StarLabsLtd/firmware #106: [Starbook MkVI-intel][ITE]laptop getting very hot

aaron-gh, 2023-05-06:
Hi,

EC version 1.12 and AMI version 1.3:

The fans do not seem to incease in speed as early as they shoudl to keep the laptop effectively cool, to the point where it is uncofortable to use on your lap or type on the keyboard for an extended period of time.

This issue is not present when using coreboot or when downgrading EC to 1.01.

Sean-StarLabs, 2023-05-06:
I'd suggest using the aggressive fan curve

aaron-gh, 2023-05-06:
Hi,

is it possible to change the fan curve in the AMI bios?

The issue does not happen when using coeboot, even on normal fan curve, just AMI.

Sean-StarLabs, 2023-05-06:
Yup

AMI can reach higher TDP

aaron-gh, 2023-05-06:
Perfect, where are these options?

Totally blind, so making an attempt to memorize the layout. e.g. a right 9 down is the battery charge limit (what are the names of the options I'm looking for in this case? I can probably find it from there.)

Sean-StarLabs, 2023-05-09:
If it's easier, you could build `ectool` from the coreboot repo, then set the fan mode with:
`sudo ectool -w 1b -z bb`

(Normal is `00`, Aggressive is `bb` and Quiet is `aa`)

aaron-gh, 2023-05-09:
Hi,

I’m asuming this wouldn't work with the ami bios?

With coreboot I can set it already, but suspend issues stop me from using that on my primary laptop.

aaron-gh, 2023-05-09:
> It would; that's why I mentioned it :D

Oh awesome!

Is there a guide to do this, and out of interest what else could be changed this way

Sean-StarLabs, 2023-05-09:
No, there won't ever be - ectool is designed for debugging, and writing to the EC should be done sparingly. That's the only value that isn't hooked to a EFi variable, so all the others would be overridden shortly anyway.

closed 2023-06-22 by Sean-StarLabs