StarLabsLtd/firmware #108: [StarBook Mk VI][EC] Charging thresholds

neeshy, 2023-06-07:
Being able to set charging thresholds (i.e. start charging at <=40%, stop charging at >=60%, via `/sys/class/power_supply/BAT0/charge_control_{start,end}_threshold`) would be nice.

Sean-StarLabs, 2023-06-12:
The problem with that is that it creates a dependency on being able to write to ACPI. Is it the greater over charge threshold you are after, or just being able to set it from the OS?

closed 2023-06-22 by Sean-StarLabs

neeshy, 2023-06-23:
Sorry for not responding sooner, I'm just now seeing your comment. Basically, I'm looking for a way to replicate the charging threshold behavior that many Thinkpads have. If `start_threshold < end_threshold` then once the battery level reaches the `end_threshold` the battery will start discharging. Once it reaches `start_threshold`, it will start charging again. Repeat ad infinitum. Currently only the upper bound is implemented.

Being able to set both values from the OS would be ideal, but it's not a big deal if it can only be set from the BIOS.

neeshy, 2023-06-23:
pinging @Sean-StarLabs

Sean-StarLabs, 2023-06-24:
> Currently only the upper bound is implemented.

No, the lower is but it's fixed at 5%. Increasing that would increase battery wear, which is why I'm hesitant to make it configurable. Just because the Lenovo has the option, doesn't mean it's a good idea ;)

neeshy, 2023-06-24:
> No, the lower is but it's fixed at 5%.

I don't think I quite follow. Do you mean that the laptop will automatically shut down once it gets to 5%? I should have clarified that by "lower bound" I meant `start_threshold`, and by "upper bound" I meant `end_threshold`.

For the Thinkpad case, if the charger is plugged in and the battery level reaches the end threshold, the battery starts to discharge _while the charger is plugged in_. As far as I can tell, the StarBook does not possess this behavior; i.e. no matter what, if the charger is plugged in it will charge. Below 5% or above it. If I set the battery limit to, say, 60% in the BIOS, this would be the equivalent of setting both start and end thresholds to 60% on a Thinkpad. For StarBooks, it would be more accurate to say that the start threshold is implemented but identical to end threshold.

> Increasing that would increase battery wear, which is why I'm hesitant to make it configurable.

On the contrary, letting the battery level go anywhere near either extreme (0 or 100) will lead to the greatest wear (for different reasons). Making sure the battery level always hovers somewhere around the middle with a small, but not excessively small gap (5-15%, to avoid trickle charging at a single set battery level) will lead to the least wear.  Basically, you want shallow charge-discharge cycles without trickle charging. Being able to configure a start threshold addresses the second part.

> Just because the Lenovo has the option, doesn't mean it's a good idea ;)

Sure, but Dell, CLEVO, and many other manufacturers have it too. It's just usually only configurable in the BIOS most of the time.

Not sure if you'll get a notification since this issue is closed, so I'm going to ping you just in case @Sean-StarLabs

Sean-StarLabs, 2023-06-24:
> Do you mean that the laptop will automatically shut down once it gets to 5%?

No, using your term of "lower bound" - that's fixed at 5%.

> .As far as I can tell, the StarBook does not possess this behavior

It does

> If I set the battery limit to, say, 60% in the BIOS, this would be the equivalent of setting both start and end thresholds to 60% on a Thinkpad. 

I'd assume it's the equivalent of 60% and 55%.

> On the contrary, letting the battery level go anywhere near either extreme (0 or 100) will lead to the greatest wear (for different reasons). Making sure the battery level always hovers somewhere [...]

No, we're not trickle-charging. We are not charging - when overcharge protection is active, the only drain on the battery is when hybrid power is required, When charging starts again, it's a rate that's optimal for the battery, which factors in things like charge level, wear and temperature. The smallest amount is optimal for reducing battery wear, and 5% is realistically the lowest you can go to.

> Not sure if you'll get a notification since this issue is closed

I get them.

neeshy, 2023-06-24:
Ah, looks like I misunderstood your earlier post. So, by setting the value to 60% it's _like_ setting `start_threshold` to 55% and `end_threshold` to 60%? You meant "5%" as a relative value? That's pretty much what I was going to set it to anyway lol.

A bit confusing since the battery level on my OS always reads 60%. Is it supposed to be that way?

Sean-StarLabs, 2023-06-25:
>  So, by setting the value to 60% it's like setting start_threshold to 55% and end_threshold to 60%?

Yes

> You meant "5%" as a relative value?

5% less than the charge limit

> Is it supposed to be that way?

Yes. Recently switched the Intel ones so they'll show raw values, had mixed feedback. Will likely do the same with the AMD ones though.