StarLabsLtd/firmware #109: EFI stub: WARNING: Failed to measure data for event

R-VdP, 2023-06-12:
On my starbook VI (Intel, coreboot), during bootup, right after the boot menu (systemd-boot), I see the following messages:
```
EFI stub: WARNING: Failed to measure data for event 1: 0x0000000000000007
EFI stub: WARNING: Failed to measure data for event 0: 0x0000000000000007
```
which seem to come from the Linux EFI stub (https://github.com/torvalds/linux/blob/858fd168a95c5b9669aac8db6c14a9aeab446375/drivers/firmware/efi/libstub/efi-stub-helper.c#LL269C12-L269C45).

The status code corresponds to `EFI_DEVICE_ERROR` (https://github.com/torvalds/linux/blob/858fd168a95c5b9669aac8db6c14a9aeab446375/include/linux/efi.h#L38).

Is this due to coreboot/EDK, or should I look higher up in the chain? Is this maybe related to the ongoing work of getting measured/secure boot to work with coreboot?

Sean-StarLabs, 2023-06-12:
What change did you make to make that appear?

R-VdP, 2023-06-12:
I was on AMI before, which showed that it successfully measured the PCRs,, but two days ago I switched to coreboot again (8.43 from testing) and ITE 1.13. Since then I see this message every time when I reboot.

Sean-StarLabs, 2023-06-14:
> Is this maybe related to the ongoing work of getting measured/secure boot to work with coreboot?

Spot on. I'm planning to make a build without that today, that'll make it to stable.

R-VdP, 2023-06-17:
It seems that the warnings are gone with 8.50 indeed.
However, is it normal that all firmware settings get reset to the default values when the firmware is updated?

Sean-StarLabs, 2023-06-19:
Yes, for coreboot, that's normal

closed 2023-06-22 by Sean-StarLabs

Ammar64, 2023-07-25:
I installed Pop OS 22.04 LTS and switched from ubuntu 23.04 (because it wasn't stable) and I see it every time I boot it's annoying to see it
I wonder if it would affect anything on my device and if there was a sulotion for it to disappear.

Sean-StarLabs, 2023-07-25:
What version and what device?

Ammar64, 2023-07-25:
I have an HP EliteBook 745 g5 laptop
CPU: AMD Ryzen 5 pro 2500
GPU: integrated AMD Radeon Vega 8 graphics

for the version I have Pop OS 22.04 LTS if you mean something else as I expect you do I have no idea what are you talking about.

Ammar64, 2023-07-25:
Oh sorry I realized this is a firmware project for some reason I thought this is the linux repo I'm not familiar with git sorry
I don't use Star Labs firmware I'm using the default one that comes with the HP laptop
I faced this problem and googled it up and this issue was the first result I got
but if you tell me a solution I would be so grateful

Sean-StarLabs, 2023-07-26:
You'll need to speak to HP