StarLabsLtd/firmware #111: [StarLite Mk IV] [Coreboot] Machine sometimes doesn't resume from sleep

ak-coram, 2023-06-23:
The power led is on, but the screen doesn't turn back on. I can still toggle one of the other leds by pressing the lock key. I can shutdown the machine by holding the power button.

I've been encountering this issue since updating to the 8.37 BIOS (I'm not sure what version I've been running before, but it was a stable coreboot release).

Version info: 
```
Linux version 6.3.8_1 (voidlinux@voidlinux) (gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39) #1 SMP PREEMPT_DYNAMIC Wed Jun 21 09:47:38 UTC 2023

Vendor: coreboot
Version: 8.51
Release Date: 06/22/2023
```

I'm periodically getting these errors in `dmesg` while running the machine:
```
[   86.401253] ACPI Error: Aborting method \_SB.PCI0.IPCM due to previous error (AE_AML_LOOP_TIMEOUT) (20221020/psparse-529)
[   86.401302] ACPI Error: Aborting method \_SB.PCI0.TSTM due to previous error (AE_AML_LOOP_TIMEOUT) (20221020/psparse-529)
[   86.401323] ACPI Error: Aborting method \_SB.PCI0.XHCI._PS0 due to previous error (AE_AML_LOOP_TIMEOUT) (20221020/psparse-529)
```

I run `powertop --auto-tune` on boot.

Sean-StarLabs, 2023-06-23:
Does it still do it if you don't run powertop?

ak-coram, 2023-06-24:
I haven't been able to reproduce this so far without running powertop / tlp. Battery life got worse though.

ak-coram, 2023-07-02:
I've had the issue again without powertop, I think it just became much more rare.

Sean-StarLabs, 2023-07-04:
Just once? Was it the first time it was suspended after being powered on, or subsequent?

ak-coram, 2023-07-04:
I was a subsequent suspend, it worked several times before without any issues. I haven't had this since, so it might be a fluke.

sim31, 2023-07-09:
I had this happen a couple of times on starbook mk VI. Except I'm not running powertop and don't get these ACPI errors. Running coreboot 8.40. Although, now that I think of it, I'm not sure if it even fell asleep during those times... I closed the lid and later when I took it out, it had the power led on as if it was not sleeping, but the screen was black. 

Not sure if I should create a separate issue.

Sean-StarLabs, 2023-07-10:
@sim31 Definately not the same issue, sounds more like #105 or #111 - not much point looking into an old release though. Please try `8.51`.

ak-coram, 2023-07-11:
I've experienced this issue again. It may be of note that when I press the power button while in this state the machine immediately turns off.

Sean-StarLabs, 2023-07-21:
Should be fixed in `8.60`, which should make it to stable mid-next week, feel free to re-open if not.

closed 2023-07-21 by Sean-StarLabs