StarLabsLtd/firmware #112: Labtop mkIII doesn't resume from sleep when I close the lid

barbastrejo12, 2023-06-24:
Have been seeing this behavior after upgrade to 8.50. I close the lid. If I let the lid open it resumes.

barbastrejo12, 2023-06-26:
Had a strange behavior yesterday, the L3 had % battery, I plug the USb-C charging cable it turns off, I try to turn it on many times for 1hour, no sign of life, red led on. I unplug the cable, leave alone for around 30minutes, press power and it starts at 47% battery.

Sean-StarLabs, 2023-07-21:
`8.60` is going up today, which'll fix a couple of things. You are flashing these with an unsupported method, before they even hit the testing remote - so you'll likely find bugs ;)

closed 2023-07-21 by Sean-StarLabs

barbastrejo12, 2023-07-22:
Thanks for the info Sean. Supported method is fwupd?

Sean-StarLabs, 2023-07-24:
Yup, that's right - it does clever stuff to get the CMOS in the right state, and that's where things like the backlight state and suspend info is stored.