StarLabsLtd/firmware #113: [StarLite Mk IV] [Coreboot] F-keys act the same regardless of holding the fn key

ak-coram, 2023-06-25:
Since BIOS version 8.51 I can't use the `F11` key anymore, it always controls the keyboard light and it doesn't matter if the `fn` key is held down (I don't even get any X events for it). A bunch of other F-keys seem to exhibit similar behavior, but notably F1 works as expected (so it doesn't seem to be an issue with the `fn` key itself).

Sean-StarLabs, 2023-06-25:
You sure it's just not function lock being on? That's the only thing that the BIOS can do to influence the keyboard, beyond that, it has nothing to do with it.

ak-coram, 2023-06-25:
You're right, my mistake! Sorry

closed 2023-06-25 by ak-coram

alexeymuranov, 2025-06-02:
I noticed a similar problem, and I do not see how the current behaviour can be expected: when the Fn lock light is **off**, some Fn keys do not function like Fn keys regardless of whether Fn modifier key is pressed. For example, F5 cannot be used to go to the presentation mode in a PDF viewer, and F10 cannot be used to save modifications in "BIOS". Some other Fn keys, like F2, still react to the Fn modifier key though.

alexeymuranov, 2025-06-12:
> What EC are you running?

```console
$ cat /sys/class/dmi/id/ec_firmware_release
1.4
```