StarLabsLtd/firmware #116: [Starbook MkVI - Intel][coreboot] Screen tearing/graphics issue

andr5w, 2023-07-24:
I'm seeing what appears to be some constant screen tearing and/or flickering (shown in the video below). The tearing/flickering occurs on both Wayland and XOrg sessions. The issue occurs immediately after a fresh OS installation.

I tried to read up on the Intel i915 driver(s) and subsequently tried setting several kernel boot parameters suggested in various support forums (e.g., `i915.enable_psr=0`) but I wasn't able to resolve the issue.

Any thoughts on how to go about troubleshooting or resolving this? Thanks.

https://github.com/StarLabsLtd/firmware/assets/75326688/bbd6c967-8570-415a-b44e-262a732fa0ac

**Machine Specs:**

- Starbook MkVI - Intel
- coreboot v8.50
- ITE/EC v1.13
- Ubuntu 23.04 (with LVM and LUKS)
   - Kernel version 6.2.0-26-generic
- 12th gen i7-1260P
- Samsung 980 Pro SSD
- 32GB crucial ram

Sean-StarLabs, 2023-07-25:
Did you have it before updating to 8.50?

andr5w, 2023-07-25:
> Did you have it before updating to 8.50?

Yes.

Sean-StarLabs, 2023-07-25:
Cool. Does

`i915.enable_dc=0 intel_idle.max_cstate=2` stop it from occurring?

andr5w, 2023-07-25:
Unfortunately not.

Sean-StarLabs, 2023-07-25:
Interesting. Just to check, how are you applying the kernel parameters?

Is there any pattern as to whether it shows on battery or power?

Can you jump up to 8.60? There were some timing changes in there, probably unrelated but might as well check if they've inadvertently solved it.

andr5w, 2023-07-25:
Sure thing. I add the parameters to `/etc/default/grub` like so:

`GRUB_CMDLINE_LINUX_DEFAULT="... i915.enable_dc=0 intel_idle.max_cstate=2 ..."` (`...` represents parameters not listed here)

and I then run `sudo update-grub` and then reboot.

The tearing/flickering occurs both when it's on battery and/or connected to a power source.

I updated to coreboot v8.60 but the issuesremains.

This could be a shot in the dark, but now that I'm thinking about it the Starbook came preinstalled with Ubuntu but disk encryption was _not_ enabled by default (if I remember correctly) and I did not see the issue when I used the Starbook for the first time. I installed POP!_OS (which configures disk encryption by default) and then immediately saw the tearing. I thought maybe there was a problem with POP!_OS and reinstalled Ubuntu but when I did I enabled disk encryption and now I see the tearing/flickering. Not sure if the two are related in any way? If I have time in the next day or two I'll try and test this hypothesis.

andr5w, 2023-07-27:
Update: I reinstalled Ubuntu 23.04 with no disk encryption but the screen tearing was still there, so I was wrong. :(

I did, however, do a bit more poking around and discovered that I'm able to reliably get the tearing to stop by turning wifi off. As soon as I turn wifi back on the tearing shows right back up. Turn off wifi, tearing stops. Turn wifi on, tearing starts.

Any thoughts as to how the wifi connection and the screen tearing could be related?

Sean-StarLabs, 2023-07-28:
If I remember correctly, that's a bug in the ucode, and only effects 5GHz networks. Are you able to test on 2.4GHz?

andr5w, 2023-07-28:
Yes! I'm able to see a difference when switching between 5GHz and 2.4GHz wifi networks, although for me it appears to be the reverse of what's stated above: I've seen no tearing (so far) when connected to a 5GHz wifi network, but the tearing starts as soon as I connect to a 2.4GHz wifi network.

Are there ucode patches/updates that fix/address this bug?

Sean-StarLabs, 2023-07-31:
I think they're only available for 23.04 now; and even that can't load the latest ones.

You can try using an older one:

```sudo dmesg | grep ty-a0-gf-a0```
Will show something like:
```[    2.908556] iwlwifi 0000:01:00.0: loaded firmware version 72.a764baac.0 ty-a0-gf-a0-72.ucode op_mode iwlmvm```

Look for the version number, `72` above, and then use the below with that version number to rename it so it won't load:
```sudo mv /lib/firmware/iwlwifi-ty-a0-gf-a0-72.ucode /lib/firmware/iwlwifi-ty-a0-gf-a0-72.bak```

andr5w, 2023-08-02:
No luck. I tried loading an older version as suggested above, but no success. I booted from an Arch Linux usb with the latest kernel (`6.4.7`) which in turn loaded iwlwifi ucode version `78`, but the flickering/tearing is still there.

Anything else I can try?

Sean-StarLabs, 2023-08-09:
If you switch bluetooth off, does that make a difference?

andr5w, 2023-08-14:
Toggling Bluetooth on/off doesn't make a difference unfortunately.

Sean-StarLabs, 2023-08-17:
Can you try disabling the power saving? You can find the name if your wireless card with `iwconfig`, it'll look something like:
```wlp44s0   IEEE 802.11  ESSID:"Star Labs"  ```
And then turn off power saving with:
```sudo iwconfig wlp44s0 power off```

andr5w, 2023-08-18:
I toggled the iwconfig power setting back and forth for the wireless card but it didn't make a difference for me. :(

andr5w, 2023-09-12:
Update: I tried booting from a Manjaro usb with kernel v6.5 and checked that the latest ucode loaded (which it did), but the issue still persists. I even tried loading Intel's experimental `xe` graphics (`i915.force_probe=!46a6 xe.force_probe=46a6`) but the system just hung.

It sounds like the issue I'm experiencing is not widespread (for which I'm grateful!). What are the chances that I have defective hardware? How could I tell? I'm willing to purchase a new wifi card but if the gpu is the issue then that won't resolve anything, right?

Sean-StarLabs, 2023-09-12:
Fixed in `8.94` which is now stable

Sean-StarLabs, 2023-09-12:
> Fixed in `8.94` which is now stable

Wrong issue....

> What are the chances that I have defective hardware?

Near zero but for PIC, you can drop the transmit power:
`sudo iwconfig wlp44s0 txpower 5`

That'll hit performance, but if you start at 5, and work up if it's too slow. 

If that does, let me know and I'll poke an Intel guy for an ETA on a fix.

andr5w, 2023-09-13:
Yep, playing with the `txpower` level made a difference, and so far keeping the level low enough has greatly reduced the tearing. I can limp along with this workaround for now, but I would be interested in hearing about Intel's ETA on a fix. Really appreciate the help!!

Sean-StarLabs, 2023-09-22:
Apparently its been fixed - just trying to find out in what and what version!

Sean-StarLabs, 2023-10-02:
Still haven't got a conclusive answer, but `8.97` contains a new version of FSP, which seems to be related. That'll go stable tomorrow so please give that a try

andr5w, 2023-10-03:
I gave it a try and updated to `8.97`, but no luck - issue is still there. :(

Sean-StarLabs, 2023-11-21:
Not getting any conclusive info - can you ping support a link this ticket? I think swapping out will be a safe step, at least to give me some ammo!

closed 2024-01-03 by Sean-StarLabs