StarLabsLtd/firmware #119: System freezes on dpms standby without kernel parameters

koeppl, 2023-07-27:
I switched yesterday to coreboot v8.50,  and experience from now on that the system freezes when the system screensaver starts, which gets triggered via `xset s` . Pressing buttons sometimes results in a white flashing of the screen with gray vertical bars. At this state, the system is no longer responding such that only a forced shutdown by a long power button press works.
The behavior can also be triggered deterministically via `xset dpms force standby`.

The kernel parameters `i915.enable_dc=0 intel_idle.max_cstate=2` suggested in https://github.com/StarLabsLtd/firmware/issues/116 helped me to resolve the problem. 

I thus wonder whether these parameters are necessary by default, or whether a bug has been introduced in the new coreboot version.

I experienced the problem with Debian 11 and Debian 12. The notebook is a StarLite Mk IV.

Sean-StarLabs, 2023-08-03:
Could be - possibly even the same as #111. Will get `8.70` to the testing remote - should be available inside of 24 hours - can you try that?

Sean-StarLabs, 2023-09-12:
Fixed in `8,94` which is now stable.

closed 2023-09-12 by Sean-StarLabs