StarLabsLtd/firmware #121: [StarBook Mk VI - AMD] Periodic system stutters / sudden reboots

dthuerck, 2023-08-15:
Hi all,

I'm running my Starbook under Ubuntu 22.04 using Mainline Kernels. Under pretty much every kernel version after 6.1, I'm experiencing semi-periodic "system stutter", i.e. Audio, Video, Cursor etc. stutter for about 2-3 secs before things return to normal. From what I could find, this is a known issue with AMD (f)TPM and requires a BIOS update to be fixed:

https://www.amd.com/en/support/kb/faq/pa-410

Do you have any plans for including the update into the next AMD BIOS version?

Furthermore - not pertaining to this issue directly, but for general information - the mileage varies a lot through different kernel versions as I've experienced. Mostly, apart from the stuttering, I have sudden reboots about twice a week or suspend is working unreliably. The good news here is that in the most recent kernels (>= 6.4.3) the suspend issues seem to be resolved, it's pretty much working smoothly for me now.

dthuerck, 2023-08-15:
Here's another discussion on the subject:

https://discussion.fedoraproject.org/t/periodic-full-system-stutters-for-2-3-seconds/71549

Sean-StarLabs, 2023-08-15:
Already on it - just fixing on regression :)

Sean-StarLabs, 2023-09-12:
Closing as the latest build uses 1.0.0.fa that fixes #75 is just going through testing now, should be up tomorrow

closed 2023-09-12 by Sean-StarLabs

dthuerck, 2023-10-09:
Sorry @Sean-StarLabs , but I think I have to reopen this - from what I get from the other threads, there's now stable new firmware yet (for AMD) that hasn't received attention here due to bricking laptops.

My current setup (Ubuntu 23.10, 6.2, Firmware 1.7.0) crashes roughly once per day - I can't use this thing as daily driver for my work anymore.

Sean-StarLabs, 2023-10-09:
> from what I get from the other threads, there's now stable new firmware yet (for AMD) that hasn't received attention here due to bricking laptops.

Not sure what you mean - but it's not stable yet, I'm just packing that up now. 

> My current setup (Ubuntu 23.10, 6.2, Firmware 1.7.0) crashes roughly once per day - I can't use this thing as daily driver for my work anymore.

Ping support - that's not the fTPM, if it's crashing

dthuerck, 2023-10-24:
For reference in case people find this later: with 1.7.0 and Ubuntu 23.10's kernel 6.5, both issues (stuttering and hangs/reboots) are gone - at least haven't encountered them in 2 full days!