StarLabsLtd/firmware #122: Publish source and build instructions

R-VdP, 2023-08-20:
Since the starlabs firmware is marketed as being open source, could you publish the source code and the build instructions to build the firmware blobs?

I understand that there are still proprietary blobs that are included, but since I suppose that those are redistributable, would it be possible to share the build scripts, the source code, and those binary blobs in a way that allows us to build the firmware ourselves?

Are you running pure upstream coreboot + edk or do you maintain your own patches?

I find the whole thing not very transparent given that you market it as open source.

Sean-StarLabs, 2023-08-20:
Everything is upstream (save for the CFR patches as theres some faffing going on between edk2 and coreboot for that), including the steps to build. Is there anything you want beyond that?

R-VdP, 2023-08-20:
So can we reproduce the binaries that are included in this repo from the upstream sources? Or does anything else go in there?
What I would be interested in, is the exact instructions to reproduce the firmware published in this repo.

Sean-StarLabs, 2023-08-21:
Cloning from our fork of coreboot would produce the same binaries.

Cloning from upstream would produce similar:
- handful of patches not yet upstream like CFR
- it'll only have the BIOS region - which is the only bit you'll need anyway

closed 2023-09-06 by Sean-StarLabs