StarLabsLtd/firmware #124: [LabTop MkIV]  power management issues

macmpi, 2023-09-06:
coreboot 5.60 (and earlier) / 5.15.0-82-generic #91~20.04.1-Ubuntu SMP Fri Aug 18 16:24:39 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
(elementary 6.1)

Laptop abruptly looses power without any system low-power warning (reported capacity around 28-30% before power loss).
In Suspend mode (manually activated and lid closed), battery seems to continue to slowly drain.

Discussed within support ticket, but reporting here for records.
Seems like some EC/calibration update is in the works that may help resolve the issue.

Sean-StarLabs, 2023-09-06:
Do you mean `8.60`? But yes, there's an EC update in the works that'll improve the calibration - but they will drain power whilst suspended, that's normal

macmpi, 2023-09-06:
> Do you mean `8.60`?

Yes indeed, sorry for the typo.

macmpi, 2023-09-16:
It seems 8.94 for L4 been pulled after a few days: is it recommended to eventually downgrade to 8.60 then?
https://fwupd.org/lvfs/devices/com.starlabs.L4.coreboot

Sean-StarLabs, 2023-09-18:
Only if you see issues, it was pulled as the 08 microcode seems to make a few distros freeze

macmpi, 2024-01-22:
> But yes, there's an EC update in the works that'll improve the calibration

Dear @Sean-StarLabs, would you have any estimate on release for such fix? Thanks for any feedback.

Sean-StarLabs, 2024-01-22:
No - it's too hard to predict time!

Sean-StarLabs, 2024-02-22:
Fixed with 406ae53abfb07e3e84f5a947c0c3a627c9ca2388

closed 2024-02-22 by Sean-StarLabs

macmpi, 2024-02-25:
Lost all Fn keys after (long) sleep after the update.
Hitting fn key simultaneouly does not help.
Lock has no visible influence.
Trackpad is also disabled…
Reboot did not help.

Sean-StarLabs, 2024-02-25:
"the update" is coreboot 24.02? If so, it sounds like the EC wasn't updated - do you know what that was?

macmpi, 2024-02-25:
yes 24.02 it was. I am now back to 8.99.
EC remained 1.07 all the way.

Sean-StarLabs, 2024-02-25:
That'll be it. You can just manually update it with:
```
sudo wget https://github.com/StarLabsLtd/firmware/raw/master/binaries/ifu.efi -O /boot/efi/ifu.efi
sudo wget https://github.com/StarLabsLtd/blobs/raw/main/mainboard/starlabs/starbook/cml/ec.bin -O /boot/efi/ec.bin
```
Then, with the charger connected, reboot to the EFI shell and enter the below commands (there is a chance it could be fs1, fs2 etc, depending on your setup. If you get a message saying can't find ifu.efi, please try them in ascending order):
```
fs0:
ifu.efi ec.bin 128
```

macmpi, 2024-02-25:
Thanks.
Before I do, what's the version it should show after that update?
Will this EC work with 8.99, or only with 24.02?

Sean-StarLabs, 2024-02-25:
24.02

It'll "work", but things won't line up, so you'll lose the same things (function keys, backlight timer etc)

macmpi, 2024-02-26:
EC and coreboot now on 24.02 (fs0 was fine).
Thanks for great support!

Isn't normal firmware update supposed to also update EC if needed?

Sean-StarLabs, 2024-02-26:
It would have done, likely, just hadn't applied it yet as there's a long list of requirements for an EC update to apply automatically

macmpi, 2024-02-28:
Just noticed:
>[35321.550117] ACPI: battery: [Firmware Bug]: (dis)charge rate invalid.

[LabTop_2024-02-28_16-09-55.zip](https://github.com/StarLabsLtd/firmware/files/14435953/LabTop_2024-02-28_16-09-55.zip)

Sean-StarLabs, 2024-02-28:
> [35321.550117] ACPI: battery: [Firmware Bug]: (dis)charge rate invalid.

5.15 won't support it