StarLabsLtd/firmware #126: [StarBook MkVI - AMD] riddled with issues after update to ami 1.8

indeedhat, 2023-09-15:
I updated my StarBook last night to the latest firmware (ami 1.8) and it is having a really bad time.

The biggest issue is that it no longer charges. I only have a usbc charger so i don't know if it will charge from the barel jack but it no longer charges via usb c. I have tried 2 seperate starlabs chargers, 3 different usb cables and a switch charger (don't know if the switch one worked before or not) no combination will charge the laptop.

I didn't realise that charging was broken until i was down to about 15% so the laptop is basically useless at this point.

When i still had charge to be playing with linux failed about 10 or so start up checks, not sure what they all were but one of them was starting the audio service, this was not a problem until the firmware update.
I did run a full system update in case but this didn't fix the issue.

Lastly (or the last thing i noticed before running out of power) is that the machine is running very slowly. Its not to the point of being completely unusable but basically felt like using a laptop with a spinning hard drive in. Boot time went from maybe 20 seconds to just over 3 minutes (not sure on the pre issue time but i timed the post issue boot time). 
The system in general is pretty unresponsive.

Im not sure what if anything i can do with my machine without charge but if there is any chance of this effecting other machines then i suggest that the update be pulled until the issue is investigated.

Sean-StarLabs, 2023-09-15:
That's strange, but all explained by one denominator. How did you flash it and what distro are you using?

indeedhat, 2023-09-15:
I updated using a manjaro live cd by using fwpudmgr installed with the flashrom-starlabs package (this is how ive updated the firmware always in the past is i cant get the starlabs package on the distro im using)

indeedhat, 2023-09-15:
thanks for the quick response by the way, i wasn't expecting a response until at least monday.

esimopoulo, 2023-09-17:
Unfortunately, I'm having the exact same issues after updating. @Sean-StarLabs I've sent you an email describing the situation entirely, but to summarise in this post.

1. StarBook MK VI AMD running Fedora 38 with latest updates and BIOS 1.7.0. Works as smooth as butter. Every BIOS update since 1.1.0 has been performed without any hiccups.
2. I have the testing LVFS repo enabled. BIOS update 1.8.0 shows up on Software Centre. I initiate the update.
3. Computer restarts, StarLabs logo shows up but update doesn't happen. Boots straight into Fedora. Update still showing up on Software Centre.
4. I make the decision to do the update via the Terminal instead. `fwupdmgr update` does the trick. System reboots, BIOS is getting updated. Laptop remains plugged in via USB-C throughout the entire process. Update finished. A "click" sound is heard through the speakers immediately after reboot and the charging light goes off.
5. Since then, symptoms are exactly like @indeedhat 's. Crippled performance, laptop not charging either via USB-C or barrel jack, plugging the power cable causes the blue power light to come on for 1 second then goes away. One extra thing for me is, the laptop doesn't switch on just by pressing the Power button. I have to press and hold for a few seconds.
6. I decided to reset BIOS to the default settings. I'm not sure if it's even related but the machine is actually snappy again.
7. I used the EFI to incrementally downgrade from 1.8.0 down to 1.5.0. No luck. Laptop still not charging whatever the BIOS version, and it's basically unusable.

Sean-StarLabs, 2023-09-18:
@indeedhat It might need flashing with a hardware programmer, but if you try disconnecting the battery, and then just using the charger - that might get it started.

@esimopoulo 1-3 are fwupd or Fedora ones, if there's an issue, I'd be surprised if they weren't already working on it as the LVFS guys are Fedora guys.

> Crippled performance, laptop not charging either via USB-C or barrel jack, plugging the power cable causes the blue power light to come on for 1 second then goes away. 

@indeedhat only had a USB-C cable. What does the LED do when you connect the DC Jack?

> I decided to reset BIOS to the default settings. I'm not sure if it's even related but the machine is actually snappy again.
> I used the EFI to incrementally downgrade from 1.8.0 down to 1.5.0. No luck. Laptop still not charging whatever the BIOS version, and it's basically unusable.

Unusable, as you can't charge it (assuming no. 6 is still the case)?

>  downgrade from 1.8.0 down to 1.5.0

The BIOS doesn't affect charging, that is down to the EC. Updating the BIOS will update the EC, but downgrading won't.  There are a few ways to downgrade the EC, but it depends on what the chargers do when connected. Let me know on the above and I can advise from there.

pastyfiend, 2023-09-18:
Hi all, have posted initially in #75 Sean as you know, but just to add here: exact same problem, but I do have a DC jack and it's charging normally with that, but nothing via USB.  Also manjaro, used fwupdmgr.

esimopoulo, 2023-09-18:
Hey @Sean-StarLabs thanks for the response. I did a proper test this time, negating one of my points from my earlier message and made some more observations as well.

To answer your questions first: 

> @esimopoulo 1-3 are fwupd or Fedora ones, if there's an issue, I'd be surprised if they weren't already working on it as the LVFS guys are Fedora guys.

1-3 was performed using Fedora Software Centre. As described, no update happened at this stage. Step 4 onwards is when I switched to fwupd and managed to update.

> > Crippled performance, laptop not charging either via USB-C or barrel jack, plugging the power cable causes the blue power light to come on for 1 second then goes away.
> 
> @indeedhat only had a USB-C cable. What does the LED do when you connect the DC Jack?

So the latest is:
- When the laptop is switched off, plugging in the USB-C cable causes the power light to come on for 4 seconds, then a clicking sound through the speakers, then machine dies. When the laptop is switched on, it just does nothing.
- **Plugging in the DC jack switches on both power and charging light, and the laptop charges normally**. Not sure if I was a total noob last time I tried, maybe I didn't plug it in properly, but the laptop actually charges fine with the DC jack, either when switched on or off.

> Unusable, as you can't charge it (assuming no. 6 is still the case)?

Correct. Laptop is snappy. However, it seems that the USB-C port is not working ok.
- Connecting a USB-C external monitor (with power delivery) doesn't work (no video output and no charging).
- Connecting monitor via a USB-C docking station doesn't work either. However, all peripherals I have connected on the docking station work fine (mouse, keyboard, webcam).
- Connecting an HDMI external monitor via a USB-C dongle doesn't produce any video output. Similar to the docking station case, other peripherals connected to the dongle such as flash drives, mouse etc work fine.

Fedora Software centre popped a new update for me (I was on 1.5.0, and 1.7.0 came up). Clicked Update, and it went through just fine. I'm now on 1.7.0, performance is normal, the only thing not properly working is the USB-C port.

indeedhat, 2023-09-18:
> if you try disconnecting the battery, and then just using the charger - that might get it started.

Sadly no this didn't help, for completeness with no battery i get no lights and it does not turn on, after putting the battery back in the blue power LED comes on for about 3-4 seconds (slightly longer than before i removed the battery) then the behavior is the same as before.

> It might need flashing with a hardware programmer

Is that just in general or is that just without a working battery/charger? 
if a fw fix becomes available and it does charge form the barrel jack then i can just take it into work and hook it up to one of the bench supplies to get it flashed (assuming i can find the correct size barrel jack)

This is a total tangent so feel free to ignore the rest of this message but i would just like to express my appreciation for Starlabs approach to support. Owning a Starbook has unfortunately not been without its frustrations (clearly), admittedly the main issue i have had has largely been out of star labs control (looking at you AMD).

It is refreshing to be able to speak to someone who isn't working from a script and who's solutions don't only amount to:
- reinstall your OS
- send it back and wait for 3 weeks
- swivel.

In this case it didn't work sure but there are not a lot of other companies that would give a suggestion like open up your machine and remove the battery as a possible interim measure to get you back up and running again.

Sean-StarLabs, 2023-09-18:
> Is that just in general or is that just without a working battery/charger?

Without a way to power it, either a working charger or power battery.

If you've got a way to power it, reverting would be possible. Just let me know if you want steps.

As for a fix, I don't think it'll take overly long to fix the USB-C port, but I'll update here tomorrow. In the meantime, please swivel :)

pastyfiend, 2023-09-18:
Completely agree with @indeedhat - everything working perfectly all the time isn't realistic or probably even reasonable to expect, but the responsiveness of the customer support at StarLabs is really impressive - the best I think I've had from any company in tech or elsewhere.

esimopoulo, 2023-09-19:
> If you've got a way to power it, reverting would be possible. Just let me know if you want steps.

Might be worth posting it here for all affected peeps Sean?

Sean-StarLabs, 2023-09-21:
This will be a specific solution for this firmware, so please don't try it in the future.

If you download these two:
https://github.com/StarLabsLtd/firmware/raw/master/StarBook/MkVI-AMD/misc/ec.bin
https://github.com/StarLabsLtd/firmware/raw/master/binaries/ifu.efi

1. Copy the files to a FAT32 USB drive
2. Connect the DC Jack (if the charging light isn't lit, this will fail spectacularly)
3. Boot to the EFI Shell (similar to https://support.starlabs.systems/kb/firmware/efi-shell)
4. Type in:
```
fs0:
ifu.efi ec.bin 128
```

After that, it'll be good.

Hoping to have a new update the start of next week to fix "easily"

indeedhat, 2023-09-21:
Thanks Sean, much appreciated.

because im gonna be running from a bench PSU, is the barrel jack center positive or center negative?

Sean-StarLabs, 2023-09-21:
Positive center.

indeedhat, 2023-09-21:
That worked a treat, thanks Sean.

For anyone else doing this:
- my usb showed up as fs1 not fs0 so you might need to change it
- one its done flashing the system shuts down and doesn't restart, don't panic like i did, it's all good

esimopoulo, 2023-09-22:
Thanks Sean, can confirm it worked for me too. Also, thanks @indeedhat my USB drive also showed up as FS1, I got an error trying to update the first time using FS0.

Also, separate question @Sean-StarLabs , I noticed you uploaded a new version of the 1.8.0 BIOS a couple of days ago. Assuming this includes the updated version of the EC software? Currently running 1.7.0. Would I be safe to update to 1.8.0 from that perspective?

Sean-StarLabs, 2023-09-22:
> a new version of the 1.8.0 BIOS

The answer is "probably" (I haven't tested that path), but long story short, you'll be better off with what you've got now.

pastyfiend, 2023-10-02:
Hi Sean, any update on a new firmware version that we can update to in the usual way that will correct this?  I haven't tried the EFI fix yet as holding out for the firmware, but if it's going to be a while I may have to bite the bullet and try it!

Sean-StarLabs, 2023-10-02:
Almost there, but just ironing out a small few things (like fan, lights, etc) that have changed due to the new AGESA. They're taking a long time to test, so I think worst case, I'll have something in the testing remote at the end of the week.

Sean-StarLabs, 2023-10-09:
`1.12` now in the testing remote - the charging light has to bit lit, so if anyone wants me to split of an EC update to fix the charging first, just let me know (although, the above one would do it).

indeedhat, 2023-10-09:
I've tried updating to 1.12 fwupdmgr does its thing but the system just sits on the starlabs bios logo on reboot and never starts the update, i left it for over an hour and it was still sat on the logo.

i tried changing the boot order to perfer bios updates just for the sake of it but still no dice.

To be clear the laptop is still running fine cutting the power has it starting up as normal with the 1.8 firmware

Sean-StarLabs, 2023-10-09:
Was that on Manjaro? Which version of fwupd (`fwupd --version`)?

pastyfiend, 2023-10-10:
Have a slightly different problem - the update installed fine, but I'm still not able to charge via USB-C (or indeed connect to external monitors through a dock etc.)  Sorry Sean!

pastyfiend, 2023-10-10:
Any trick to this Sean?  E.g. when to do it (starting off/on etc.), barrel power in/out etc?  Have tried several times but can't crack it.

Sean-StarLabs, 2023-10-10:
1. Ensure the laptop is turned **off** and connect the charger.
2. Press and **hold** the `Power` key.

   - Ignore any activity on the screen, including messages or the screen turning on.

3. Once the charging LED turns **off** and then back **on**, release the `Power` key.

pastyfiend, 2023-10-10:
Thanks.  Tried that several times, but no joy I'm afraid.  Any thoughts on what I might be doing wrong?

pastyfiend, 2023-10-10:
Afraid that hasn't worked either.

Sean-StarLabs, 2023-10-10:
What is the output of
```
cat /sys/class/dmi/id/bios_version
cat /sys/class/dmi/id/ec_firmware_release
```

pastyfiend, 2023-10-10:
cat /sys/class/dmi/id/bios_version
1.12.0

cat /sys/class/dmi/id/ec_firmware_release
2.3

Sean-StarLabs, 2023-10-10:
Hmm, are you able to send a video of the attempted EC reset to support@starlabs.systems?

pastyfiend, 2023-10-10:
Just to say now the EC reset is done, this is working perfectly.  Thanks Sean

indeedhat, 2023-10-10:
> Was that on Manjaro? Which version of fwupd (`fwupd --version`)?

Sorry Sean by the time i noticed the email about this reply i had already overwritten the usb drive with a newer version of manjaro so i cant say for sure. With a more recent manjaro live usb it did the update just fine. The iso i was using was 22.1 i think so pretty damn old.

So far everything seems to be working fine for me on 1.12, ill keep a close eye out for issues over the next couple of days but its looking promising. Thanks :)

> I have noticed that, since updating to 1.8 (and still with 1.12), I have more WiFi issues than previously... It's hard to be sure that this is related to the firmware (but I've tried various kernel versions), so I'm posting here. Often, the device won't detect any networks until I unload and reload the iwlwifi module; or, if it does successfully connect, then the speeds are very limited. In dmesg, I see a lot of error messages from iwlwifi (and seemingly its firmware), often related to `net/mac80211/scan.c:440 __ieee80211_scan_completed+0x2f9/0x310 [mac80211]` or `iwl_trans_txq_send_hcmd+0x376/0x380 [iwlwifi]`. Can anybody confirm or help debug this?
> 
> (Edit to add: for instance, just now, the device seemed to fail with the corresponding error message `iwlwifi 0000:02:00.0: Unhandled alg: 0x703`, after which the link speed dropped to ~600 kbps(!). Then the device reset itself and eventually re-associated, but still with a number of errors saying `Scan failed! ret -5`...)

I've had no issues with the wifi on any of the firmware versions, i have just checked my dmesg logs and im not seeing any similar logs to you.

tsmithe, 2023-10-10:
That's interesting @indeedhat; I wonder what the difference is on my system. What kernel version are you using?

Sean-StarLabs, 2023-10-11:
@tsmithe I'd look at the ucode version (grep dmesg for ucode) - that's the biggest influence on the wireless. Oversimplifying it, the firmware pretty much just turns it on.

@indeedhat When you left it overnight, was it on charge? If so, which charger? I don't need anything verifying, thanks

indeedhat, 2023-10-11:
it was fully charged when i went to bed so i unplugged it, i was using usbc for charging and thats whats not working now. I assume that like last time the barrel jack will be working but i wont know for sure until tomorrow

tsmithe, 2023-10-12:
Hi Sean, the reported ucode version is `72.daa05125.0 ty-a0-gf-a0-72.ucode`.  How does that seem?  (I will also check that the antenna cables are tightly connected...)

Sean-StarLabs, 2023-10-12:
> How does that seem?

New

Trying rolling it back to that previous version (you can just rename them). I'm using `72.a764baac.0` and zero issues there.

indeedhat, 2023-10-12:
Before reverting to 1.8 i decided to do an EC reset on the off chance.
That seems to have gotten everything working again, fingers crossed it sticks.

indeedhat, 2023-10-13:
After leaving the laptop off charge over night again it once again didn't charge via usb c and monitors didn't work. EC reset fixed the issue again but if it does the same tonight then ill have to revert back to 1.8

Sean-StarLabs, 2023-10-13:
It will - just testing a fix, will aim for Monday

indeedhat, 2023-10-14:
sounds good, in the meantime so long as i don't leave it unplugged for an extended period of time then it seems to keep working fine, i can deal with that until the fix comes through.

dthuerck, 2023-10-18:
Hi, 

I did the update to 1.13.0 yesterday. So far, haven't had any issues with charging via USB-C. However, one issue persists: In order to boot up the laptop, I need to press the "on" button for at  least 1-2 secs. If pressed for a shorter amount of time, the power light shortly flashes and I can hear a loud "pop", but the machine does not continue to boot.

indeedhat, 2023-10-20:
I just attempted the update myself but the 1.13 update no longer seems to be in the repos

Sean-StarLabs, 2023-10-23:
> I just attempted the update myself but the 1.13 update no longer seems to be in the repos

"repos" is GitHub or LVFS? Nothings changed - it was only ever in Git, didn't push it to the LVFS

indeedhat, 2023-10-24:
ah right, i was checking lvfs, i didn't realise. in light of @dthuerck's issues ill stick with 1.7 for now

Sean-StarLabs, 2023-10-27:
3 days of testing `1.14.0`, seems to fix all know bugs with any kernel. FYI - if you update with type-c connected, you'll need to do an EC reset afterwards to get the PD working

dthuerck, 2023-10-27:
Thanks @Sean-StarLabs , really appreciate your commitment to the community and the product!

Just installed 1.14.0 and will test it extensively over the next week - work-related travel coming up.

For the others: had some problems with Turbo Boost in the past (if under load, at some point the USB-C would stop working for external displays). Deactivating Turbo Boost certainly helps, but isn't great for working with the machine. After some more experimentation, I've used `ryzenadj` to throttle the TDWs a bit: `ryzenadj --tctl-temp=85 --fast-limit=25000 --slow-limit=20000 --stapm-limit=12500`. That limits the boost a little. So far, it feels great - quick responses, no more waiting time and no sudden fan bursts.

Thanks to this and the latest updates, my Starbook is now a great working machine! Thanks, guys!

rmsthebest, 2023-10-28:
I just performed the update from 1.7 -> 1.14 with dc jack power connected.
I now have the same problem described here.
No power through usb-c, no display through usb-c, other peripherals like keyboard does work.
Popping sounds if I start computer with usb-c connected.

I think I did the EC-reset (hold power button when powering on until the lights blink). Nothing changed.

```
cat /sys/class/dmi/id/bios_version
1.14.0
cat /sys/class/dmi/id/ec_firmware_release
2.6
```

indeedhat, 2023-10-29:
I updated with dc power connected, i still had to do a EC reset to get usb-c charging to work again but everything seems to be working properly for the last 24 hours so its looking good on my end.

Thanks for all your work on this :)

tsmithe, 2023-10-30:
I have found that, if I suspend the laptop by closing the lid (on battery), it seems to wake itself up immediately.  But I have also been playing with ASPM, following the idea [here](https://old.reddit.com/r/AMDLaptops/comments/16lz0oh/anyone_managed_to_get_pci_express_activestate/k4klnmg/) that it may help extend the battery life. Maybe this is the problem... (I have also noticed the keyboard dropping keystrokes when touch typing; maybe again this is related..)

Otherwise, everything seems well!  And I haven't encountered #75 again (though I haven't done an exhaustive test).

indeedhat, 2023-10-30:
when i got home from work today i had to do another EC reset to get it charging, no other issues so far though.

indeedhat, 2023-11-02:
Charging has been fine since but i am now getting brief disconnects of usb devices connected via a usbc dock, monitors seem fine but mouse, keyboard and DAC are all having periodic cut outs for a second or so.

dthuerck, 2023-11-02:
So, I've been test-driving `1.14.0` all week extensively. When charging, everything is fine. However, when I use it on battery, these are the problems I encountered:

1. Same as @tsmithe, I noticed that the keyboard frequently drops keypresses (which is a problem for coders...).
2. On 3 separate occasions, I was unable to charge the laptop without doing an EC reset. While in operation, it also ddn't show the "plugged in" symbol neither was it charging.
3. On 2 separate occasions, power consumption (reported through `powertop`) went up from ~6-9 watts to 19 watts even though no heavy computation was going on.

lil0r, 2023-11-03:
had the same problems as @dthuerck as well and consequently downgraded to 1.7.0. however ec is still on 2.6.

lil0r, 2023-11-04:
had to revert to EC 2.1 since I had to do an EC reset on every second USB-C charge.
1.14.0 was not really usable due to the missed key strikes - actually attributed it to the flimsy keyboard until I saw this thread.
Is there any hope that eventually coreboot will be available and fix these?

Sean-StarLabs, 2023-11-04:
coreboot available, yes but it won't fix anything - the bootloader is a tiny slither of the BIOS and unrelated to anything above or the other issue

dthuerck, 2023-11-07:
@lil0r since I have the same thing - essentially have to do an EC reset before every charge - could you write down the instructions for reverting EC back to 2.1 here? All I can find is an `ec.bin` file in the repo. Thank you :)

Sean-StarLabs, 2023-11-07:
> This will be a specific solution for this firmware, so please don't try it in the future.
>
> If you download these two: https://github.com/StarLabsLtd/firmware/raw/master/StarBook/MkVI-AMD/misc/ec.bin [...]

They are here, but just pushed 1.16 - it'll likely need the defaults loading in the BIOS and/or EC reset - still testing all the various kernels, but USB-Cs happy.

lil0r, 2023-11-07:
@dthuerck - did the procedure that sean was discribing.
@Sean-StarLabs what is the recommended bios/ec combination & version? I am slightly confused which version has which quirks and which to choose? Also you mentioned coreboot is available for AMD, where could I find that? Didnt see it on LVFS.

Sean-StarLabs, 2023-11-08:
> what is the recommended bios/ec combination & version? I
Either 1.7.0 and 2.00
or
1.16.0 and 2.07

> ? Also you mentioned coreboot is available for AMD, where could I find that? 

No I didn't?

[27 more replies]

commit c8d2d0bf (2023-12-18): Added StarBook MkVI-AMD ami 1.16.0

>     Update PI to 1.0.0.fa\n>     Call ALIB via a Q Event Dont notify AFN4 of state changes\n>     Disable Force Mirror Flag\n

Fixes #126
Fixes #75
closed 2023-12-18 by Sean-StarLabs