StarLabsLtd/firmware #128: [StarBook MkVI - AMD] firmware update to ami 1.10.0 bricked my machine

lil0r, 2023-09-29:


Sean-StarLabs, 2023-10-02:
They're very similar to the LabTop Mk IV in terms of layout - so you can follow https://support.starlabs.systems/kb/guides/installing-slash-recovering-firmware-with-external-programmer

I'd recommend recovering with 1.7.0, which is the latest stable version

lil0r, 2023-10-03:
Thanks Sean,
are there any other ways to recover?
I ordered a ch341a, can you specify the 1.8v converter ?

thanks

Sean-StarLabs, 2023-10-05:
You can use a 12-pin FFC, but with AMD boards, the clip is more straightforward. 

Chip is a W25R128JWSIQ; flashrom 1.2 and higher supports them.

lil0r, 2023-10-05:
Thanks Sean,

The most recent bios would be https://github.com/StarLabsLtd/firmware/raw/master/StarBook/MkVI-AMD/ami/1.10.0/efi-B6-A.zip ?
I am still surprised how I ended up in this situation, the last 30 years I have never ever had a bad flash.

lil0r, 2023-10-06:
Thanks Sean, what is the 1.10.0 for? Is it not the right version to flash?
EFI flash did recognise & successfully flash the platform.

Sean-StarLabs, 2023-10-09:
1.10.0 is in the embarge remote (not our test, basically "alpha").
1.08.0 is testing

How do you get to https://support.starlabs.systems/kb/firmware/efi-shell?

lil0r, 2023-10-09:
recovered via ch341a + 1.8v mod to 1.7.0.
md5sumed the eeprom dump and the vanilla 1.10.0.rom file and they differ. Not sure what happend there.
How can I find out which version is the current stable/beta/alpha version?
Since I have a way to recover now, happy to also be testing the highly anticipated AMD coreboot version…

Sean-StarLabs, 2023-10-09:
Awesome.

It's the links under Releases - https://support.starlabs.systems/kb/firmware/starbook-mk-vi-amd

1.12.0 the latest, currently in testing remote