StarLabsLtd/firmware #129: How to make fwupd to work on arch linux?

Barba732, 2023-10-02:
I have the fwupd 1.9.5 and flashrom 1.2 installed. Do I need to have those manjaro packages of starlabs?

Sean-StarLabs, 2023-10-02:
Shouldn't make too much difference when you get it from , you just need something newer than 1.2 - https://support.starlabs.systems/kb/firmware/lvfs-requirements

Barba732, 2023-10-02:
I see, I asked because I wanted to test if switching branch works, not working:
`# fwupdmgr switch-branch
no devices
`

Sean-StarLabs, 2023-10-02:
Which model and which way?

Barba732, 2023-10-02:
lite mkIII, and ami>coreboot. I know it's beta yet, just want to make sure it works on arch when ready to make the switch

Barba732, 2023-10-02:
I have testing enabled, flashrom 1.2-4 and BIOS unlocked

Sean-StarLabs, 2023-10-02:
What does `sudo flashrom -p internal` say?

Barba732, 2023-10-02:
flashrom v1.2 on Linux 6.5.5-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No DMI table found.
WARNING: No chipset found. Flash detection will most likely fail.
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.

Sean-StarLabs, 2023-10-02:
That's too old -  need a newer version

Barba732, 2023-10-02:
not sure why the text got disconfigure like that...what version should I use? that one from aur?

Sean-StarLabs, 2023-10-02:
1.3 would definately work, you can try the one from AUR - I don't know how they work in terms of what commit it'll use, so don't know if it'll work or not

Barba732, 2023-10-02:
Could it be done with the following comand? 
sudo flashrom -p internal -w coreboot.rom -i bios --ifd -N

Barba732, 2023-10-02:
But still:
$ fwupdmgr switch-branch
no devices

Barba732, 2023-10-02:
No, it needed a reboot:
$ fwupdmgr switch-branch
Dispositivo selezionato: Lite (bios)
╔══════════════════════════════════════════════════════════════════════════════╗
║ Passare dal ramo predefinito a coreboot?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release includes the following features:                         ║
║                                                                              ║
║ • Fix the inconsistant memory map in edk2                                    ║
║ • Disable DPST to fix a flicker with recent kernels                          ║
║ • Dont let coreboot try to change EC values                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Le conseguenze del cambio di ramo del firmware sono state comprese? [y|N]:

closed 2023-10-03 by Sean-StarLabs