StarLabsLtd/firmware #136: [StarBook MkVI - Intel] Wonky system power states

bryanvobo, 2023-10-26:
I've had power state problems for a while but am just now getting around to opening this issue about it.

I was previously running `8.95`/`1.18` which had a problem where it would power on/start the laptop any time I plugged in the barrel power connector, regardless of whether the `coreboot` setting for that was `Enabled` or `Disabled`. I upgraded to `8.97`/`1.19` which _did_ fix the plug-in-power-on problem but also introduced a new problem where any time I rebooted or powered off the laptop it wouldn't "take" and I had to manually hold down the power button for several seconds.

I downgraded back to `8.95`/`1.18` because I'd rather deal with plug-in-power-on than manual-intervention-to-reboot-or-poweroff, but instead of only having the plug-in-power-on problem it kept the new problem as well.

Hoping to get rid of both problems, I switched to the AMI firmware (I flashed [1.3.0.bin](https://github.com/StarLabsLtd/firmware/blob/master/StarBook/MkVI-Intel/ami-flashrom/1.3.0/1.3.0.bin) but `F2`'ing into it says `BIOS Version 1.2.0` and `EC Version 01.05`) but whatever `8.97`/`1.19` introduced has mostly persisted. Now the laptop does eventually reboot on its own after a ~15 second delay but powering off still requires that I hold down the power button for several seconds.

I noticed that `8.97`/`1.19` has been removed from [this git repo](https://github.com/StarLabsLtd/firmware/tree/master/StarBook/MkVI-Intel) (though it's still available from [fwupd](https://fwupd.org/lvfs/devices/com.starlabs.B6-I.coreboot)) so I'm guessing that somebody is already aware that it included something bad. How do I unbreak what has become broken?

---

Details on how I upgrade/downgrade the firmware/ITE/EC:

I've never been able to use `fwupdmgr` from my internal-SSD OS (Arch Linux with `flashrom 1.2-4` and `fwupd 1.9.6-1`), it errors with:
```
failed to open device for prepare: failed to read layout from Intel ICH descriptor
```

A [Star Labs support page](https://support.starlabs.systems/kb/firmware/lvfs-requirements) warns that the "disk cannot be encrypted with LUKS", which my disk _is_, so I assumed that may somehow be the reason. I haven't troubleshot that any further because what does work is booting from an Ubuntu installation USB:
```
$ sudo dd if=/path/to/ubuntu-23.04-desktop-amd64.iso of=/dev/sdX bs=1M conv=fsync oflag=sync status=progress
```
installing `flashrom` packages from [Canonical's repo](http://archive.ubuntu.com/ubuntu/pool/main/f/flashrom/), and running:
```
# flashrom -p internal -w <firmware.file> -i bios --ifd -N
```

To update the ITE/EC, I followed the directions from [this Star Labs page](https://support.starlabs.systems/kb/firmware/efi-shell) - unpack the contents of `efi-B6-I.zip` (`1.18.rom`/`1.19.rom`, `ifu.efi`, `startup.nsh`) to the root of another USB drive, boot to `EFI Shell` while it's inserted, and let it countdown to run `startup.nsh`.

I'm not sure if I need to update the firmware and ITC/EC individually but that's what I've been doing. Until I switched over to the AMI firmware, where all I did was flash `1.3.0.bin` without also running an EFI shell script.

divico, 2023-10-26:
> I've never been able to use `fwupdmgr` from my internal-SSD OS (Arch Linux with `flashrom 1.2-4` and `fwupd 1.9.6-1`), it errors with:
> 
> ```
> failed to open device for prepare: failed to read layout from Intel ICH descriptor
> ```

This is because  flashrom in arch  is too old. As explained in the Starlabs documentation you need a more recent version. I'm also on Arch and with this it works

```
pacman -Qs flashrom
local/flashrom-starlabs 1.3.0rc2-1

```
I used the [PKGBUILD from Manjaro](https://gitlab.manjaro.org/packages/community/flashrom-starlabs) to install it

Sean-StarLabs, 2023-10-26:
> A [Star Labs support page](https://support.starlabs.systems/kb/firmware/lvfs-requirements) warns that the "disk cannot be encrypted with LUKS", which my disk is, so I assumed that may somehow be the reason.

That's for standalone EC updates - the Mk VI doesn't use those.

bryanvobo, 2023-10-26:
Thank you for the `flashrom` build script. It will be more convenient to update without booting from a USB stick. Somehow my brain didn't register the lower version even though `flashrom 1.2-4` is right there staring me in the face.

And I'll stop doing the standalone EFI shell updates for EC and only flash the main firmware.

How do you recommend I proceed, to make the laptop behave normally? Should I try flashing [8.95.rom](https://github.com/StarLabsLtd/firmware/blob/master/StarBook/MkVI-Intel/coreboot/8.95/8.95.rom) again? Or an older version? Or wait for a newer version?

commit 4a0a1103 (2023-10-26): Added StarBook MkVI-Intel coreboot 8.99

Fixes #137
Fixes #136
Fixes #135
closed 2023-10-26 by Sean-StarLabs