StarLabsLtd/firmware #138: [StarBook MkVI - AMD] WRT: Invalid buffer destination

rmsthebest, 2023-11-02:
This is not a new error message, it has been printed probably since the laptop was delivered. But since both bluetooth and wifi seems to work it hasn't bothered me too much.

On every boot I see these two messages printed
[   60.914973] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[   60.262639] iwlwifi 0000:02:00.0: WRT: Invalid buffer destination

The output of dmesg isn't always this verbose though so I'm posting it in hopes it has something useful in it,
```
[   59.609765] iwlwifi 0000:02:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x20
[   59.609793] iwlwifi 0000:02:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[   59.609806] iwlwifi 0000:02:00.0: WFPM_AUTH_KEY_0: 0x90
[   59.609821] iwlwifi 0000:02:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[   59.611118] iwlwifi 0000:02:00.0: loaded PNVM version 181407b3
[   59.626406] iwlwifi 0000:02:00.0: Detected RF GF, rfid=0x10d000
[   59.696400] iwlwifi 0000:02:00.0: base HW address: 7c:b5:66:66:84:b0
[   59.713948] iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0
[   60.262639] iwlwifi 0000:02:00.0: WRT: Invalid buffer destination
[   60.418447] iwlwifi 0000:02:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x20
[   60.418478] iwlwifi 0000:02:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[   60.418492] iwlwifi 0000:02:00.0: WFPM_AUTH_KEY_0: 0x90
[   60.418507] iwlwifi 0000:02:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[   60.506695] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506711] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506717] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506722] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506726] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506731] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.506735] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.511058] iwlwifi 0000:02:00.0: Registered PHC clock: iwlwifi-PTP, with index: 0
[   60.576239] iwlwifi 0000:02:00.0: WRT: Invalid buffer destination
[   60.732018] iwlwifi 0000:02:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x20
[   60.732040] iwlwifi 0000:02:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[   60.732059] iwlwifi 0000:02:00.0: WFPM_AUTH_KEY_0: 0x90
[   60.732072] iwlwifi 0000:02:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[   60.820031] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820049] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820055] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820060] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820065] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820069] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.820074] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   60.888956] Bluetooth: hci0: Waiting for firmware download to complete
[   60.888962] Bluetooth: hci0: Firmware loaded in 1490741 usecs
[   60.889061] Bluetooth: hci0: Waiting for device to boot
[   60.914973] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[   60.914973] Bluetooth: hci0: Device booted in 25363 usecs
[   60.915134] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0041-0041.ddc
[   60.921341] Bluetooth: hci0: Applying Intel DDC parameters completed
[   60.929256] Bluetooth: hci0: Firmware timestamp 2023.28 buildtype 1 build 68350
[   60.959703] iwlwifi 0000:02:00.0: Microcode SW error detected. Restarting 0x0.
[   60.959781] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
[   60.959783] iwlwifi 0000:02:00.0: Transport status: 0x0000004B, valid: 6
[   60.959786] iwlwifi 0000:02:00.0: Loaded firmware version: 83.e8f84e98.0 ty-a0-gf-a0-83.ucode
[   60.959789] iwlwifi 0000:02:00.0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL
[   60.959792] iwlwifi 0000:02:00.0: 0x000002F0 | trm_hw_status0
[   60.959794] iwlwifi 0000:02:00.0: 0x00000000 | trm_hw_status1
[   60.959796] iwlwifi 0000:02:00.0: 0x004DC410 | branchlink2
[   60.959798] iwlwifi 0000:02:00.0: 0x000165E4 | interruptlink1
[   60.959800] iwlwifi 0000:02:00.0: 0x000165E4 | interruptlink2
[   60.959802] iwlwifi 0000:02:00.0: 0x00016DBE | data1
[   60.959803] iwlwifi 0000:02:00.0: 0x00000010 | data2
[   60.959805] iwlwifi 0000:02:00.0: 0x00000000 | data3
[   60.959807] iwlwifi 0000:02:00.0: 0x003D6A99 | beacon time
[   60.959809] iwlwifi 0000:02:00.0: 0x00049694 | tsf low
[   60.959811] iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
[   60.959813] iwlwifi 0000:02:00.0: 0x00000000 | time gp1
[   60.959814] iwlwifi 0000:02:00.0: 0x0005D3B8 | time gp2
[   60.959816] iwlwifi 0000:02:00.0: 0x00000001 | uCode revision type
[   60.959818] iwlwifi 0000:02:00.0: 0x00000053 | uCode version major
[   60.959820] iwlwifi 0000:02:00.0: 0xE8F84E98 | uCode version minor
[... 173 lines cut ...]
[   61.740172] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   61.740176] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   61.740179] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.261154] iwlwifi 0000:02:00.0: WRT: Invalid buffer destination
[   65.417973] iwlwifi 0000:02:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x20
[   65.418000] iwlwifi 0000:02:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[   65.418020] iwlwifi 0000:02:00.0: WFPM_AUTH_KEY_0: 0x90
[   65.418034] iwlwifi 0000:02:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[   65.505333] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505350] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505357] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505362] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505366] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505371] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.505375] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)
[   65.654149] wlp2s0: authenticate with 04:f0:21:32:39:ce
[   65.663522] wlp2s0: send auth to 04:f0:21:32:39:ce (try 1/3)
[   65.695094] wlp2s0: authenticated
[   65.697331] wlp2s0: associate with 04:f0:21:32:39:ce (try 1/3)
[   65.698970] wlp2s0: RX AssocResp from 04:f0:21:32:39:ce (capab=0x11 status=0 aid=2)
[   65.705256] wlp2s0: associated
[   67.046360] systemd-journald[544]: /var/log/journal/1bd6b73e9a374a13ab2923b04df7ff5f/user-1000.journal: Journal file uses a different sequence number ID, rotating.
[   67.088537] pipewire[876]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[  241.527626] warning: `Socket Thread' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
```

Sean-StarLabs, 2023-11-02:
I think `quiet` kernel param should stop it from showing

mysfitt, 2023-11-06:
I can confirm that the exact same issue is present on the Intel variant of the MK VI and has been present through all firmware versions. In more recent versions of the kernel (6.5+) it has gotten more noisy. Also, bluetooth audio, eg. to a headset or speaker, stutters badly. Switching to the linux-lts (6.1.x) kernel on arch has resolved the stutter, but the error messages are still present. I have assumed that the bug lies in the firmware for the Intel  AX210 wireless chipset.

closed 2023-11-21 by Sean-StarLabs