StarLabsLtd/firmware #139: [Starbook MkVI - Intel][coreboot] Don't wake-up from suspend on lid close

divico, 2023-11-21:
Closing the lid when the laptop is in suspend will wake it up.

I've observed this on Coreboot `8.99` and ITE `1.21`

This sequence reproduces the issue:
1. With the lid open, put the machine in suspend 
2. blue LED is dimming, machine is  in suspend
3. Closed the lid
4. The machine wakes up (blue LED is on)

Below some more details. Note that the machine is configured to suspend on power key press (works fine). 

`logind` events
```
journalctl -fu systemd-logind
systemd-logind[1035]: Power key pressed short.
systemd-logind[1035]: The system will suspend now!
systemd-logind[1035]: Lid closed.
systemd-logind[1035]: Operation 'sleep' finished.
systemd-logind[1035]: Lid opened.
```
Same sequence running `acpi_liste`
```
acpi_listen
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
button/power PBTN 00000080 00000000
button/power LNXPWRBN:00 00000080 00000005
battery PNP0C0A:00 00000080 00000001
button/lid LID close
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
button/lid LID open
button/right RIGHT 00000080 00000000 K
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001

```

Sean-StarLabs, 2023-11-21:
It can be observed on any release we've ever done - it's how it should be. Why would you not want it to?

Sean-StarLabs, 2023-11-21:
I'm not against switching it off, unless someone complains - but I don't get this:

> I don't want to rely on the lid to put the machine to sleep, because I often do long calculations that I don't want to interrupt when moving the laptop from one meeting room to another. Therefore, I always explicitly put the machine into suspend by pressing the power button.

End result is S3 - why are you saying via the Power Button is different?

divico, 2023-11-21:
Bad English, sorry about that, let me rephrase.

_ I often do long calculations that I don't want to interrupt when I need to carry the laptop with the lid closed for a short time (e.g. switch meeting room). Therefore, I don't want the rely on lid events for entering suspend (S3).  When I need to carry the laptop around for a longer time, I close the lid and put it in a bag. If it was in suspend before, it will wake up when the lid is closed and stay on in the bag._

Could you make this configurable in the coreboot menu?

Sean-StarLabs, 2023-11-21:
Wouldn't that mean you've set the OS to ignore the lid?

divico, 2023-11-21:
> Wouldn't that mean you've set the OS to ignore the lid?

Yes in `/etc/systemd/logind.conf` I have

``` 
HandleLidSwitch=ignore
HandleLidSwitchExternalPower=ignore
HandleLidSwitchDocked=ignore
```
I can close the lid and the machine stays on, as expected. However, once in S3, closing the lid will wake it up.

Sean-StarLabs, 2023-11-21:
Cool - so it's a software problem. What distro + de?

divico, 2023-11-21:
I'm on Arch with `gnome 45`.

Sean-StarLabs, 2023-11-21:
Eesh, probably not going to be much help... but, have you checked the schema values match the ones you just posted?

divico, 2023-11-21:
I understand that this is not the right place if it is a software issue. Could you please elaborate on how you came to the conclusion that is is a software issue?

Why is there no separate entry for the lid in `cat /proc/acpi/wakeup`?

```
Device	S-state	  Status   Sysfs node
RP01	  S3	*enabled   pci:0000:00:1c.0
XHCI	  S3	*enabled   pci:0000:00:14.0
GLAN	  S4	*disabled
```

I was thinking of disabling lid events here.

divico, 2023-11-21:
> For your use case, you said it should be ignoring the lid - which it's not.

Yes it does ignore the lid when booted in OS, but it does not when in S3.

There are no options anymore in gnome 45 to handle lid switch (neither in settings nor in tweaks). I don't think that the option in previous version would handle the case in S3.

I will dig into this but I don't understand how the OS is supposed to inhibit a lid  event while the machine is in S3. Isn't the wake up from S3 happening at a lower level, EC---> CPU or something like this?

Sean-StarLabs, 2023-11-21:
> There are no options anymore in gnome 45 to handle lid switch (neither in settings nor in tweaks). I don't think that the option in previous version would handle the case in S3.

Weird. It does ignore it on 44.3 (Ubuntu).

> how the OS is supposed to inhibit a lid event while the machine is in S3. 

It's not meant to inhibit it (it doesn't have that much control), it's just, in your case, not meant to do anything if the value changes. The "lid" is super basic, there is a variable in stored in ACPI - IIRC, 1 is open and 0 is closed.

divico, 2023-11-21:
I can reproduce the same behavior on  Ubuntu `22.04.3 LTS` booted from a USB disk. Closing the lid while in S3 wakes the machine up. I used gnome tweaks to disable suspend on lid close, which worked in the OS. 
So, it is exactly the same behavior as on my arch install with gnome `45`. You should be able to reproduce.

I will further look into how is this ACPI variable handled when in S3.

Sean-StarLabs, 2023-11-21:
I did test with 22.04 (installed, not live) - just the ac-action and battery-action set to none, right?

divico, 2023-11-22:
> just the ac-action and battery-action set to none, right?

These settings have no effect once it is in S3.

I could reproduce the issue with

```
ubuntu@ubuntu:~$ gsettings list-recursively | grep lid-close
org.gnome.settings-daemon.plugins.power lid-close-ac-action 'suspend'
org.gnome.settings-daemon.plugins.power lid-close-battery-action 'suspend'
org.gnome.settings-daemon.plugins.power lid-close-suspend-with-external-monitor false
```

as well as with

```
ubuntu@ubuntu:~$ gsettings list-recursively | grep lid-close
org.gnome.settings-daemon.plugins.power lid-close-ac-action 'nothing'
org.gnome.settings-daemon.plugins.power lid-close-battery-action 'nothing'
org.gnome.settings-daemon.plugins.power lid-close-suspend-with-external-monitor false
```

Same for the tweak to suspend when laptop is closed, it has no effect when in S3.
[image]

Below movie that shows the issue. This is Ubuntu `22.04.3 LTS` live.

https://github.com/StarLabsLtd/firmware/assets/11789268/1d3a7b28-6850-430c-a19b-73d2e9add8d5

Sean-StarLabs, 2023-11-22:
Have you tried with `sudo systemctl suspend`?

divico, 2023-11-22:
> Have you tried with `sudo systemctl suspend`?

yes same issue

divico, 2023-11-23:
One thing I don't get: how is it actually supposed to be?

Because you said
> It can be observed on any release we've ever done - it's how it should be. Why would you not want it to?

but later

> Weird. It does ignore it on 44.3 (Ubuntu).

In addition, I tested this on another laptop (thinkpad x230) and it behaves as I expect: when in S3, closing the lid does not wake up.  

So is it about fixing a bug or inhibiting a feature?

closed 2023-12-07 by Sean-StarLabs

tsmithe, 2023-12-10:
Hi @Sean-StarLabs, I've been experiencing this on the AMD variant (or I must be hallucinating?); is it / will it be closed in an upcoming firmware there?

Sean-StarLabs, 2023-12-11:
Yup - they all use the same EC so whatever the load behaviour ends up being, it will be the same for all of them

divico, 2024-01-12:
Finally I could test the fix in `9.01` and it is indeed solved. Thanks!

divico, 2024-02-26:
Today I updated to the stable `24.02` coreboot firmware and the issue is back even though the release note says

```Only wake from sleep when the lid is closed with an active PD connection.```

I tried with and without PD connection.

Sean-StarLabs, 2024-02-26:
Interesting, as coreboot doesn't have anything to do with it.

What's the EC version?

divico, 2024-02-26:
Still `1.21`

```
cat /sys/class/dmi/id/ec_firmware_release
1.21
```
but I did try `1.23` for some time in between, in case this is be relevant

Sean-StarLabs, 2024-02-26:
It might be worth checking what's changed on your distro, as the lid code in coreboot hasn't changed for quite a while: https://github.com/StarLabsLtd/coreboot/commits/starlabs-4.23/src/ec/starlabs/merlin/acpi/lid.asl

But, I'd recommend using 24.02 to match coreboot. Letting it latch should be enough to update it, just shutdown, disconnect the charger and wait for 12 seconds. Connect the charger and it *might* automatically go up to 24.02 (long list of requirements...)

divico, 2024-02-27:
So I have now also managed to get EC `24.02` and the issue is gone. Thanks for your help.

One more question...How can I figure out whether a coreboot firmware release comes with a new ite or not? Is this for stable releases always the case? Because from the release note it is not clear to me that `24.02`  was also supposed to update the ite, that is why I thought `1.21` is the lastest release

Sean-StarLabs, 2024-02-27:
No easy way to tell - but they'll all have them going forward so the versions match

commit 8032b0c6 (2023-12-07): Added StarBook MkVI-Intel coreboot 9.00

>     Rebased on coreboot 4.22\n>     Rebased on edk2/master\n>     Enabled PMC muxing\n>     Remove the option to turn on when connecting a charger\n>     Make PECI IRQ driven\n>     Empty the keyboard buffer faster\n>     Switch the compiler from Keil to SDCC\n>     Automatically power on after FMF\n>     Fixed an issue that stopped the scheduler running\n>     Reduced false over-current protection warnings\n>     Only wakeup when closing the lid with CC1 and CC2\n>     \n>     Fixes #144\n>     Fixes #143\n>     Fixes #140\n>     Fixes #139\n