StarLabsLtd/firmware #140: USB  over-current condition, S3 suspend fails - Starbook VI Intel

mysfitt, 2023-11-25:
Overall, Suspend has been working pretty reliably in recent firmware versions, but I occasionally see this in my journal upon a failure:

> Nov 25 01:31:55 kernel: usb usb1-port6: over-current condition
> Nov 25 01:31:55 kernel: xhci_hcd 0000:00:14.0: USB core suspending port 1-6 not in U0/U1/U2
> Nov 25 01:31:55 kernel: usb 1-6: Failed to suspend device, error -32
> Nov 25 01:31:55 kernel: usb usb1-port7: over-current condition
> Nov 25 01:31:56 kernel: usb usb1-port6: over-current condition

I believe that is occurs when connected to power, but I haven't isolated that to a 100% correlation. In this particular example, I was definitely connected to mains when initiating suspend. I have also seen the over-current condition on USB ports seemingly randomly at other times when I am not initiating S3. I have only noticed this error in 8.97 and 8.99 of coreboot. My EC version is 1.21. It's interesting to note that there are no external USB devices connected when I see this, so I'm guessing the usb1 bus is the internal one for the camera, fingerprint reader, etc. 

Is there any risk of damage to the hardware, or is this merely an annoyance?

Sean-StarLabs, 2023-12-01:
There's nothing to worry about but I'll doubt check it, see if we can quieten it down. Which StarBook?

mysfitt, 2023-12-04:
It's the 1.0 StarBook with the 12th gen Intel I7 1260p processor. Thanks for checking into it!

commit 8032b0c6 (2023-12-07): Added StarBook MkVI-Intel coreboot 9.00

>     Rebased on coreboot 4.22\n>     Rebased on edk2/master\n>     Enabled PMC muxing\n>     Remove the option to turn on when connecting a charger\n>     Make PECI IRQ driven\n>     Empty the keyboard buffer faster\n>     Switch the compiler from Keil to SDCC\n>     Automatically power on after FMF\n>     Fixed an issue that stopped the scheduler running\n>     Reduced false over-current protection warnings\n>     Only wakeup when closing the lid with CC1 and CC2\n>     \n>     Fixes #144\n>     Fixes #143\n>     Fixes #140\n>     Fixes #139\n