StarLabsLtd/firmware #143: [StarBook mkVI Intel coreboot] Missing CPPC support breaks Linux CPU scheduling

R-VdP, 2023-11-29:
I noticed that since Linux 6.5, performance of my starbook mkVI is pretty bad. Linux seems to prefer E cores over P cores most of the time.

I filed a [bug report](https://bugzilla.kernel.org/show_bug.cgi?id=218195) on the kernel bug tracker and the conclusion seems to be that there is an issue with the CPPC/OSC support in the firmware, see the bug report for more details.

This is killing performance of my i7 processor.

 I am running the coreboot version 8.60 firmware (since there are some open issues for 8.99 still).

Sean-StarLabs, 2023-11-29:
The base `8.60` used didn't support CPPC, latest does.

> since there are some open issues for 8.99 still

Not that I can see

R-VdP, 2023-11-29:
I upgraded to 8.99 but the kernel still does not manage to get CPC info from the CPU:

```
$ dmesg | rg -i cppc
[    0.000000] Command line: initrd=\efi\nixos\7pnavbs3q41mr5i33sjaifz2y2zyv9bm-initrd-linux-6.6.3-initrd.efi init=/nix/store/m7y6rpq5p69pxkh7276ddd19xsf75q77-nixos-system-starbook-20231129.dirty/init video=DP-1:3840x2160@60 iomem=relaxed i915.enable_fbc=1 i915.enable_psr=2 mem_sleep_default=deep nvme.noacpi=1 systemd.gpt_auto=false "dyndbg=file cppc_acpi.c +p" loglevel=4
[    0.062752] Kernel command line: initrd=\efi\nixos\7pnavbs3q41mr5i33sjaifz2y2zyv9bm-initrd-linux-6.6.3-initrd.efi init=/nix/store/m7y6rpq5p69pxkh7276ddd19xsf75q77-nixos-system-starbook-20231129.dirty/init video=DP-1:3840x2160@60 iomem=relaxed i915.enable_fbc=1 i915.enable_psr=2 mem_sleep_default=deep nvme.noacpi=1 systemd.gpt_auto=false "dyndbg=file cppc_acpi.c +p" loglevel=4
[    0.372136] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372138] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372195] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372196] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372239] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372240] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372281] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372283] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372322] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372323] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372360] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372361] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372399] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372400] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372442] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372443] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372482] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372483] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372523] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372524] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372562] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372563] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372602] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372603] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372643] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372644] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372681] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372682] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372722] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372723] ACPI CPPC: CPPC v2 _OSC not acked
[    0.372762] ACPI CPPC: begin acpi_cppc_processor_probe
[    0.372763] ACPI CPPC: CPPC v2 _OSC not acked
[    0.380999] ACPI CPPC: No CPC descriptor for CPU:0
[    0.381065] ACPI CPPC: No CPC descriptor for CPU:1
[    0.381153] ACPI CPPC: No CPC descriptor for CPU:2
[    0.381214] ACPI CPPC: No CPC descriptor for CPU:3
[    0.381385] ACPI CPPC: No CPC descriptor for CPU:4
[    0.381442] ACPI CPPC: No CPC descriptor for CPU:5
[    0.381607] ACPI CPPC: No CPC descriptor for CPU:6
[    0.381665] ACPI CPPC: No CPC descriptor for CPU:7
[    0.381824] ACPI CPPC: No CPC descriptor for CPU:8
[    0.381936] ACPI CPPC: No CPC descriptor for CPU:9
[    0.382034] ACPI CPPC: No CPC descriptor for CPU:10
[    0.382153] ACPI CPPC: No CPC descriptor for CPU:11
[    0.382253] ACPI CPPC: No CPC descriptor for CPU:12
[    0.382338] ACPI CPPC: No CPC descriptor for CPU:13
[    0.382392] ACPI CPPC: No CPC descriptor for CPU:14
[    0.382468] ACPI CPPC: No CPC descriptor for CPU:15
```

Sean-StarLabs, 2023-11-29:
Pass on what that means - if you check the SSDT, it's there. Maybe kernel build option/param or something distro related?

R-VdP, 2023-11-29:
Well, in the [kernel bug](https://bugzilla.kernel.org/show_bug.cgi?id=218195), they mention that they require OSC support for CPPC v2, which should be provided by the firmware but apparently isn't in this case.

Do you see `/proc/sys/kernel/sched_itmt_enabled` on any system on your side with the i7 1260P and coreboot?

Sean-StarLabs, 2023-11-29:
Sure do:
```
sean@starbook:~/Desktop/acpi$ grep -ri cpc
grep: ssdt.dat: binary file matches
ssdt.dsl:        Name (GCPC, Package (0x17)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            GCPC [0x03] = 0x21
ssdt.dsl:            GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (GCPC) /* \_SB_.CP00.GCPC */
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x21
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
ssdt.dsl:        Method (_CPC, 0, NotSerialized)  // _CPC: Continuous Performance Control
ssdt.dsl:            \_SB.CP00.GCPC [0x03] = 0x1A
ssdt.dsl:            \_SB.CP00.GCPC [0x16] = 0x0A28
ssdt.dsl:            Return (\_SB.CP00.GCPC)
sean@starbook:~/Desktop/acpi$ cat /proc/sys/kernel/sched_itmt_enabled
[... 123 lines cut ...]
/sys/devices/system/cpu/cpu7/acpi_cppc/nominal_freq:2600
/sys/devices/system/cpu/cpu7/acpi_cppc/lowest_freq:0
/sys/devices/system/cpu/cpu7/acpi_cppc/nominal_perf:33
/sys/devices/system/cpu/cpu7/acpi_cppc/lowest_perf:1
/sys/devices/system/cpu/cpu7/acpi_cppc/reference_perf:33
/sys/devices/system/cpu/cpu7/acpi_cppc/wraparound_time:18446744073709551615
/sys/devices/system/cpu/cpu8/acpi_cppc/feedback_ctrs:ref:45785259936 del:22089880138
/sys/devices/system/cpu/cpu8/acpi_cppc/lowest_nonlinear_perf:13
/sys/devices/system/cpu/cpu8/acpi_cppc/highest_perf:37
/sys/devices/system/cpu/cpu8/acpi_cppc/nominal_freq:2600
/sys/devices/system/cpu/cpu8/acpi_cppc/lowest_freq:0
/sys/devices/system/cpu/cpu8/acpi_cppc/nominal_perf:26
/sys/devices/system/cpu/cpu8/acpi_cppc/lowest_perf:1
/sys/devices/system/cpu/cpu8/acpi_cppc/reference_perf:26
/sys/devices/system/cpu/cpu8/acpi_cppc/wraparound_time:18446744073709551615
/sys/devices/system/cpu/cpu9/acpi_cppc/feedback_ctrs:ref:30707020448 del:15237582281
/sys/devices/system/cpu/cpu9/acpi_cppc/lowest_nonlinear_perf:14
/sys/devices/system/cpu/cpu9/acpi_cppc/highest_perf:37
/sys/devices/system/cpu/cpu9/acpi_cppc/nominal_freq:2600
/sys/devices/system/cpu/cpu9/acpi_cppc/lowest_freq:0
/sys/devices/system/cpu/cpu9/acpi_cppc/nominal_perf:26
/sys/devices/system/cpu/cpu9/acpi_cppc/lowest_perf:1
/sys/devices/system/cpu/cpu9/acpi_cppc/reference_perf:26
/sys/devices/system/cpu/cpu9/acpi_cppc/wraparound_time:18446744073709551615
```

R-VdP, 2023-11-29:
Hmm, could you share the kernel config from this system so that I can compare it? Which distro is this?
And this is on coreboot 8.99?

Sean-StarLabs, 2023-11-29:
It's just Ubuntu - zero changes:
[kernel.txt](https://github.com/StarLabsLtd/firmware/files/13503783/kernel.txt)

Yeah, just checked, 8.60 has it too - though the acpigen was meant to be broken but apparently not

R-VdP, 2023-11-29:
I just booted a Ubuntu 23.05 live iso and had the same issue there. No itmt scheduling and no cppc.

Is there any chance that there's something to with the firmware on my specific model (mkVI, i7 1260P, coreboot) ?

What could we do to rule out an issue with the firmware?

It's really very annoying to not have proper scheduling on this processor.

divico, 2023-11-30:
> Do you see `/proc/sys/kernel/sched_itmt_enabled` on any system on your side with the i7 1260P and coreboot?

Not sure if this is still relevant to you, but I don't have that file neither on arch kernel `6.1.63-1-lts` and coreboot `8.99` (same CPU).

R-VdP, 2023-12-01:
@Sean-StarLabs did you manage to test from your side on the same system? Did you get the same results?

The fact that I'm not the only one with this issue, seems to indicate that there is something going wrong here.

I've got this issue on NixOS and had the same problem on a Ubuntu live ISO, and the other reporter has the same issue on Arch.

Sean-StarLabs, 2023-12-01:
> @Sean-StarLabs did you manage to test from your side on the same system? Did you get the same results?

Not yet

R-VdP, 2023-12-07:
Thanks @Sean-StarLabs, that seems to fix it indeed!

I had to update the EC firmware separately using the EFI shell though, it was still at 1.13. Is it not supposed to be automatically updated with the mirror thingy?

Sean-StarLabs, 2023-12-07:
It'll only do it if the requirements are met - it's a long list, but from 1.22 onwards, it'll be a lot shorter.

R-VdP, 2023-12-07:
Oh I see, ok, cool. Thanks again for the fix for the scheduling!

divico, 2023-12-09:
> > @r-vdp  how did you install `9.00`? It's not in lvfs-testing
> 
> I downloaded the .rom from this repo, and flashed it using flashrom:
> 
> ```
> sudo flashrom -p internal -w 9.00.rom -i bios --ifd -N
> ```

Thanks. I'm not sure this is safe / recommended. 
@Sean-StarLabs will `9.00` be available on lvfs? If not, is it fine to use  flashrom directly as above?

Sean-StarLabs, 2023-12-09:
It'll be in the testing remote early next week.

Flashrom can brick it (one in 510ish chance)

divico, 2023-12-16:
> It'll be in the testing remote early next week.
Could you please make `9.00`available in lvfs

commit 8032b0c6 (2023-12-07): Added StarBook MkVI-Intel coreboot 9.00

>     Rebased on coreboot 4.22\n>     Rebased on edk2/master\n>     Enabled PMC muxing\n>     Remove the option to turn on when connecting a charger\n>     Make PECI IRQ driven\n>     Empty the keyboard buffer faster\n>     Switch the compiler from Keil to SDCC\n>     Automatically power on after FMF\n>     Fixed an issue that stopped the scheduler running\n>     Reduced false over-current protection warnings\n>     Only wakeup when closing the lid with CC1 and CC2\n>     \n>     Fixes #144\n>     Fixes #143\n>     Fixes #140\n>     Fixes #139\n