StarLabsLtd/firmware #145: [StarBook Mk VI - AMD] Small EC Bugs

Sean-StarLabs, 2023-12-18:


dthuerck, 2023-12-27:
Thanks for the new firmware - it's a much needed improvement over the last versions/attempts (although I'm sad to say I've moved to a Framework 13 as a daily driver due to the stream of problems with the Starbook AMD). 

After testing it for a few days (Flash 1.16.0, do EC reset, install fresh Fedora 39), two issues remain:
- The number of missed keystrokes (*on battery only*) is dramatically reduced since the previous versions. However, it's still there. I had three different people test it just in case the problem is me ;)
- Loud "CLICK" noise when I shut down the machine *on battery power only*.

Interestingly, everything is fine while connected to AC!

Sean-StarLabs, 2023-12-27:
> The number of missed keystrokes (on battery only) is dramatically reduced since the previous versions. However, it's still there. I had three different people test it just in case the problem is me ;)

You're running one of the testing ECs

> Loud "CLICK" noise when I shut down the machine on battery power only.

It's not a bug; "design choice" is probably the phrase I'd use. It shuts off the amp when it goes from S5 to G3.

dthuerck, 2023-12-28:
> > The number of missed keystrokes (on battery only) is dramatically reduced since the previous versions. However, it's still there. I had three different people test it just in case the problem is me ;)
> 
> You're running one of the testing ECs
> 

Sorry, then I must have misunderstood - I thought that the newest 1.16.0 release includes a fixed EC. Or is it that issue here which will conclude in a new EC? ;)

> > Loud "CLICK" noise when I shut down the machine on battery power only.
> 
> It's not a bug; "design choice" is probably the phrase I'd use. It shuts off the amp when it goes from S5 to G3.

Thanks got clearing that up! It's not a problem per se of course, I was just afraid that something is going wrong.

lil0r, 2023-12-31:
Another feature has popped up that seemed related:
**Suffering from charging issue and powerdrain during deep sleep.**
BIOS 1.16.0
EC 2.1

Battery drains about 1% per hour while in deep sleep.

Computer is configured for deep sleep (RAM only) and I do acknowledge I have 2x32GB (factory) DIMMS, however, still way to much. I loose about 50% charge in 2.5 days. 

Other observation when plugged in at 100%, (barrel or usb-c) state alternates between charging and full.

tried a couple of things:
- EC reset and load bios defaults
- different ubuntu & custom kernels

tsmithe, 2024-01-02:
Another one: when suspending while on USB power (`systemctl suspend`), the system (sometimes?) seems to resume itself after less than a second.  According to `dmesg`, it does actually reach suspend (at least from the perspective of the Linux kernel), but then immediately wakes up again:
```
[31363.132441] smpboot: CPU 15 is now offline
[31363.133647] ACPI: PM: Low-level resume complete
```
Is this because I closed the lid?  If so, I'd really like it to be possible to suspend the machine, close the lid, and have it *not* wake up!

Sean-StarLabs, 2024-01-03:
> Sorry, then I must have misunderstood - I thought that the newest 1.16.0 release includes a fixed EC. Or is it that issue here which will conclude in a new EC? ;)

1.16.0 doesn't have a forced EC, so it won't change what you have installed. The only things I've got on my list to change in the EC are:
* fix the keyboard backlight after S3
* increasing the debounce of the chargers

To put the EC back to stable, you can either flash it manually, or use ectool from coreboot. If you want me to upload the EC on it's own, just let me know but the EC tool command is:
```
sudo ectool -w 05 -z aa
```
Then shutdown with the charger connected (not reboot) and it'll turn back on running the stable EC.

lil0r, 2024-01-03:
Thanks Sean,

any pointers on the powerdrain?

I am currently playing around with pm-suspend and various quirks settings.

Syslog only shows noticeable events after waking up
Wifi wakes up and triggers a hardware reset, but is working fine - very rarely bluetooth does not come up
[  174.215694] iwlwifi 0000:02:00.0: Microcode SW error detected. Restarting 0x0.
[  174.215801] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:

ACPI error I cannot decode, but also points toward wireless
[  174.955773] ACPI: \: failed to evaluate _DSM bf0212f2-788f-c64d-a5b3-1f738e285ade (0x1001)

Sean-StarLabs, 2024-01-03:
> any pointers on the powerdrain?

Is it present with the stable EC?

lil0r, 2024-01-03:
> > any pointers on the powerdrain?
> 
> Is it present with the stable EC?

Which one is considered stable? Sorry bit confused by now

Sean-StarLabs, 2024-01-03:
> 1.16.0 doesn't have a forced EC, so it won't change what you have installed. The only things I've got on my list to change in the EC are:
>
> * fix the keyboard backlight after S3
> * increasing the [...]

This ^^

tsmithe, 2024-01-03:
I have `2.6` in `/sys/class/dmi/id/ec_firmware_release` -- is that the stable one?

lil0r, 2024-01-03:
> This ^^

ok, interesting... 

`cat /sys/class/dmi/id/ec_firmware_release 
1.7
`

Do I need to perform an EC reset?

Sean-StarLabs, 2024-01-03:
> I have 2.6 in /sys/class/dmi/id/ec_firmware_release -- is that the stable one?

No

> Do I need to perform an EC reset?

After the ectool command, no

lil0r, 2024-01-04:
> > any pointers on the powerdrain?
> 
> Is it present with the stable EC?

seems to be also present with EC 1.7. I never paid attention before.
Anyone else experience a 1% battery charge reduction per hour while in S3/suspend to RAM?

cdelledonne, 2024-01-04:
>  Anyone else experience a 1% battery charge reduction per hour while in S3/suspend to RAM?

I do! I'm running firmware 1.16.0 and EC 2.0.

lil0r, 2024-01-06:
> ok, interesting...
>
> `cat /sys/class/dmi/id/ec_firmware_release 1.7 `
>
> Do I need to perform an EC reset?

Sean - It is a bit hit & miss. I had to revert to EC 2.6 since 1.7 kept throttling my CPU to 400mhz hard. I think around 85°C
I got stuttering in my system under heavy load and an annoying spin up and down of the fan.
2.6 lets my CPU go to 100°C with  constant fan and about 3Ghz.

lil0r, 2024-01-06:
@Sean-StarLabs. How do I get to EC 2.7? I am somehow stuck on EC 2.6 and USB-PD does not work on that version.
flashing 1.16.0 package I am still on EC2.6.

Sean-StarLabs, 2024-01-11:
Pushed 2.07 to the misc folder

lil0r, 2024-01-12:
> Pushed 2.07 to the misc folder

cheers, happy to test. What is the recommended install & reset procedure?
What has been updated?

Sean-StarLabs, 2024-01-12:
> What has been updated?

Absolutely nothing - you asked for it?

lil0r, 2024-01-12:
> > What has been updated?
> 
> Absolutely nothing - you asked for it?

Ah thanks, understand - thought there were changes made to it. I am staying on 2.6 and wait for the next release... thanks

tsmithe, 2024-01-14:
I let my StarBook run out of charge last night and this morning USB PD wouldn't work, so I had to dig out my barrel cable.  I'm on BIOS 1.16 and EC 2.6.  Is this still expected?  (I thought the USB PD empty-battery issues had been fixed.)

Sean-StarLabs, 2024-01-15:
Yes, new BIOS isn't going to fix a bug in the EC. Downgrading to a stable one would be the best bet.

Sean-StarLabs, 2024-01-15:
> 1.16.0 doesn't have a forced EC, so it won't change what you have installed. The only things I've got on my list to change in the EC are:
>
> * fix the keyboard backlight after S3
> * increasing the [...]

Either these to go to 1.7.0 or you can grab 2.00 from [here](https://github.com/StarLabsLtd/firmware/raw/master/StarBook/MkVI-AMD/misc/2.00.bin) and use `ifu.efi 2.00.bin 128`

tsmithe, 2024-01-15:
Ah, so the ectool command writes some magic which tells the EC to update itself from the BIOS (which is 1.7 for me)? Whereas using ifu (from the EFI shell?) will write the given binary to the EC?

Sean-StarLabs, 2024-01-15:
Yup

tsmithe, 2024-01-24:
I have noticed, in `powertop`, a "USB Storage" device that seems to be constantly drawing 4W of power.  I don't have anything plugged in, so I think it is the device "05e3:0761 Genesys Logic, Inc. Genesys Mass Storage Device" that appears to be part of the StarBook platform and that seems to show up as `/dev/sda` on my system (but which reports "no medium" if I try to inspect its partition table..).  I'm not sure what's going on, but I'd like to save those 4W if possible!

tsmithe, 2024-01-24:
Oh, right, thanks!

Sean-StarLabs, 2024-01-24:
I'll add two things to my list:
* Add an option to disable the card reader (I never use mine)
* Disable USB power in S3

lil0r, 2024-01-27:
Hi Sean, is there any update on the fixed EC? Had to do an EC reset twice this week to get USB-C PD working again. Also I seem to have misplaced my barrel charger, hence I am samewhat dependent ;)

Sean-StarLabs, 2024-01-27:
The power sequence is just getting SDCC treatment. If you're dependent, maybe a stable EC is that way to go?

lil0r, 2024-02-04:
> The power sequence is just getting SDCC treatment. If you're dependent, maybe a stable EC is that way to go?

Sean, as mentioned 1.7.0 turned out to be a hog on performance. EC2.6 has much better thermal performance but other issues.... Folks with AMD seem to be currently trapped between a rock and a hard place

Sean-StarLabs, 2024-02-04:
> 1.7.0 turned out to be a hog on performance. EC2.6 has much better thermal performance but other issues.... 

Why are you only considering those?

> Folks with AMD seem to be currently trapped between a rock and a hard place

Nope, most are happily running the latest stable versions - only known bug there is the KBLT after S3.

tsmithe, 2024-02-07:
On EC 1.7 / BIOS 1.16, I still experience missed keypresses (even plugged in, seemingly particularly noticeable after resuming from suspend).  I'm quite careful to make each keypress distinct, but it's still quite annoying;  I don't think I type particularly fast!

Is this what you mean by "KBLT after S3"?  If so, I'm looking forward to the fix!

tsmithe, 2024-02-07:
Oh, anything to be done about the dropped keypresses?  I don't think it's just psychosomatic ...

Sean-StarLabs, 2024-02-07:
> Oh, anything to be done about the dropped keypresses? I don't think it's just psychosomatic ...

Use 2.00

lil0r, 2024-02-10:
> Why are you only considering those?
>
>
> Nope, most are happily running the latest stable versions - only known bug there is the KBLT after S3.

well I am on 2.6 I still experience:
- USB PD not charging, about once a week. fixed by EC reset.
- KBLT not working (minor)
- machine not resuming/display not turning on, fixed by turning off & on

Which version should I be using?

tsmithe, 2024-02-14:
On EC 1.7 (& BIOS 1.16), I have now also encountered USB PD stopping working; it is the case just now.  I have not been able to deduce the conditions under which this still happens.

Sean-StarLabs, 2024-02-14:
> I have not been able to deduce the conditions under which this still happens.

Don't worry about it

> Which version should I be using?

2.00 is the latest stable

tsmithe, 2024-03-25:
Hi again,  I wonder if there will be a stable firmware release soon, with the bugs listed above fixed?  (Or at least the EC 2.00 bundled, rather than 1.7?)

Sean-StarLabs, 2024-03-25:
Yeah, couple of days  - the version number is just being annoying!

commit 2dc10430 (2024-03-26): Added StarBook MkVI-AMD ami 24.03.00

>     Fixed an issue that stopped the backlight working after S3\n>     Enabled support for PD 2.0 chargers\n>     Disable USB power in S3\n>     Improved battery reporting\n

Fixes #145
closed 2024-03-26 by Sean-StarLabs