StarLabsLtd/firmware #147: [StarBook Mk V] fwupd fails to connect / activate

sees7ern, 2024-01-06:
**Describe the bug**
Attempting to update StarBook Mk V coreboot via fwupd. Worked previously and still works perfectly on my StarLite Mk IV, having followed the information under https://starlabsltd.github.io/firmware/methods/. Both systems run Elementary OS 7.1 with Coreboot.

> fwupd is already the newest version (1.8.9+j).

> libflashrom1 is already the newest version (1.2-10+j).

> Hit:10 https://ppa.launchpadcontent.net/starlabs/coreboot/ubuntu jammy InRelease          

> Hit:11 https://ppa.launchpadcontent.net/starlabs/main/ubuntu jammy InRelease

Now any command involving fwupdmgr does no longer work on my StarBook Mk V. Each time running into the following error, including after re-installing fwupd:
> Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

also following error persists on my StarBook Mk V:
> systemctl status fwupd.service
× fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: signal) since Sat 2024-01-06 13:42:41 GMT; 1min 11s ago
       Docs: https://fwupd.org/
    Process: 20585 ExecStart=/usr/libexec/fwupd/fwupd (code=killed, signal=ABRT)
   Main PID: 20585 (code=killed, signal=ABRT)
        CPU: 101ms

Jan 06 13:42:41 ZZSL systemd[1]: Starting Firmware update daemon...
Jan 06 13:42:41 ZZSL fwupd[20585]: **
Jan 06 13:42:41 ZZSL fwupd[20585]: ERROR:../src/xb-lzma-decompressor.c:130:xb_lzma_decompressor_convert: code should not be reached
Jan 06 13:42:41 ZZSL fwupd[20585]: Bail out! ERROR:../src/xb-lzma-decompressor.c:130:xb_lzma_decompressor_convert: code should not be reached
Jan 06 13:42:41 ZZSL systemd[1]: fwupd.service: Main process exited, code=killed, status=6/ABRT
Jan 06 13:42:41 ZZSL systemd[1]: fwupd.service: Failed with result 'signal'.
Jan 06 13:42:41 ZZSL systemd[1]: Failed to start Firmware update daemon.

> journalctl -xeu fwupd.service
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit fwupd.service has entered the 'failed' state with result 'signal'.
Jan 06 13:42:06 ZZSL systemd[1]: Failed to start Firmware update daemon.
░░ Subject: A start job for unit fwupd.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit fwupd.service has finished with a failure.
░░ 
░░ The job identifier is 8078 and the job result is failed.
Jan 06 13:42:41 ZZSL systemd[1]: Starting Firmware update daemon...
░░ Subject: A start job for unit fwupd.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit fwupd.service has begun execution.
░░ 
░░ The job identifier is 8182.
Jan 06 13:42:41 ZZSL fwupd[20585]: **
Jan 06 13:42:41 ZZSL fwupd[20585]: ERROR:../src/xb-lzma-decompressor.c:130:xb_lzma_decompressor_convert: code should not be reached
Jan 06 13:42:41 ZZSL fwupd[20585]: Bail out! ERROR:../src/xb-lzma-decompressor.c:130:xb_lzma_decompressor_convert: code should not be reached
Jan 06 13:42:41 ZZSL systemd[1]: fwupd.service: Main process exited, code=killed, status=6/ABRT
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ An ExecStart= process belonging to unit fwupd.service has exited.
░░ 
░░ The process' exit code is 'killed' and its exit status is 6.
Jan 06 13:42:41 ZZSL systemd[1]: fwupd.service: Failed with result 'signal'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit fwupd.service has entered the 'failed' state with result 'signal'.
Jan 06 13:42:41 ZZSL systemd[1]: Failed to start Firmware update daemon.
░░ Subject: A start job for unit fwupd.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit fwupd.service has finished with a failure.
░░ 
░░ The job identifier is 8182 and the job result is failed.

**Steps to Reproduce**
Followed all steps outlined under https://starlabsltd.github.io/firmware/methods/ to set up fwupd. 

> fwupdmgr refresh

> fwupdmgr update

> systemctl status fwupd.service

> journalctl -xeu fwupd.service

Errors mentioned above occur

**Expected behavior**
Update process commences.

**Other**
Have tried to find a solution online, including the StarLabs sites, but have not found any solution. Would appreciate any help. Thank you!

Sean-StarLabs, 2024-01-06:
Do the stock packages work?

sees7ern, 2024-01-07:
Sorry, what do you mean with stock packages int his case? referring to fwupd?

jozic, 2024-01-07:
Linux Mint 21.2, StarBook Mk VI
things used to work but now it fails with similar issue
```
➜  ~ fwupdmgr --version -v
(fwupdmgr:2889): GLib-DEBUG: 13:56:25.529: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:2889): GLib-GIO-DEBUG: 13:56:25.537: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.537: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.538: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.538: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.538: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.538: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:2889): dconf-DEBUG: 13:56:25.539: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:2889): GLib-GIO-DEBUG: 13:56:25.541: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)                                                                                                                                                              /25.0s
➜  ~ 

```

Sean-StarLabs, 2024-01-07:
> Sorry, what do you mean with stock packages int his case? referring to fwupd shipped with Elementary OS (did not test if this is the case)?

Yes i.e. remove the ones from the PPA

sees7ern, 2024-01-08:
uninstalled fwupd completely and re-installed it various ways, following the official guidelines in each case. encountered the following errors:

**FLATPAK**
> sudo -i flatpak run org.freedesktop.fwupd update
> Loading…                 [****                                   ]19:25:21.313 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
> Loading…                 [************************************** ]
> WARNING: This package has not been validated, it may not work properly.
> Devices with no available firmware updates: 
>  • Star Drive PCIe SSD
>  • SuperIO FuSuperioIt89Device

-----

**SNAP**
> fwupdmgr refresh
> Updating lvfs
> Downloading…             [***************************************]
> (/snap/fwupd/5533/bin/fwupdmgr:7577): Fwupd-CRITICAL **: 19:19:31.218: fwupd_checksum_guess_kind: assertion 'checksum != NULL' failed
> Downloading…             [***************************************]
> Successfully downloaded new metadata: 0 local devices supported

Sean-StarLabs, 2024-01-08:
Neither the snap or flatpaks work - just needs to be via apt

sees7ern, 2024-01-08:
have just installed it via apt:

> fwupd is already the newest version (1.8.9+j).

> fwupdmgr refresh
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

sees7ern, 2024-01-08:
if it has any relevance

> Linux 6.2.0-39-generic

Sean-StarLabs, 2024-01-08:
> fwupd is already the newest version (1.8.9+j).

That's not the stock package.

First, make sure whatever snap/flatpak stuff has been done, has been undone. Then run
```
sudo apt install ppa-purge
sudo ppa-purge ppa:starlabs/main
sudo ppa-purge ppa:starlabs/coreboot
```

sees7ern, 2024-01-08:
so, i got rid with anything linked to fwupd via purge command. there is also nothing left from the snap and flatpak version as far as I can see. i removed the ppa previously. however, it does not seem to have improved the situation.

> sudo ppa-purge ppa:starlabs/main
Updating packages lists
PPA to be removed: starlabs main
Warning:  Could not find package list for PPA: starlabs main

> sudo ppa-purge ppa:starlabs/coreboot
Updating packages lists
PPA to be removed: starlabs coreboot
Warning:  Could not find package list for PPA: starlabs coreboot

> fwupdmgr refresh
Command 'fwupdmgr' not found, but can be installed with:
sudo snap install fwupd  # version 1.9.11, or
sudo apt  install fwupd  # version 1.7.9-1~22.04.3
See 'snap info fwupd' for additional versions.

now if i try to re-install fwupd, i get the following:

> sudo apt install fwupd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 fwupd : Depends: libfwupd2 (= 1.7.9-1~22.04.3) but 1.8.9+j is to be installed
         Depends: libfwupdplugin5 (= 1.7.9-1~22.04.3) but it is not going to be installed
         Recommends: bolt but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

Sean-StarLabs, 2024-01-08:
> so, i got rid with anything linked to fwupd via purge command

Purge alone wont do it.

Can you post:
```
cat /etc/apt/sources.list.d/*starlabs*
```

sees7ern, 2024-01-08:
cat /etc/apt/sources.list.d/*starlabs*
cat: '/etc/apt/sources.list.d/*starlabs*': No such file or directory

Sean-StarLabs, 2024-01-08:
Right, and this:
```
sudo apt update && sudo apt list | grep fwupd
```

sees7ern, 2024-01-08:
`sudo apt update && sudo apt list | grep fwupd
Get:1 http://mirrors.ubuntu.com/mirrors.txt Mirrorlist [495 B]
Hit:4 https://brave-browser-apt-release.s3.brave.com stable InRelease                                                                                    
Hit:5 https://ppa.launchpadcontent.net/elementary-os/stable/ubuntu jammy InRelease                                                                       
Hit:6 https://ppa.launchpadcontent.net/kisak/kisak-mesa/ubuntu jammy InRelease
Hit:7 https://deb.librewolf.net focal InRelease                     
Hit:8 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:9 https://ppa.launchpadcontent.net/elementary-os/os-patches/ubuntu jammy InRelease
Hit:10 https://ppa.launchpadcontent.net/philip.scott/pantheon-tweaks/ubuntu jammy InRelease
Hit:3 https://mirrors.hosterion.ro/ubuntu jammy-updates InRelease
Hit:2 https://ubuntu-mirror.magnetic-it.com/ubuntu jammy InRelease
Fetched 495 B in 6s (85 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up-to-date.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

fwupd-amd64-signed-template/jammy-updates 1:1.4-0ubuntu0.1 amd64
fwupd-doc/jammy-updates,jammy-updates 1.7.9-1~22.04.3 all
fwupd-signed/jammy-updates,now 1.51.1~22.04.1+1.4-0ubuntu0.1 amd64 [installed]
fwupd-tests/jammy-updates 1.7.9-1~22.04.3 amd64
fwupd-unsigned-dev/jammy-updates 1:1.4-0ubuntu0.1 amd64
fwupd-unsigned/jammy-updates 1:1.4-0ubuntu0.1 amd64
fwupd/jammy-updates 1.7.9-1~22.04.3 amd64
fwupdate/jammy 12-7 amd64
gir1.2-fwupd-2.0/jammy-updates 1.7.9-1~22.04.3 amd64
gir1.2-fwupdplugin-1.0/jammy-updates 1.7.9-1~22.04.3 amd64
libfwupd-dev/jammy-updates 1.7.9-1~22.04.3 amd64
libfwupd2/now 1.8.9+j amd64 [installed,local]
libfwupdplugin-dev/jammy-updates 1.7.9-1~22.04.3 amd64
libfwupdplugin5/jammy-updates 1.7.9-1~22.04.3 amd64
plasma-discover-backend-fwupd/jammy-updates 5.24.7-0ubuntu0.1 amd64
`

Sean-StarLabs, 2024-01-08:
Reinstall libfwupd2 then it should work

sees7ern, 2024-01-08:
`sudo apt install libfwupd2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libfwupd2 is already the newest version (1.8.9+j).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.`

`sudo apt install fwupd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 fwupd : Depends: libfwupd2 (= 1.7.9-1~22.04.3) but 1.8.9+j is to be installed
         Depends: libfwupdplugin5 (= 1.7.9-1~22.04.3) but it is not going to be installed
         Recommends: bolt but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`

Sean-StarLabs, 2024-01-08:
Reinstall, not install:
```
sudo apt install --reinstall libfwupd2
```

sees7ern, 2024-01-08:
`sudo apt install --reinstall libfwupd2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reinstallation of libfwupd2 is not possible, it cannot be downloaded.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
`

Sean-StarLabs, 2024-01-08:
Hmm, what's `sudo apt remove libfwupd2` say?

sees7ern, 2024-01-08:
`sudo apt remove libfwupd2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  io.elementary.feedback libadwaita-1-0 libjcat1 liblightdm-gobject-1-0 lightdm numlockx switcheroo-control
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED
  io.elementary.greeter io.elementary.onboarding io.elementary.settings-daemon libfwupd2 switchboard-plug-about switchboard-plug-pantheon-shell
  switchboard-plug-security-privacy wingpanel-indicator-a11y
0 to upgrade, 0 to newly install, 8 to remove and 0 not to upgrade.
After this operation, 6,785 kB disk space will be freed.
Do you want to continue? [Y/n] 
`

sees7ern, 2024-01-08:
it is nothing urgent to resolve the fwupd issue. the laptop runs stable enough. happy to give you more time to think and reply. already appreciate all your support very much, and apologise for any inconvenience. worst case, i can always re-install my system and try again, even though it is not my preferred solution.

Sean-StarLabs, 2024-01-10:
Try remove this first, only proceed if it doesn't want to remove anything else:
```
sudo apt remove libfwupdplugin5
```
If it lets you remove it, do:

```
sudo apt install libfwupd2=1.7.9-1~22.04.3
```
And if not:
```
sudo apt install libfwupd2=1.7.9-1~22.04.3 libfwupdplugin5=1.7.9-1~22.04.3

```

sees7ern, 2024-01-10:
```
sudo apt remove libfwupdplugin5
[sudo] password for cb:                          
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'libfwupdplugin5' is not installed, so not removed
0 to upgrade, 0 to newly install, 0 to remove and 14 not to upgrade.
```

```
sudo apt install libfwupd2=1.7.9-1~22.04.3
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be DOWNGRADED:
  libfwupd2
0 to upgrade, 0 to newly install, 1 to downgrade, 0 to remove and 14 not to upgrade.
Need to get 106 kB of archives.
After this operation, 14.3 kB disk space will be freed.
Do you want to continue? [Y/n] y
Get:1 http://mirrors.ubuntu.com/mirrors.txt Mirrorlist [823 B]
Get:2 http://mirror.cov.ukservers.com/ubuntu jammy-updates/main amd64 libfwupd2 amd64 1.7.9-1~22.04.3 [106 kB]
Fetched 106 kB in 1s (190 kB/s)    
dpkg: warning: downgrading libfwupd2:amd64 from 1.8.9+j to 1.7.9-1~22.04.3
(Reading database ... 212004 files and directories currently installed.)
Preparing to unpack .../libfwupd2_1.7.9-1~22.04.3_amd64.deb ...
Unpacking libfwupd2:amd64 (1.7.9-1~22.04.3) over (1.8.9+j) ...
Setting up libfwupd2:amd64 (1.7.9-1~22.04.3) ...
Processing triggers for libc-bin (2.35-0ubuntu3.5) ...
```

**with and without**
`sudo add-apt-repository ppa:starlabs/main
sudo add-apt-repository universe`

**i now get the following**
```
fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported

```

Sean-StarLabs, 2024-01-10:
> with and without

Stay without - that's just going to re-add the newer builds

> i now get the following

Cool. What's the output of:
```
sudo apt install flashrom
sudo flashrom -p internal
```

sees7ern, 2024-01-10:
```
sudo flashrom -p internal
flashrom unknown on Linux 6.5.0-14-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76b21000.
Found chipset "Intel Tiger Lake U Premium".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00500000-0x00ffffff) is read-write.
FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
OK.
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
No operations were specified.
```

Sean-StarLabs, 2024-01-10:
```
sudo wget https://raw.githubusercontent.com/fwupd/fwupd/1_7_X/plugins/flashrom/flashrom.quirk -O /usr/share/fwupd/quirks.d/flashrom.quirk
sudo systemctl restart fwupd
```
Should do it

sees7ern, 2024-01-10:
```
sudo wget https://raw.githubusercontent.com/fwupd/fwupd/1_7_X/plugins/flashrom/flashrom.quirk -O /usr/share/fwupd/quirks.d/flashrom.quirk
--2024-01-10 21:12:42--  https://raw.githubusercontent.com/fwupd/fwupd/1_7_X/plugins/flashrom/flashrom.quirk
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3236 (3.2K) [text/plain]
Saving to: ‘/usr/share/fwupd/quirks.d/flashrom.quirk’

/usr/share/fwupd/quirks.d/flashrom.qui 100%[==========================================================================>]   3.16K  --.-KB/s    in 0s      

2024-01-10 21:12:42 (11.3 MB/s) - ‘/usr/share/fwupd/quirks.d/flashrom.quirk’ saved [3236/3236]

```

`sudo systemctl restart fwupd`

```
fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
```

Sean-StarLabs, 2024-01-10:
Assuming requirements are met (battery >30%, charger connected etc), try a reboot just incase elem is holding onto something, and then can you post the output of:
```
sudo fwupdtool get-devices --plugins flashrom --show-all 
```

sees7ern, 2024-01-10:
Confirming that laptop has been rebooted, has over 80% of battery and has charger connected:

`sudo fwupdtool get-devices --plugins flashrom --show-all 
21:20:43:0908 FuHistory            schema version 8 is unknown
Loading…                 [***************************************]
No detected devices
`

Sean-StarLabs, 2024-01-11:
Cool - `sudo rm /var/lib/fwupd/pending.db`

sees7ern, 2025-01-18:
Apologies for commenting here after you closed the issue, but I want to add the following as it may help others in the future:

I tried many things to resolve the issue but nothing worked, including the above.

It was only an install of Elementary OS 8 Circe replacing Elementary OS 7.1 that resolved the issue on my StarBook Mk V. I have since been able to update to firmware version 24.10 following the guidance given here: https://support.starlabs.systems/kb/firmware/getting-started

Also, I can confirm that the same goes for my StarLite MK IV.