StarLabsLtd/firmware #148: [StarBook Mk VI - Intel - coreboot 8.99] Machine freezes after coming out of Sleep

vchongmorrison, 2024-01-11:
I updated coreboot from 8.40 to 8.99, which resulted in the machine unable to boot correctly while coming out of Sleep mode. The cursor displays as a grey square, 'Authentication error' blinks on the screen, and keyboard is unresponsive so logging in was not possible. This does not happen when powering on after Shutdown - forced or otherwise.

Downgrading the firmware to 8.60 appears to fix the issue. When I re-upgrade to 8.99 the same issue returned, hence I downgraded again to 8.60 for now.

EC firmware release was 1.1 throughout.

Sean-StarLabs, 2024-01-11:
Which StarBook? 12th or 13th Gen?

vchongmorrison, 2024-01-11:
> Which StarBook? 12th or 13th Gen?

12th Gen Intel® Core™ i7-1260P × 16