StarLabsLtd/firmware #150: Cannot update coreboot for StarBook VI (Intel)

zeroskyx, 2024-02-27:
Greetings,

attempting to update coreboot via `sudo fwupdmgr update` fails with `failed to open device for prepare: failed to read layout from Intel ICH descriptor`

fwupdmgr (and the whole system) is updated. System is `Linux zeroskyx-star6 6.7.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 23 Feb 2024 16:31:48 +0000 x86_64 GNU/Linux` running on Arch/ Gnome 45.

Help is greatly appreciated.

Thanks in advance
-Simon

Sean-StarLabs, 2024-02-27:
Hi

> sudo fwupdmgr update

That'll break the database, see [here](https://support.starlabs.systems/kb/firmware/lvfs-terminal)

But assuming it's not [this](https://support.starlabs.systems/kb/firmware/lvfs-troubleshooting), then unfortunately, not a foggy. I'd hazard a guess that something in your kernel config is blocking access, as having a version of flashrom that predates decriptors would be impressive.

zeroskyx, 2024-02-28:
Hi Sean,

thanks for the feedback! Turns out it was something else: the [flashrom](https://archlinux.org/packages/extra/x86_64/flashrom/) package on Arch (v1.2) doesn't appear to work with the update.  
Installing [flashrom-git](https://aur.archlinux.org/packages/flashrom-git) from the AUR (v1.4 -- don't be fooled by the version in the title) solved the issue and the update worked nicely.

Maybe putting this piece of information into the update README could also help others :)

Enjoy!

Sean-StarLabs, 2024-02-28:
I think [this](https://support.starlabs.systems/kb/firmware/lvfs-requirements) covers it already

closed 2024-02-29 by zeroskyx