StarLabsLtd/firmware #151: Labtop MKIV function keys

BrtSak, 2024-03-07:
The "special" function keys don't seem to work with the 24.02 coreboot firmware. I am referring to the volume, keyboards backlight, screen brightness, etc. These worked fine with 8.99 release.

Sean-StarLabs, 2024-03-07:
What's your EC version? If it's mismatched, i.e. not 24.02., it'll need updating. You can just manually update it with:

```
sudo wget https://github.com/StarLabsLtd/firmware/raw/master/binaries/ifu.efi -O /boot/efi/ifu.efi
sudo wget https://github.com/StarLabsLtd/blobs/raw/main/mainboard/starlabs/starbook/cml/ec.bin -O /boot/efi/ec.bin
```
Then, with the charger connected, reboot to the EFI shell and enter the below commands (there is a chance it could be fs1, fs2 etc, depending on your setup. If you get a message saying can't find ifu.efi, please try them in ascending order):
```
fs0:
ifu.efi ec.bin 128
```

BrtSak, 2024-03-08:
Coreboot was at 24.02, but EC was at 1.07 I think. I updated the EC according to your directions.

The keyboard backlight now has two levels. Great! 😊

Thank You for the updates and support.

BrtSak, 2024-03-11:
Update:

With 24.02 EC you linked above, the Labtop MKIV would not charge. I was getting a message that the ac power is connected and charging, but the battery kept discharging.

At 32% battery, I downgraded firmware to 8.99. The EC was still at 24.02. By then the battery was down to 26%. So, I downloaded the ITE 1.07 (https://github.com/StarLabsLtd/firmware/tree/master/LabTop/MkIV/ite/1.07) to a usb flash stick and it was auto loaded from the EFI shell. I am confused now because EC shows up as version 1.7. Shouldn't it be 1.07?

The battery is charging but which EC should I be using?

macmpi, 2024-03-11:
Got all keys back with EC & coreboot 24.02.
Had to reset power-management also (ctrl-atl-del-power), and I'm in for battery re-calibration (EFI boot and `mm` commands) to see if power-management gets stabilized...

However, lock key save-state is lost post 24.02 (both coreboot & EC): it is never active after reboot.

Sean-StarLabs, 2024-03-11:
> With 24.02 EC you linked above, the Labtop MKIV would not charge. I was getting a message that the ac power is connected and charging, but the battery kept discharging.

Sounds like it was calibrating.

> The battery is charging but which EC should I be using?

24.02

> and I'm in for battery re-calibration

What do you mean?

> However, lock key save-state is lost post 24.02 (both coreboot & EC): it is never active after reboot.

That's not a firmware bug, feel free to ping support.

macmpi, 2024-03-14:
> 24.02
> 
> > and I'm in for battery re-calibration
> 
> What do you mean?

Since I'm still having sudden power-off @ about reported 23% battery, I went to battery re-calibration procedure mentioned by support (boot under `efi`,  issue a few `mm` commands, reboot and let 3 battery discharge/charge cycles).
After all this, I'm not convinced new EC improved situation.

> > However, lock key save-state is lost post 24.02 (both coreboot & EC): it is never active after reboot.

Went on inverting lock-key setting in coreboot menu.

Sean-StarLabs, 2024-03-14:
> Since I'm still having sudden power-off @ about reported 23% battery, I went to battery re-calibration procedure mentioned by support (boot under efi, issue a few mm commands, reboot and let 3 battery discharge/charge cycles).
After all this, I'm not convinced new EC improved situation.

What MM commands? Do you want to ping me a link or let me know your name so I can find what you've done?

> Went on inverting lock-key setting in coreboot menu.

Just press the lock key?

BrtSak, 2024-03-15:
So less than a week after I updated the firmware to 24.02 and then manually updated the ITE/EC to 24.02 my WIFI module appears to be dead. The network manager doesn't detect it. I made sure it is enabled in the platform setup menu. I also downgraded back to 8.96/1.07 and it's still dead. It's most likely a coincidence. 

But now I have no WIFI. The WIFI pcb is soldered down in the Labtop. Can it be de-soldered easily? It looks like it is soldered down from the bottom. Is it a ball grid array footprint or similar? What are my options to get this fixed?

Sean-StarLabs, 2024-03-15:
Better to look at software than firmware, but if you need to replace it, it's BGA. Depends on where you are in the world, but around here, it usually costs about £30.

macmpi, 2024-03-18:
> > With 24.02 EC you linked above, the Labtop MKIV would not charge. I was getting a message that the ac power is connected and charging, but the battery kept discharging.
> 
> Sounds like it was calibrating.

I'm also experiencing this: LabTop is discharging while connected to my (working) charger...to suddenly power-off after a while.
If that's part of automated re-calibration procedure, can you please explain/document in more detail what's going on, and how long/how many cycles will it go through.

Currently I just see the % gauge  going down while plugged, and then sudden power-off.
Leaving plugged does not re-charge either: have to `ctrl-atl-del-power` to restore some life and use the computer for a little while...until it depletes again.
Really not intuitive process without a more detailed explanation.

divico, 2024-03-22:
> Really not intuitive process without a more detailed explanation.

I totally agree. Users definitely deserve a better explanation of why their machines suddenly force a hard shutdown.

BrtSak, 2024-03-22:
FYI. I have been on the 24.02 firmware/ec for a week now. I set the max battery level to 60%/80%. But it always charges to 100%.

I rolled back to 8.96/1.07 for now. I feel like the Labtop MKIV firmware needs more work. No 24.02 issues so far with my Starbook MKVI.

divico, 2024-03-24:
> FYI. I have been on the 24.02 firmware/ec for a week now. I set the max battery level to 60%/80%. But it always charges to 100%.

I experience the same issue on the MKVI. Charge limit is not always honored.

> No 24.02 issues so far with my Starbook MKVI.

MKVI also has the issue of calibration forcing a hard shutdown, see #153

Sean-StarLabs, 2024-03-25:
> I experience the same issue on the MKVI. Charge limit is not always honored.

There must be a misunderstanding in what you're expecting of it - "not always" isn't possible i.e. "always" or "never" are plausible. What are you expecting to happen?

divico, 2024-03-26:
> There must be a misunderstanding in what you're expecting of it - "not always" isn't possible i.e. "always" or "never" are plausible. What are you expecting to happen?

I have the charge limit set to 60% in CFR/coreboot. However, I have often found that the laptop has been charged to 100% above this threshold. I have not found the cause yet, but it seems to happen more often with the barrel connected.

Sean-StarLabs, 2024-03-26:
Feel free to log an issue with an EC dump

TomWilford, 2024-10-14:
> What's your EC version? If it's mismatched, i.e. not 24.02., it'll need updating. You can just manually update it with:
>
> ```
> sudo wget [...]

The first wget should be this now, from the looks of it
```
sudo wget https://github.com/StarLabsLtd/firmware/raw/main/binaries/ifu.efi -O /boot/efi/ifu.efi
```