StarLabsLtd/firmware #152: [Starbook MK VI - Intel] USB over-current condition and USB disconnects under high load

mysfitt, 2024-03-11:
I've been noticing for [some time now](https://github.com/StarLabsLtd/firmware/issues/140#issue-2010755846) that my i7-1260p has been having disconnects with AX210 bluetooth chipset. I recently bought a trackball that uses a USB dongle with  2.4ghz RF instead of bluetooth, and it also disconnected today, and I think I've figured out the connection.  When the machine is under high load, I will get this in the kernel log:
```
Mar 11 12:47:55 gryphon kernel: usb 1-6: USB disconnect, device number 5
Mar 11 12:47:56 gryphon kernel: usb usb1-port7: over-current condition
Mar 11 12:47:56 gryphon kernel: usb 1-7: USB disconnect, device number 6
Mar 11 12:47:57 gryphon kernel: usb usb1-port6: over-current condition
```
In this particular example, I was running a build for OpenWRT, which is multi-threaded and involves  not only a kernel build, but a full system build with an arm cross-compiler.  Other examples of disconnects have been when playing a game on Steam with a bluetooth controller, and when playing music to a bluetooth speaker on Spotify, which is a crappy Electron app that consumes way too many system resources. I know that the Bluetooth chipset is connected through the USB bus, so when I had the failure today (that did not involve bluetooth) I realized that it's not Bluetooth being flaky, it's the USB bus.  Note that I do have my machine set to "performance" in the BIOS. I'm running the latest version as of this writing, Coreboot 24.02.

Is this something that can be fixed in firmware, or am I actually seeing a hardware failure?
### Steps to Reproduce
1. connect a bluetooth device, such as a mouse or speaker
2. Put a high load on the system. I'd recommend a large compile, such as the kernel. A benchmarking suite should also do the trick.
3. Monitor the kernel log for the over-current condition message

Sean-StarLabs, 2024-03-11:
Are you outside the EU by any chance?

mysfitt, 2024-03-12:
Yes, I'm in the US. I'm same guy that's been bugging you about flashrom not working on Arch :) What would country code have to do with it? Wifi regulatory domain and Radar detection stuff?

Sean-StarLabs, 2024-03-14:
It's something to do with certain channels shouldn't be used at max power. I'm pretty sure we can limit it from the firmware so I'll take a look

mysfitt, 2024-03-15:
Cool. Let me know if there's useful logs I can provide. I keep a pretty good history buffer in my journal.

Sean-StarLabs, 2024-04-26:
Can you try 24.04 (once it's jumped through the LVFS's hoops), with PCI Clock Power Management, PCI ASPM and PCI L1 Substates all disabled?

mysfitt, 2024-10-08:
@Sean-StarLabs I just wanted to follow up and note that I haven't been able to test 24.04 since it's still not in LVFS or LVFS-testing after 6 months :/ Was the build found to be unstable? The Bluetooth/USB issue continues to be a problem for me. Strangely, if I boot off an old 6.1 that I still have on the machine, the problem with bluetooth seems to go away. So perhaps it's some interaction with firmware versions or kernel driver versions.

Sean-StarLabs, 2024-10-09:
I checked this dby, and thought, must be fixed as I haven't heard anything - turns out I just never promoted 24.04 to stable, sorry!

24.04 is good to use, but literally uploading 24.09 at the moment - that's probably a better bet, should be stable by Friday.

codemacabre, 2024-10-11:
I have issues with the Bluetooth adapter not being found on my StarBook Mk VIr2; not sure if it's the same or related issue. It's really sporadic; it sometimes works after installing the OS, but then on subsequent restarts seemingly disappears from the system and no amount of tinkering with things like `sudo systemctl enable bluetooth` or `bluetoothctl` shows any available Bluetooth adapter. Reinstalling the OS then leads to the Bluetooth adapter not being found at all.

I'm temporarily using a Bluetooth dongle, but would prefer this to be fixed.

I'm on coreboot 24.04 (using `fwupdmgr get-devices --show-all-devices` doesn't list the BIOS for some reason, but `cat /sys/class/dmi/id/bios_version` lists the correct version).

Sean-StarLabs, 2024-10-11:
24.09 should help, shutting down and letting it go to G3 should solve until that's available.

> doesn't list the BIOS for some reason

Either old version of fwupd, or iomem needs to be set to relaxed,

codemacabre, 2024-10-11:
> letting it go to G3 should solve until that's available.

That worked. Thank you! I'll look forward to the 24.09 release.

> Either old version of fwupd, or iomem needs to be set to relaxed,

I just reinstalled fwupd (to version 1.8.12-2) and added `iomem=relaxed` to my GRUB, restarted, but no change to my `get-devices` output.

Sean-StarLabs, 2024-10-11:
1.8.12-2 predates your laptop

codemacabre, 2024-10-11:
Ah, the downside to using Debian stable, I suppose.

Sean-StarLabs, 2024-10-11:
Isn't the kernel in Debian Stable still 6.1? Because that predates the laptop :)

mysfitt, 2024-10-15:
@Sean-StarLabs When 24.09 drops, I'll give it a go and see if there's any change. I suspect this might be a kernel thing since I've seen some similar issues reported on different distros and hardware. I strongly suspect radio interference with the WiFi since Intel chips have been known to have that problem.  But then there's also the USB weirdness that I've observed, so it's hard to know which one is the culprit.

Sean-StarLabs, 2024-10-22:
How'd it go?

closed 2024-11-21 by Sean-StarLabs