StarLabsLtd/firmware #153: [Starbook MK VI - Intel] won't charge on USB PD

divico, 2024-03-13:
I've been running coreboot `24.02` and EC `24.02` for a few days without any problems, but today it suddenly stopped charging via USB PD. It was connected to a Dell WD 17, but plugging in the Starlabs charger didn't solve the problem.
As I was in a business video call, I was unable to restart and it completely drained the battery until a hard shutdown.  I had to plug in the barrel jack to get it to charge again.

I have no idea how to reproduce this as it seems to be working fine now. The system was running at quite a high load, maybe this is related to #152 

I'm happy to help with debugging if you have any ideas on how to track this down.

mysfitt, 2024-03-14:
@divico The webcam is also on the internal USB bus. There is definitely something weird going on with USB on these machines. The trigger seems to be a high load condition.

Sean-StarLabs, 2024-03-14:
Battery calibration has absolutely nothing to do with USB, it's i2c...

divico, 2024-03-14:
I'm very much looking forward to this fix. Will it be available on lvfs-testing?

It may just be calibration, but it is a serious usability issue if it suddenly forces a hard shutdown. 
I'm not competent to understand the code of the fix. I hope the fix is not just about switching off calibration, as I see there is an option to disable it.

While I can't say much about the cause, the correlation between the battery/calibration/non-charging issue and the camera turning on and off was clear. It never happened before and it never happened again.

One question which may be related: today, the power led turned BLUE when connecting the USB PD charger. It did charge the battery but I never saw it blue before. What does BLUE mean?

Sean-StarLabs, 2024-03-14:
> What does BLUE mean?

Charging the upper limit of the battery.

closed 2024-03-20 by Sean-StarLabs