StarLabsLtd/firmware #154: Star Lite Mk III Not charging on certain usb-c chargers / docking stations

only1thor, 2024-03-17:
I have noticed that the MK III does not change on certain chargers. 
The Lenovo ADLX65YCC3D 65Watt usb-C charger for thinkpads is the one i have on hand, that does not charge the laptop.
It works fine on other laptops, like the thinkpad it came with naturally, and also my framework laptop. 

Due to this i decided to make a small usb-c PD to 12v cable, and found that it still does not charge. 
That lead me to check the voltage on the cable on the original charger vs the Thinkpad charger. 
And i found that it seems like the Thinkpad charger does not supply 12V when it is requested by the PD chip. 
And i can confirm that 12v is not on the list of voltages listed on the info sticker on the Thinkpad charger. 
(although it lists 20,15,9 and 5V)

The star light MK III does go into the "charging" state when i connect it to the Thinkpad charger, but it slowly depletes.

Is there a way to rectify this? or  does the MK III simply not support charging from a charger that does not supply 12v?

Sean-StarLabs, 2024-03-17:
I'd suggest checking your EC is up to date, but they support PD 3.0 - other standards won't work.

Not sure I exactly get what you tried, but putting a fixed voltage into a PD device is a bad idea.

only1thor, 2024-03-26:
Can the EC be updated via `fwupd`?

The fixed voltage was to the barrel connector on the laptop. 
i was trying to hack a solution that would allow me to charge using ThinkPad chargers i have laying arround. 
The USB-C PD to 12V adapter i made used a "ZY12PDN" board that negotiates for 12V from a PD charger. 
I wired the 12V output to a barrel jack that fit the MKIII, and checked the voltages before connecting. 
Thats when i noticed the voltage was at 5V when i used the ThinkPad USB-C chargers. 

The result is that i still can not charge on thinkpad usb-c chargers since even the hacky cabel i made only delivers 5V when used on the thinkpad chargers.

Sean-StarLabs, 2024-03-26:
> Can the EC be updated via fwupd?

No, only the EFI Shell.

Oh, I'm with you. To answer your original question, no - it has to be 12V.

only1thor, 2024-04-02:
i see. 
i found [this guide on installing coreboot](https://support.starlabs.systems/kb/guides/installing-coreboot) (seems risky on the mk III)

and just found [this guide](https://support.starlabs.systems/kb/firmware/efi-shell) on updating via EFI shell. I assume thats the right one?
and these are the binaries i need to install with EFI shell?
https://github.com/StarLabsLtd/firmware/tree/master/StarLite/MkIII

Sean-StarLabs, 2024-04-02:
>  I assume thats the right one?

Yes

EC can be downloaded [here](https://github.com/StarLabsLtd/firmware/blob/master/StarLite/MkIII/ITE/1.14/efi-I3.zip).

closed 2024-04-02 by only1thor