StarLabsLtd/firmware #155: [Starbook MK VI - Intel] USB-C repeated disconnection and reconnection

divico, 2024-03-26:
Since the upgrade to coreboot `24.02` and EC `24.02`, I cannot connect my smartphone using USB-C. The phone shows repeatedly charging and discharging and the phone is never recognised by the laptop. Using a USB-A port works fine.

Here some logs generated when connecting the phone on the USB-C port.

```
journalctl --system -f
Mär 26 10:13:43 starbook kernel: usb usb1-port1: attempt power cycle
Mär 26 10:13:44 starbook kernel: usb 1-1: new full-speed USB device number 13 using xhci_hcd
Mär 26 10:13:44 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:13:44 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:13:44 starbook kernel: usb 1-1: device not accepting address 13, error -71
Mär 26 10:13:44 starbook kernel: usb 1-1: new full-speed USB device number 14 using xhci_hcd
Mär 26 10:13:44 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:13:44 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:13:45 starbook kernel: usb 1-1: device not accepting address 14, error -71
Mär 26 10:13:45 starbook kernel: usb usb1-port1: unable to enumerate USB device
Mär 26 10:14:48 starbook kernel: usb 1-1: new full-speed USB device number 15 using xhci_hcd
Mär 26 10:14:48 starbook kernel: usb 1-1: device descriptor read/64, error -71
Mär 26 10:14:48 starbook kernel: usb 1-1: device descriptor read/64, error -71
Mär 26 10:14:49 starbook kernel: usb 1-1: new full-speed USB device number 16 using xhci_hcd
Mär 26 10:14:49 starbook kernel: usb 1-1: device descriptor read/64, error -71
Mär 26 10:14:49 starbook kernel: usb 1-1: device descriptor read/64, error -71
Mär 26 10:14:49 starbook kernel: usb usb1-port1: attempt power cycle
Mär 26 10:14:49 starbook kernel: usb 1-1: new full-speed USB device number 17 using xhci_hcd
Mär 26 10:14:49 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:14:50 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:14:50 starbook kernel: usb 1-1: device not accepting address 17, error -71
Mär 26 10:14:50 starbook kernel: usb 1-1: new full-speed USB device number 18 using xhci_hcd
Mär 26 10:14:50 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:14:50 starbook kernel: usb 1-1: Device not responding to setup address.
Mär 26 10:14:50 starbook kernel: usb 1-1: device not accepting address 18, error -71
Mär 26 10:14:50 starbook kernel: usb usb1-port1: unable to enumerate USB device
```

Sean-StarLabs, 2024-03-26:
What phone? Pixel by any chance?

divico, 2024-03-26:
> What phone? Pixel by any chance?

Yes Pixel 6a

Sean-StarLabs, 2024-03-26:
Not familiar myself, but there's a way to make the Pixel stop trying to charge the StarBook - think it shows as a notification when you connect it

divico, 2024-03-26:
> Not familiar myself, but there's a way to make the Pixel stop trying to charge the StarBook - think it shows as a notification when you connect it

_Charge connected device_ on the phone is off. This problem does not occur with previous firmware or with another laptop.

Sean-StarLabs, 2024-04-17:
Fixed in 24.04

closed 2024-04-17 by Sean-StarLabs