StarLabsLtd/firmware #156: [Starbook MK VI - AMD] Fan curve adjustment 24.03 is way more noise than before

lil0r, 2024-03-31:
Flashed Bios 24.03 (EC 24.03) before easter and I noticed a significant change in noise through fan behaviour.
Single simple workloads like browsing web with chrome or firefox cause to spin the fan up. Running a windows VM sounds like a jet taking off. 
Beforehand I was on Bios 1.16.0 (EC 2.6) which was almost always silent until I compiled a kernel or something.

Is there any way to change powermanagement to something more like EC 2.6?

Sean-StarLabs, 2024-03-31:
Not from the firmware (although changing the fan curve will indirectly change it).

lil0r, 2024-03-31:
Thanks Sean,
what can I do to tame the jet engine?

Sean-StarLabs, 2024-03-31:
Change the fan curve, or performance settings in the os

lil0r, 2024-04-01:
Thanks Sean,

Havent changed anything other than the bios/EC update - could you please elaborate what has changed in regard to PM & fan management in the two verions? Otherwise, it will be trial and error...

Sean-StarLabs, 2024-04-01:
Nothing specific (granted, AMD change logs leave a lot to be desired) but something's getting lost in translation. Fan curve has 3 settings as does the performance, sure, trail and error, but turn one down, if it's not quiet enough, turn the other down

lil0r, 2024-04-01:
Fan curve in the  Bios 1.16.0 (EC 2.6) previously were a bit odd - Normal profil was much quiter than quite profil.
First thing I played around obviously was the Bios setting, without much effect.
Since that lead to nowhere, I started manually tweaking the ryzen PM Table. But really, I dont want to reverse engineer the changes - hence I asked.

`root@StarBook:/usr/src/linux# /usr/local/bin/ryzenadj -i
CPU Family: Cezanne
SMU BIOS Interface Version: 19
Version: v0.15.0 
PM Table Version: 400005
|        Name         |   Value   |     Parameter      |
|---------------------|-----------|--------------------|
| STAPM LIMIT         |    15.000 | stapm-limit        |
| STAPM VALUE         |     2.841 |                    |
| PPT LIMIT FAST      |    30.000 | fast-limit         |
| PPT VALUE FAST      |     3.844 |                    |
| PPT LIMIT SLOW      |    25.000 | slow-limit         |
| PPT VALUE SLOW      |     2.749 |                    |
| StapmTimeConst      |   150.000 | stapm-time         |
| SlowPPTTimeConst    |     5.000 | slow-time          |
| PPT LIMIT APU       |    25.000 | apu-slow-limit     |
| PPT VALUE APU       |     2.749 |                    |
| TDC LIMIT VDD       |    33.000 | vrm-current        |
| TDC VALUE VDD       |     0.332 |                    |
| TDC LIMIT SOC       |    13.000 | vrmsoc-current     |
| TDC VALUE SOC       |     1.162 |                    |
| EDC LIMIT VDD       |    90.000 | vrmmax-current     |
| EDC VALUE VDD       |    16.135 |                    |
| EDC LIMIT SOC       |    17.000 | vrmsocmax-current  |
| EDC VALUE SOC       |     0.000 |                    |
| THM LIMIT CORE      |   100.000 | tctl-temp          |
| THM VALUE CORE      |    45.875 |                    |
| STT LIMIT APU       |     0.000 | apu-skin-temp      |
| STT VALUE APU       |     0.000 |                    |
| STT LIMIT dGPU      |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU      |     0.000 |                    |
| CCLK Boost SETPOINT |    95.000 | power-saving /     |
| CCLK BUSY VALUE     |    12.374 | max-performance    |
`

Sean-StarLabs, 2024-04-01:
Did you try the fan curve in the version that you are running?

lil0r, 2024-04-01:
> Did you try the fan curve in the version that you are running?

running the quite fancurve set in the bios, yes.

Sean-StarLabs, 2024-04-03:
And which distro are you using?

Sean-StarLabs, 2024-04-04:
What versions?

alternateved, 2024-04-04:
I observe similar behaviour as @evansimo in regards to charging. Version for firmware is 24.3.0.

Sean-StarLabs, 2024-04-04:
EC, BIOS or both?

evansimo, 2024-04-04:
For me it's both

Sean-StarLabs, 2024-04-04:
Okay, happens post S3 maybe?

evansimo, 2024-04-04:
Good question. No - laptop was shut down, and it's been on, uninterrupted since this morning

Sean-StarLabs, 2024-04-04:
Hmm, happening now? Are you able to provide an EC dump? (`ectool -d`)

evansimo, 2024-04-04:
It's gone up to 94 since my last message, and it stopped again. As if some charge is slowly creeping in...

I tried to install ectools and succeeded, but the ectool command is not available. Do you know if there's an easy way to do it on Fedora 39?

alternateved, 2024-04-04:
My battery went to 96% and is currently stuck, but I've rebooted half an hour ago. My dump here on Fedora Beta 40:
```
EC RAM:

00: 18 03 00 00 01 00 00 00 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
30: 00 bb aa 00 00 00 bb 00 00 00 00 00 00 00 00 00 
40: 00 31 36 2f 33 33 2f 35 34 00 00 32 30 32 34 2f 
50: 30 33 2f 31 39 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 36 4d 00 00 16 01 00 00 00 05 00 00 00 00 00 01 
80: 07 01 05 01 fc 15 1e 2d a5 16 00 00 08 00 00 d9 
90: 15 ed 32 60 00 17 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

Ectool installed like so:
```sh
git clone http://review.coreboot.org/coreboot.git
cd coreboot/util/ectool
make
sudo make install
```

Sean-StarLabs, 2024-04-04:
> might be useless.

Maybe. 
```
10: 00 00 00 00 00 00 00 00 00 00 [00] 00 00 00 00 00 

```
00 is 100%, bb is 80% and aa is 60% - so the EC is doing what it's been told to do. Is it still set to 80% in AMI?

> I've also noticed after upgrade to 24.03 that in order to wake laptop from suspend I need to use power button. Before any key would be enough. 

Yes, that was changed in the common code a little while ago, it saves power.  Lid can also wake.

alternateved, 2024-04-04:
> 00 is 100%, bb is 80% and aa is 60% - so the EC is doing what it's been told to do. Is it still set to 80% in AMI?

Yup, it surely is.

[image]

> Yes, that was changed in the common code a little while ago, it saves power. Lid can also wake.

That's great!

Sean-StarLabs, 2024-04-04:
Cheers

Did you install it recently? Just as most of the EC memory is zeroed, and only the old EC used to do that

alternateved, 2024-04-04:
Hmm, `fwupdmgr get-history` displays `Last modified: 2024-04-02 18:04` so maybe that was the time. I don't remember exactly. That update showed up in Gnome Software recently and I've installed it.

rmsthebest, 2024-04-06:
Since this is becoming a catch-all issue for 24.03 amd fw update, I also have a question.
This release is the first that got tagged with "Issues found using FwHunt from Binarly" on LVFS.

I have held off updating for this reason. Is it an old issue only getting tagged now/false positive/nothing to worry about?

alternateved, 2024-04-07:
I've noticed that charge limit is not working anymore. Tested it with 80% and 60% limits and the cap does nothing.

Sean-StarLabs, 2024-04-26:
Fixed in 24.04

closed 2024-04-26 by Sean-StarLabs