StarLabsLtd/firmware #157: [Starbook MK VI - AMD] Battery drainage issue

lil0r, 2024-04-28:
Issue is that in S3 sleep battery drains about 1% per hour. I have read here that more people are suffering from this. Essentially, I have to run it plugged in and transport shut down. issue persist since I have the Starbook, 

- with all Bios & EC versions I have tested. currently running EC/BIOS 24.4.
- Stock & custom Kernels
- Encyrpted & un-encrypted swap/root
- Ubuntu/Gentoo/Arch

Which led me to believe it is a hardware issue. I would like to collect other peoples experience and configuration to identify the culprit. 
I have two theories
- 2x 32GB RAM -> large amount of ram, refresh drains the battery
- NVME -> incompatible NVME drains the battery (I have noticed that I have the PCIGen4 in my AMD model that should go into Intel model)

CONFIG  NVME
        **01:00.0 Non-Volatile memory controller: Phison Electronics Corporation E18 PCIe4 NVMe Controller (rev 01) (prog-if 02 [NVM Express])**
        **Subsystem: Phison Electronics Corporation E18 PCIe4 NVMe Controller**
        
CONFIG RAM
	**Part Number: F4-3200C22-32GRS**
	**Part Number: F4-3200C22-32GRS**

Sean-StarLabs, 2024-04-28:
What are you expecting? i.e. why do you think it's an issue?

> should go into Intel mode

Not sure where you got that from - that SSD doesn't have a specific SOC requirement.

Sean-StarLabs, 2024-04-29:
"sleep" is what exactly?

Maybe you're after S4 (Hibernate), not S3 (Suspend). No laptop can do weeks in S3; the target consumption is 1.5W, so even with the biggest battery you can put in a laptop, 100Wh, you'd be talking ~2.5 days. 

Ours are very efficient, so we can get more with a small battery, but pushing it further would just be grasping for minutes and compromising usability.

The drain in S4 is impercevable.

lil0r, 2024-04-30:
I am talking about closing the lid and picking up the laptop a couple of days later, opening the laptop and it resumes - not booting back from hibernate. Both my macbook and my thinkpad (win11) are able to do that for more than two days, with the macbook significantly outperforming the thinkpad.

Sean-StarLabs, 2024-04-30:
> I am talking about closing the lid and picking up the laptop a couple of days later, opening the laptop and it resumes - not booting back from hibernate.

I don't follow - if the OS is configured to hibernate, closing the lid will put it into hibernation - and opening it will resume.

> Both my macbook and my thinkpad (win11) are able to do that for more than two days, with the macbook significantly outperforming the ThinkPad.

Neither are comparable - I'm not sure if Apple has a name for it, but what they do is pretty much the same a hibernating.

lil0r, 2024-05-05:
root@starbook:~# ./amd_s2idle.py 
Location of log file (default s2idle_report-2024-05-05.txt)? 
Debugging script for s2idle on AMD systems
💻 Star Labs StarBook (B6) running BIOS 24.4 (24.4.0) released 04/25/2024 and EC 24.4
🐧 Ubuntu 24.04 LTS
🐧 Kernel 6.8.7
🔋 Battery BAT0 (Standard SR Real Battery) is operating at 97.14% of design
Checking prerequisites for s2idle
✅ Logs are provided via systemd
✅ AMD Ryzen 7 5800U with Radeon Graphics (family 19 model 50)
✅ LPS0 _DSM enabled
❌ ACPI FADT doesn't support Low-power S0 idle
✅ HSMP driver `amd_hsmp` not detected (blocked: False)
❌ PMC driver `amd_pmc` did not bind to any ACPI device
✅ GPU driver `amdgpu` available
❌ System isn't configured for s2idle in firmware setup
✅ NVME Phison Electronics Corporation E18 PCIe4 NVMe Controller is configured for s2idle in BIOS
✅ GPIO driver `pinctrl_amd` available
Your system does not meet s2idle prerequisites!
Explanations for your system

🚦 The kernel didn't emit a message that low power idle was supported
	Low power idle is a bit documented in the FADT to indicate that
	low power idle is supported.
	Only newer kernels support emitting this message, so if you run on
	an older kernel you may get a false negative.
	When launched as root this script will try to directly introspect the
	ACPI tables to confirm this.

🚦 AMD-PMC driver is missing
	The amd-pmc driver is required for the kernel to instruct the
	soc to enter the hardware sleep state.
	Be sure that you have enabled CONFIG_AMD_PMC in your kernel.
	If CONFIG_AMD_PMC is enabled but the amd-pmc driver isn't loading
	then you may have found a bug and should report it.

🚦 The system hasn't been configured for Modern Standby in BIOS setup
	AMD systems must be configured for Modern Standby in BIOS setup
	for s2idle to function properly in Linux.
	On some OEM systems this is referred to as 'Windows' sleep mode.
	If the BIOS is configured for S3 and you manually select s2idle
	in /sys/power/mem_sleep, the system will not enter the deepest hardware state.
root@starbook:~#

Sean-StarLabs, 2024-05-05:
What's the context?

lil0r, 2024-05-05:
> What's the context?

https://community.frame.work/t/responded-amd-7040-sleep-states/38101

Sean-StarLabs, 2024-05-05:
Right - but what's the relevance of Framework not supporting S3?

lil0r, 2024-05-05:
The Framework AMD seems to have similar issues and the are mantaining the above AMD debugging script https://gitlab.freedesktop.org/drm/amd/-/blob/master/scripts/amd_s2idle.py

Sean-StarLabs, 2024-05-05:
The link you posted is about the lack of S3 support and losing power in s2idle.

We support S3 and use pretty much the minimal amount of power possible.

To save beating around the bush, your convinced that S3 can use less power? Do you want me to tag in an AMD engineer to confirm what I'm saying?

lil0r, 2024-05-05:
sure

CommonLoon102, 2024-06-02:
@lil0r When you close the lid on Win11 then it will go to S3 and if you don't open the lid for a few hours then Win11 will wake itself up and go to hibernate (saving RAM contents to disk and then does a complete shutdown). From that point on that won't drain the battery at all.

lil0r, 2024-06-09:
thanks @CommonLoon102 ,
my windows machine does do the same and the difference between suspend to RAM and suspend to DISK is clear. It boots and resumes. Whatever magic MACs may be doing it sleeps for weeks and  instantly wakes up.
The issue is that the starbook consumes over 800mAh in S3. So I cannot leave it or transport it really in the current configuration.

Sean-StarLabs, 2024-06-09:
> sure

No dice 

>  in the current configuration

Enable S4, or make Linux put it in S4 after a couple of hours like Windows. We can't do anything to help as consumption in S3 is already below average.

lil0r, 2024-06-09:
I would expect power consumption in S3 to be at least half of what it is.
[image]

lil0r, 2024-06-09:
btw booting from cold to grub is quite slow, any chance to speed this up? Or is this another quirk we AMD guys have to live with?

Sean-StarLabs, 2024-06-09:
> I would expect power consumption in S3 to be at least half of what it is

Why? That's not realistic for the hardware.

> btw booting from cold to grub is quite slow, any chance to speed this up

Next one does speed a bit

lil0r, 2024-06-10:
Above statistic is from a Lenovo Thinkpad P14s with similar specs, so would expect better not worse performance.

Sean-StarLabs, 2024-06-10:
From what I've gleamed from the internet, I'd say the P14s is drastically different save for the processor and display size. Either way, the StarBook is running the best it can - nothing we can improve on.

closed 2024-06-10 by Sean-StarLabs