StarLabsLtd/firmware #159: Cannot update firmware on starbook mkVI

r-vdp, 2024-05-14:
I ran `fwupdmgr refresh; fwugdmgr get-updates; fwupdmgr update` but it fails with the following output:
```
device version not updated on success, 24.03 != 24.02
```

The full report is this:
```
                 {
                           "ReportVersion" : 2,
                           "MachineId" : "a5e9a65b9760ef7db737912add43ceb396ae9a1db4a0af0d57a0c0e6f4b75058",
                           "Metadata" : {
                             "DistroId" : "nixos",
                             "DistroVersion" : "24.05"
                           },
                           "Reports" : [
                             {
                               "Checksum" : "03ab13e9e266a361eedfbad8b4be3e3d1d5bdda2",
                               "ReleaseId" : null,
                               "Protocol" : "org.flashrom",
                               "UpdateState" : 3,
                               "InstallDuration" : 28,
                               "Guid" : [
                                 "7227cf32-bb0f-5894-8d66-c91a43c10eae"
                               ],
                               "Plugin" : "flashrom",
                               "VersionOld" : "24.02",
                               "VersionNew" : "24.03",
                               "Flags" : 549860802603,
                               "Created" : 1715677133,
                               "Modified" : 1715677273,
                               "Metadata" : {
                                 "HostEnclosureKind" : "9",
                                 "HostSku" : "B6-I",
                                 "DistroVersion" : "24.05",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.9.19",
                                 "RuntimeVersion(org.kernel)" : "6.9.0",
                                 "CpuArchitecture" : "x86_64",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.4.9",
                                 "HostFamily" : "B6-I",
                                 "HostBiosVersion" : "24.02",
                                 "HostVendor" : "Star Labs",
                                 "FwupdSupported" : "True",
                                 "PowerState" : "ac-charging",
                                 "KernelName" : "Linux",
                                 "DistroPrettyName" : "NixOS 24.05 (Uakari)",
                                 "BatteryLevel" : "50",
                                 "SystemIntegrityOld" : "UEFI:BootOrder=c8767cb2ce42263788b9b549e146e00ab1efbdf6da9b06bd311dca832caf79da\nUEFI:Boot0001=e19705946a08a2305f8fc84dae26f1704c56104cd15d1b135a9d6272cb2f3ff3\nUEFI:Boot0000=88047d5437636859eee3716e2ca0851eab4cb4615c84a6e2b57c694bc084cf1d\nUEFI:BootCurrent=9b4fb24edd6d1d8830e272398263cdbf026b97392cc35387b991dc0248a628f9\nUEFI:Boot0004=6f40a6eb59891b8e5a1aae03ca82871c997ac5eaccb7bae68013103f68e65210\nUEFI:OsIndicationsS
upported=14e4b8ac16877a5d25661d84f8558c1481bf795a165d92feee46191b3224cde5\nUEFI:Boot0003=73cc6ec78694814a76bfacf4ceb34bcc977ac1862c77d6c2326077de6637bac8\nACPI:TPM2=be1bb447921719c65ed9e3bd78c237f94622683308480c923b97e8f92a65e88b",
                                 "RuntimeVersion(com.hughsie.libjcat)" : "0.2.1",
                                 "DistroName" : "NixOS",
                                 "LidState" : "open",
                                 "RuntimeVersion(com.hughsie.libxmlb)" : "0.3.19",
                                 "KernelVersion" : "6.9.0",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.4.9",
                                 "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.6",
                                 "HostFirmwareMajorRelease" : "18",
                                 "HostProduct" : "StarBook",
                                 "BootTime" : "1715675525",
                                 "BatteryThreshold" : "30",
                                 "CpuModel" : "Intel 12th Gen Intel Core™ i7-1260P",
                                 "HostBaseboardManufacturer" : "Star Labs",
                                 "HostFirmwareMinorRelease" : "02",
                                 "HostBiosMajorRelease" : "00",
                                 "HostBiosVendor" : "coreboot",
                                 "KernelCmdline" : "mem_sleep_default=deep init=/nix/store/5wwrym3k4k9z3mva4xw2g3kmjbbr3rc3-nixos-system-starbook-20240512.2057814.20240513.a8155c2/init i915.enable_fbc=1 i915.enable_psr=2 iomem=relaxed video=DP-1:3840x2160@60 nvme.noacpi=1",
                                 "CompileVersion(com.hughsie.libjcat)" : "0.2.1",
                                 "HostBaseboardProduct" : "StarBook",
                                 "CompileVersion(com.hughsie.libxmlb)" : "0.3.19",
                                 "DistroId" : "nixos",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.9.19",
                                 "DisplayState" : "connected",
                                 "HostBiosMinorRelease" : "00"
                               }
                             }
                           ]
                         }
```

Sean-StarLabs, 2024-05-14:
NixOSs fwupd isn't built with flashrom support, you'll need to either build it yourself or use another distro.

r-vdp, 2024-05-14:
I did build it myself with flashrom support, it has worked without issues before.

I just rebooted and in the firmware UI, it reports the firmware as 24.03 and the EC as 24.02. Is that expected?

Sean-StarLabs, 2024-05-14:
Yeah, likely just hasn't had a chance to do the EC yet

closed 2024-05-14 by Sean-StarLabs