StarLabsLtd/firmware #160: Starbook mkVI stopped charging over usb C

r-vdp, 2024-05-14:
Suddenly my starbook mkVI stopped charging over usb C.

The laptop reports that it's charging but it doesn't draw any power and the battery percentage keeps on going down.

The charging LED flashes multiple times when I plug a cable.

I don't have the barrel charger with me to test currently.

The laptop will intermittently charge for a bit, and then stops again. It is plugged into the starbook power brick.

What can I do? This is obviously a major issue.

Sean-StarLabs, 2024-05-14:
It's calibrating; either let is run to empty, or update the firmware as the latest build has an option to disable the calibration

r-vdp, 2024-05-14:
I just rebooted and set the LEARN setting to disabled, but it's still discharging.

The EC did not get updated though, maybe that's related?

r-vdp, 2024-05-14:
@Sean-StarLabs the battery ran down to 0%, the laptop shut off, and now it doesn't turn on anymore. No LEDs are on either, and the power button doesn't do anything. 

Is this normal? How can I get the laptop to turn on again?

Sean-StarLabs, 2024-05-14:
With the charger in or out?

r-vdp, 2024-05-14:
> With the charger in or out?

In both cases there's no response from the laptop. Black screen and no LEDs.

r-vdp, 2024-05-14:
Any idea? My laptop is bricked now.

divico, 2024-05-14:
> Any idea? My laptop is bricked now.

It is most likely not bricked and will start just fine if you plug in the barrel jack. I had the same issue #153 which is solved by disabling LEARN on latest firmware.

r-vdp, 2024-05-14:
> It is most likely not bricked and will start just fine if you plug in the barrel jack. I had the same issue #153 which is solved by disabling LEARN on latest firmware.

Ok. Well it's effectively bricked right now because I won't have access to the barrel jack cable until beginning next week. 

This is really very annoying.

Sean-StarLabs, 2024-05-14:
> I had the same issue https://github.com/StarLabsLtd/firmware/issues/153

No LEDs isn't the same,,,

So, updated to 24.03, did you shut down after installing or rebooting the BIOS setup?
After that, the battery ran down to 0%. Was that with or without the charger connected?
Which charger and cable are you using?

r-vdp, 2024-05-14:
I rebooted after the firmware flash and went into the firmware setup to reapply my settings. 

Then it ran the battery down with the charger connected. It's the starbook charger that came with the laptop, connected with a good quality usb cable. 

Once the battery was empty, the laptop turned off and since then I can't turn it on anymore.

Sean-StarLabs, 2024-05-14:
Not shutting down means a bunch of things don't get reset, so when the BIOS you just flashed starts for the first time - it's reading its config from junk data - very unlikely that it'll work.

All you can really "try" is disconnecting the battery, leaving it for about 15 minutes, and then reconnecting it. If that doesn't work, the BIOS region will need to be reprogrammed with a hardware programmer. If you need help with that, ping support as there a few ways to play it.

closed 2024-05-15 by Sean-StarLabs

r-vdp, 2024-05-15:
> Closed > #160 <https://github.com/StarLabsLtd/firmware/issues/160>>  as completed.

Hi Sean

Could you explain me a bit how the procedure goes so that I can estimate whether I can do this myself or not? 

Thanks, 
Ramses

Sean-StarLabs, 2024-05-16:
Pretty much the same as this: https://support.starlabs.systems/kb/guides/installing-slash-recovering-firmware-with-external-programmer

Only real difference is that you can't use a clip, it has to be a 12-pin FPC.