StarLabsLtd/firmware #161: I can't update my current coreboot by fwupd

stalkerg, 2024-05-17:
In the log I can see:
```
 5月 17 15:36:43 cachyos fwupd[23140]: 06:36:43.239 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
```

and in stdout:
```
device version not updated on success, 24.04 != 8.95
```

It's a little bit scary to reboot the laptop now.

stalkerg, 2024-05-17:
It's arch linux (cachyos flavor).
I just run `fwupd update`. 

This is actual update log:
```
07:08:43.423 FuEngine             resetting update motd timeout
07:08:43.434 FuMain               Daemon ready for requests (locale ru_RU.UTF-8)
07:08:48.955 FuPluginLinuxLockdown failed to ensure attribute fix flags: missing executable grubby in PATH
07:08:48.973 FuPluginTpm          failed to get eventlog reconstruction: no event log data
07:08:48.978 FuEngine             skipping writing HSI attrs to database as unchanged
07:08:56.703 FuDeviceList         no replug or re-enumerate required
07:08:56.927 FuCabinet            verified metadata B62-I.coreboot.metainfo.xml: 3
07:08:56.927 FuCabinet            processing file: B62-I.coreboot.metainfo.xml
07:08:56.927 FuCabinet            processing release: 24.04
07:08:57.217 FuCabinet            verified payload 24.04.rom: 3
07:08:57.310 FuEngine             composite update 1: dbee8bd3b1ae0316ad143336155651eedb495a0e 8.95->24.04 (bios, order:0: priority:0)
07:08:57.310 FuEngine             install phase now composite-prepare
07:08:57.310 FuDeviceList         no replug or re-enumerate required
07:08:57.374 FuEngine             install phase now prepare
07:08:57.374 FuDeviceList         no replug or re-enumerate required
07:08:57.385 FuEngine             prepare -> FuFlashromDevice:
  DeviceId:             dbee8bd3b1ae0316ad143336155651eedb495a0e
  Name:                 StarBook (bios)
  Status:               device-busy
  Guid:                 7227cf32-bb0f-5894-8d66-c91a43c10eae ← FLASHROM\VENDOR_Star Labs&PRODUCT_StarBook&REGION_BIOS
  Guid:                 12de99d7-d328-5bc2-b391-64eab51edea4 ← FLASHROM\GUID_12122d1c-c383-5583-9cb7-3ba8d220913d
  Guid:                 29ce8545-a416-5f70-8735-a2f6e795a3ea ← Star Labs&B62-I&StarBook&B62-I&Star Labs&StarBook
  Guid:                 0d61c51d-51db-58cf-8901-61a3c2803dcd ← Star Labs&B62-I&StarBook&B62-I
  Guid:                 c6bfd6ae-2931-5711-88c1-2923b0141cb7 ← Star Labs&B62-I&StarBook
  Guid:                 b57d4d60-b1af-584d-81b3-c3eb7d9e60b5 ← Star Labs&B62-I&Star Labs&StarBook
  Guid:                 595c5861-a105-509b-8dd6-f77070345286 ← Star Labs&B62-I&StarBook&B62-I&coreboot
  Branch:               coreboot
  Plugin:               flashrom
  Protocol:             org.flashrom
  Flags:                internal|require-ac|supported|registered|needs-shutdown|can-verify|can-verify-image|updatable-hidden
  Problems:             update-in-progress
  Vendor:               Star Labs
  VendorId:             DMI:coreboot
  Version:              8.95
  VersionFormat:        pair
  Icon:                 computer
  Created:              2024-05-17
  Modified:             2024-05-17
  UpdateError:          An update is in progress
  PhysicalId:           flashrom
  LogicalId:            bios
  AcquiesceDelay:       2500
  CustomFlags:          reset-cmos
  FirmwareSizeMax:      33554432
  Order:                0
  PciBcrAddr:           0
  InternalFlags:        ensure-semver|md-set-signed|md-set-flags|update-pending|host-firmware
  PrivateFlags:         reset-cmos
  Inhibit:              [update-in-progress] An update is in progress

07:08:57.385 FuPluginFlashrom     Reading ich descriptor...
07:08:57.388 FuPluginFlashrom     done.
07:08:57.390 FuDeviceList         no replug or re-enumerate required
07:08:57.390 FuDeviceList         no replug or re-enumerate required
07:08:57.390 FuEngine             install phase now detach
07:08:57.390 FuDeviceList         no replug or re-enumerate required
07:08:57.390 FuEngine             detach -> FuFlashromDevice:
  DeviceId:             dbee8bd3b1ae0316ad143336155651eedb495a0e
  Name:                 StarBook (bios)
  Status:               device-restart
  Percentage:           1
  Guid:                 7227cf32-bb0f-5894-8d66-c91a43c10eae ← FLASHROM\VENDOR_Star Labs&PRODUCT_StarBook&REGION_BIOS
  Guid:                 12de99d7-d328-5bc2-b391-64eab51edea4 ← FLASHROM\GUID_12122d1c-c383-5583-9cb7-3ba8d220913d
  Guid:                 29ce8545-a416-5f70-8735-a2f6e795a3ea ← Star Labs&B62-I&StarBook&B62-I&Star Labs&StarBook
  Guid:                 0d61c51d-51db-58cf-8901-61a3c2803dcd ← Star Labs&B62-I&StarBook&B62-I
[... 210 lines cut ...]

fwupd versions:
```
fwupdmgr --version                                                                                                                                       127 
compile   org.freedesktop.fwupd         1.9.19
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.9.0-3-cachyos
runtime   org.freedesktop.fwupd         1.9.19
```

```
flashrom --version
flashrom v1.3.0 on Linux 6.9.0-3-cachyos (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
```

Sean-StarLabs, 2024-05-17:
I would assume `update-in-progress` is because you've already run `fwupdmgr update` and its waiting to be shutdown to complete the update

stalkerg, 2024-05-17:
Should I  just reboot?

Sean-StarLabs, 2024-05-17:
Definately not. Shutdown.

stalkerg, 2024-05-17:
Thanks! It's working now and shows the proper version. Hmm, I think Fwupd should show a different message in such a case; it is very confusing. Also, it seems like Fwupda now constantly asks me to send a report about an update error. 

Thank you for your job and a good laptop!

closed 2024-05-17 by stalkerg