StarLabsLtd/firmware #162: [StarLite Mk IV] Battery not recognised

CommonLoon102, 2024-05-28:
The OS (Ubuntu 24.04) cannot detect any battery, it doesn't indicate anything.
`upower -i /org/freedesktop/UPower/devices/battery_BAT0` gives only `(null)`. There is no battery info in the settings either.
```
native-path: (null)
power supply: no
updated: Thu 01 Jan 1970 01:00:00 AM CET (1716922293 seconds ago)
has history: no
has statistics: no
unknown
  warning-level: unknown
  battery-level: unknown
  percentage: 0% (should be ignored)
  icon-name: '(null)'
```

If I open the BIOS on boot and quit without any change, then battery indication works correctly. But only if I enter the BIOS on boot. If I don't, the OS doesn't know about the battery at all. I had the same issue on Ubuntu 22.
I haven't used the device for a while (months). I don't remember having this issue before. I haven't updated or changed the BIOS, neither the OS. Then I upgraded to Ubuntu 24, but the issue is the same.

OS: Ubuntu 24.04

BIOS Version: 1.0.0
EC Version: 1.04
BIOS Vendor: American Megatrends

Sean-StarLabs, 2024-05-28:
We're no longer releasing updates for the AMI firmware, so I'd recommend installing coreboot.

CommonLoon102, 2024-05-28:
Thanks for the quick reply. Do I have the latest firmware for AMI then?

Sean-StarLabs, 2024-05-28:
Yes

CommonLoon102, 2024-05-28:
Thanks. I found it in the meantime as well. There was only one initial version.
https://github.com/StarLabsLtd/firmware/tree/master/StarLite/MkIV/AMI/1.0.0
Then I will just open the BIOS every time I boot...

Sean-StarLabs, 2024-05-28:
Out of interest, why do you not want to use coreboot?

CommonLoon102, 2024-05-28:
Because what if it is even worse? I have a workaround I can live with. I don't take unnecessary risks :)

Sean-StarLabs, 2024-05-28:
Haha, it's most definately not - but you can always switch back!

CommonLoon102, 2024-05-28:
Maybe on one rainy Sunday evening I will give it a try.

CommonLoon102, 2024-06-02:
Although it is not raining, but a Saturday evening...
So I have realised that my workaround was just random luck. It doesn't matter if I enter the BIOS or anything.
I decided to give coreboot a try, but after `fwupdmgr switch-branch` I only get `no devices`. What should I do?

Sean-StarLabs, 2024-06-02:
> Can this be a problem? I have the disk encrypted with LUKS.

No

Assuming requirements are met, would need to see the output of `sudo fwupdtool get-devices --plugins flashrom --verbose`

CommonLoon102, 2024-06-02:
Inspecting the output of `sudo fwupdtool get-devices --plugins flashrom --verbose` gave me the idea to plug in the charger. That was the issue. Now the flashing is in progress. Thanks.

CommonLoon102, 2024-06-02:
Now it is running on coreboot. I restarted 10 times and shows the battery percentage 10 out of 10. Closing this. Thanks again.

closed 2024-06-02 by CommonLoon102