StarLabsLtd/firmware #164: [StarLite Mk IV] Missing coreboot options?

CommonLoon102, 2024-06-02:
Since I have switched to coreboot, I wanted to install the Coreboot Configurator, but it has been retired.
https://support.starlabs.systems/kb/guides/coreboot-configurator

Specifically, I wanted two things. Turn off the microphone and limit the max charge to 90%.
Both options are missing from the BIOS.

I suppose turning off the webcam would turn off the microphone as well? Because for input device I see "Microphone - USB 2.0 Camera". OK, so I tried it, the answer is yes.

What I still miss though is limiting the max charge. Is there a reason why the option is not showing up in the BIOS?

Sean-StarLabs, 2024-06-03:
The Lite Mk IV doesn't have an option to limit the charge level

CommonLoon102, 2024-06-03:
Is this a hardware limitation?

closed 2024-06-03 by CommonLoon102