StarLabsLtd/firmware #165: [StarBook Mk V] Unable to update firmware

cyb3rmonk, 2024-06-09:
Hi,

I see that there's a firmware update available for my StarBook, but I'm unable to do the update using fwupdmgr. Here's the

Here's the firmware I have:
```
# fwupdmgr get-devices
├─System Firmware:
│ │   Device ID:          c738f5c2f089f407525d70d40c11ec7dbeb1e687
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.2.0
│ │   Minimum Version:    1.2.0
│ │   Vendor:             Star Labs (DMI:American Megatrends International, LLC.)
│ │   Update State:       Success
│ │   GUID:               5a29ddbb-11dd-476d-a641-a45c1db63d5c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
(...)
```

And here's what happens when I try to update:
```
# fwupdmgr refresh --force && fwupdmgr update
Updating lvfs
Downloading…
Successfully downloaded new metadata: 2 local devices supported
Devices with no available firmware updates: 
 • Star Drive PCIe SSD
╔════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.2.0 to 24.3.0?                  ║
╠════════════════════════════════════════════════════════════════╣
║ This release includes the following features:                  ║
║                                                                ║
║ • Fixed an issue that stopped the backlight working after S3   ║
║ • Enabled support for PD 2.0 chargers                          ║
║ • Disable USB power in S3                                      ║
║ • Improved battery reporting                                   ║
║                                                                ║
║ StarBook must remain plugged into a power source for the       ║
║ duration of the update to avoid damage.                        ║
╚════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]: y
```

Then I reboot and the firmware is still 1.2.0
I see this dmesg log that seems to be related:
```
[   15.913653] systemd[1]: Starting Firmware update daemon...
[   15.994985] fwupd[1944]: 03:12:52.955 FuPluginFlashrom     FCH device found but SMBus revision 0x51 does not match known values.
[   15.994985] fwupd[1944]: Please report this to flashrom@flashrom.org and include this log and
[   15.994985] fwupd[1944]: the output of lspci -nnvx, thanks!.
[   15.994985] fwupd[1944]: 03:12:52.955 FuPluginFlashrom     Could not determine chipset generation.
[   15.995368] fwupd[1944]: 03:12:52.955 FuPluginFlashrom     Laptops, notebooks and netbooks are difficult to support and we
[   15.995368] fwupd[1944]: recommend to use the vendor flashing utility. The embedded controller
[   15.995368] fwupd[1944]: (EC) in these machines often interacts badly with flashing.
[   15.995368] fwupd[1944]: See the manpage and https://flashrom.org/Laptops for details.
[   15.995368] fwupd[1944]: If flash is shared with the EC, erase is guaranteed to brick your laptop
[   15.995368] fwupd[1944]: and write may brick your laptop.
[   15.995368] fwupd[1944]: Read and probe may irritate your EC and cause fan failure, backlight
[   15.995368] fwupd[1944]: failure and sudden poweroff.
[   15.995368] fwupd[1944]: You have been warned.
```

cyb3rmonk, 2024-06-09:
Extra info:

```
$ lspci -nnvx
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne Root Complex [1022:1630]
	Subsystem: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne Root Complex [1022:1630]
	Flags: fast devsel
00: 22 10 30 16 00 00 00 00 00 00 00 06 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 22 10 30 16
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne IOMMU [1022:1631]
	Subsystem: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne IOMMU [1022:1631]
	Flags: fast devsel, IRQ 255
	Capabilities: <access denied>
00: 22 10 31 16 00 00 10 00 00 00 06 08 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 22 10 31 16
30: 00 00 00 00 40 00 00 00 00 00 00 00 ff 01 00 00

00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge [1022:1632]
	Flags: fast devsel, IOMMU group 0
00: 22 10 32 16 00 00 00 00 00 00 00 06 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge [1022:1632]
	Flags: fast devsel, IOMMU group 1
00: 22 10 32 16 00 00 00 00 00 00 00 06 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

00:02.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge [1022:1634] (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
	Flags: bus master, fast devsel, latency 0, IRQ 28, IOMMU group 2
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: fcf00000-fcffffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: [disabled] [64-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport
00: 22 10 34 16 07 04 10 00 00 00 04 06 10 00 81 00
10: 00 00 00 00 00 00 00 00 00 01 01 00 f1 01 00 00
20: f0 fc f0 fc f1 ff 01 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 50 00 00 00 00 00 00 00 ff 00 12 00

00:02.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge [1022:1634] (prog-if 00 [Normal decode])
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
	Flags: bus master, fast devsel, latency 0, IRQ 29, IOMMU group 3
	Bus: primary=00, secondary=02, subordinate=02, sec-latency=0
	I/O behind bridge: [disabled] [32-bit]
	Memory behind bridge: fce00000-fcefffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: [disabled] [64-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport
00: 22 10 34 16 07 04 10 00 00 00 04 06 10 00 81 00
10: 00 00 00 00 00 00 00 00 00 02 02 00 f1 01 00 00
20: e0 fc e0 fc f1 ff 01 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 50 00 00 00 00 00 00 00 ff 00 12 00

00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge [1022:1632]
	Flags: fast devsel, IOMMU group 4
00: 22 10 32 16 00 00 00 00 00 00 00 06 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

[... 276 lines cut ...]
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:            TPM 2.0 Device
│     Current version:    3.73.0.5
│     Vendor:             Advanced Micro Devices, Inc. (TPM:AMD)
│     GUIDs:              9305de1c-1e12-5665-81c4-37f8e51219b8 ← TPM\VEN_AMD&DEV_0001
│                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ← TPM\VEN_AMD&MOD_AMD
│                         65a3fced-b423-563f-8098-bf5c329fc063 ← TPM\VEN_AMD&DEV_0001&VER_2.0
│                         5e704f0d-83cb-5364-8384-f46d725a23b8 ← TPM\VEN_AMD&MOD_AMD&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│   
```
```

Sean-StarLabs, 2024-06-09:
>  So I'm guessing either...

So, fwupd does a lot of things - flashrom is only used when switching branch, or updating coreboot. What you're trying to do is a capsule update, so anything flashrom related can be ignored.. That said, the stuff you're seeing is because the version of flashrom you've got installed is too old for the processor.

For capsule updates, all it's doing is copying a capsule to the EFI partition and then booting an EFI binary that'll install it (it'll show in the Boot Menu as "Linux Firmware Updater").

There are too many variables in Arch to say where to look; if you want to dig into it, I'd suggest scheduling the update and then rebooting manually and selecting the "Linux Firmware Updater" to see if it's an issue with it, just not being called. If it is being called, then I've only ever seen it be tripped up by some file on the EFI partition that confuses things - cautiously removing things that aren't essential is the first step, and failing that, you have to pul logs for the EFI binary.

Best bet, just update it from the EFI Shell - https://support.starlabs.systems/kb/firmware/efi-shell

cyb3rmonk, 2024-06-10:
> So, fwupd does a lot of things - flashrom is only used when switching branch, or updating coreboot. What you're trying to do is a capsule update, so anything flashrom related can be ignored..

didn't know, thanks for the info!

> Then I reboot and the firmware is still 1.2.0

FIXED - I was able to install the latest version just now. Turns out the problem was my system - I have a non-standard `/boot` and `/esp` setup, so fwupdmgr was writing `fwupdx64.efi` and `fwupd-xxx.cap` files outside `/esp` partition.

The "0x51" error threw me off because it seemed related, and OP from #65 had the same message so I thought it indicated the problem was not with my install.

... so anyway sorry for wasting your time

> That said, the stuff you're seeing is because the version of flashrom you've got installed is too old for the processor.

For the record, I checked and I'm using the latest version of fwupd and flashrom. Re-reading the logs, it seems like fwupd daemon is checking capabilities of plugins when it starts, and the SMBus thing is something the flashrom plugin sees as unsupported.

closed 2024-06-10 by cyb3rmonk

Sean-StarLabs, 2024-06-10:
> I'm using the latest version of fwupd and flashrom. Re-reading the logs, it seems like fwupd daemon is checking capabilities of plugins when it starts, and the SMBus thing is something the flashrom plugin sees as unsupported.

It's because flashrom is outdated - just because it's arch, doesn't mean it's the latest ;)