StarLabsLtd/firmware #170: [Starbook IV - AMD] Can't Install Coreboot / No Devices (Star Drive SATA SSD)

pelter, 2024-06-14:
On a StarLite IV, looking to change from AMD to Coreboot.

- StarLite IV
- Ubuntu 24.04

```bash
uname -a
Linux dkeSL 6.8.0-35-generic #35-Ubuntu SMP PREEMPT_DYNAMIC Mon May 20 15:51:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

Prerequisites run OK:

- [X] BIOS lock disabled.

```bash
sudo add-apt-repository ppa:starlabs/main
sudo add-apt-repository universe
sudo apt update
sudo apt install fwupd libflashrom1
```

```bash
systemctl status fwupd
● fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Fri 2024-06-14 14:22:22 BST; 2min 54s ago
       Docs: https://fwupd.org/
   Main PID: 9419 (fwupd)
      Tasks: 6 (limit: 9218)
     Memory: 8.5M (peak: 9.7M)
        CPU: 1.188s
     CGroup: /system.slice/fwupd.service
             └─9419 /usr/libexec/fwupd/fwup
```

```bash
fwupdmgr --version
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         2.0.0
runtime   org.kernel                    6.8.0-35-generic
```

```bash
fwupdmgr refresh --force
updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 2 local devices supported
```

```bash
fwupdmgr switch-branch
no devices
```

```bash
fwupdmgr update
Devices with the latest available firmware version:
 • System Firmware
 • UEFI dbx
Devices with no available firmware updates: 
 • Star Drive SATA SSD
```

```bash
fwupdmgr refresh --force && fwupdmgr update
Updating lvfs
Downloading…             [     \                                 ]
Successfully downloaded new metadata: 2 local devices supported
Devices with the latest available firmware version:
 • System Firmware
 • UEFI dbx
Devices with no available firmware updates: 
 • Star Drive SATA SSD
```

```bash
fwupdmgr get-devices
Star Labs StarLite
│
├─GeminiLake [UHD Graphics 605]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              6f398ef1-6999-553b-822a-e5798e7d3809 ← PCI\VEN_8086&DEV_3184
│                         a5316ef3-61f0-5047-a470-25fd2bcfe359 ← PCI\VEN_8086&DEV_3184&SUBSYS_80862212
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Pentium Silver N5030 CPU @ 1.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000024
│     Vendor:             Intel
│     GUIDs:              54817722-b948-55a7-8324-3d4736e90181 ← CPUID\PRO_0&FAM_06&MOD_7A
│                         f02ec022-4f79-53c2-a68f-08ea48563cb4 ← CPUID\PRO_0&FAM_06&MOD_7A&STP_8
│     Device Flags:       • Internal device
│   
├─Star Drive SATA SSD:
│     Device ID:          c46944fe84f2b057692258cc87a812cc86364881
│     Summary:            ATA drive
│     Current version:    SBFMX1.5
│     Vendor:             Seagate (ATA:0x1BB1)
│     Serial Number:      A46207231E8900001411
│     GUIDs:              76fae88f-f1c3-5816-a6b8-c86c967e0fd2 ← IDE\Star_Drive_SATA_SSD_____________________SBFMX1.5
│                         0966c9ae-3ad1-57f1-a328-3cee48807461 ← IDE\0Star_Drive_SATA_SSD_____________________
│                         1bd403d3-8f7e-5dca-b0c1-6535d48f4e85 ← Star Drive SATA SSD
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          7e9f6fea6862e5ba4d994bf493952b63072af6c0
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.0.0
│ │   Minimum Version:    1.0.0
│ │   Vendor:             Star Labs (DMI:American Megatrends Inc.)
│ │   Update State:       Success
│ │   GUID:               b1ff7b4d-8b6a-4f50-a047-cddfdb75fa28
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUID:             f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            TPM 2.0 Device
      Current version:    403.0.0.0
      Vendor:             Intel (TPM:INTC)
      GUIDs:              34801700-3a50-5b05-820c-fe14580e4c2d ← TPM\VEN_INTC&DEV_0000
                          8e1cbc5d-5a11-5149-bfea-b6065d5296ba ← TPM\VEN_INTC&MOD_Intel
                          03f304f4-223e-54f4-b2c1-c3cf3b5817c6 ← TPM\VEN_INTC&DEV_0000&VER_2.0
                          52d7b679-db28-5bf7-bd87-41d77aeec600 ← TPM\VEN_INTC&MOD_Intel&VER_2.0
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

Sean-StarLabs, 2024-06-14:
Can you post:
```
sudo fwupdtool get-devices --plugins flashrom --verbose
fwupdmgr get-devices --show-all
apt list --installed | grep fwupd
````

pelter, 2024-06-15:
```bash
sudo fwupdtool get-devices --plugins flashrom --verbose
```

```bash
07:04:36.229 FuDebug              verbose to info (on console 1)
07:04:36.233 FuEngine             starting fwupd 2.0.0…
07:04:36.258 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]07:04:36.265 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
07:04:36.266 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
07:04:36.266 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
07:04:36.267 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
07:04:36.268 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]07:04:36.274 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]07:04:36.308 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
07:04:36.309 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]07:04:36.313 FuContext            SMBIOS Manufacturer=Star Labs
07:04:36.313 FuContext            SMBIOS EnclosureKind=a
07:04:36.313 FuContext            SMBIOS Family=I4
07:04:36.313 FuContext            SMBIOS ProductName=StarLite
07:04:36.313 FuContext            SMBIOS ProductSku=I4
07:04:36.313 FuContext            SMBIOS BiosVendor=American Megatrends Inc.
07:04:36.313 FuContext            SMBIOS BiosVersion=1.0.0
07:04:36.313 FuContext            SMBIOS BiosMajorRelease=05
07:04:36.313 FuContext            SMBIOS BiosMinorRelease=0d
07:04:36.313 FuContext            SMBIOS FirmwareMajorRelease=01
07:04:36.313 FuContext            SMBIOS FirmwareMinorRelease=00
07:04:36.313 FuContext            SMBIOS BaseboardManufacturer=Star Labs
07:04:36.313 FuContext            SMBIOS BaseboardProduct=StarLite
07:04:36.313 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
07:04:36.314 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]07:04:36.314 FuEngine             loading metadata for remote 'vendor-directory'
07:04:36.314 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
07:04:36.315 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
07:04:36.315 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
07:04:36.332 FuEngine             2638 components now in silo
Loading…                 [***                                    ]07:04:36.371 FuEngine             plugins disabled: modem_manager, uefi_capsule, acpi_dmar, acpi_facp, acpi_ivrs, acpi_phat, algoltek_aux, algoltek_usb, amd_pmc, amd_gpu, analogix, android_boot, ata, audio_s5gen2, aver_hid, bcm57xx, bios, ccgx, ccgx_dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros_ec, dell, dell_dock, dfu, dfu_csr, ebitdo, elantp, elanfp, emmc, ep963x, fastboot, focalfp, fpc, fresco_pd, genesys, genesys_gl32xx, goodixmoc, goodixtp, gpio, hailuck, igsc, intel_me, intel_usb4, iommu, jabra, jabra_gnp, kinetic_dp, lenovo_thinklmi, linux_display, linux_lockdown, linux_sleep, linux_swap, linux_tainted, logind, logitech_hidpp, logitech_bulkcontroller, logitech_rallysystem, logitech_scribe, logitech_tap, mediatek_scaler, msr, mtd, nitrokey, nordic_hid, nvme, optionrom, parade_lspcon, pci_bcr, pci_mei, pci_psp, pixart_rf, qsi_dock, realtek_mst, redfish, rts54hid, rts54hub, steelseries, scsi, synaptics_cape, synaptics_cxaudio, synaptics_mst, synaptics_prometheus, synaptics_rmi, synaptics_vmm9, system76_launch, test, test_ble, thelio_io, thunderbolt, ti_tps6598x, tpm, uefi_dbx, uefi_esrt, uefi_pk, uefi_recovery, uf2, upower, usi_dock, vbe, vli, wacom_raw, wacom_usb, wistron_dock
Loading…                 [***                                    ]07:04:36.371 FuContext            battery threshold now 30
Loading…                 [ -                                     ]07:04:36.372 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
Loading…                 [************************************** ]07:04:36.401 FuEngine             FuUsbBackend:
  Name:                 usb
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
DoneColdplug:           true
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false

07:04:36.402 FuEngine             resetting update motd timeout

No detected devices
07:04:36.402 FuMain               No detected devices

dke@dkeSL:~/Dropbox$ sudo fwupdtool get-devices --plugins flashrom --verbose > star1.txt
07:04:46.140 FuDebug              verbose to info (on console 1)
07:04:46.141 FuEngine             starting fwupd 2.0.0…
07:04:46.152 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
[... cut ...]
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

```bash
apt list --installed | grep fwupd
```

```bash
fwupd-signed/noble,now 1.52+1.4-1 amd64 [installed]
fwupd/noble,now 1.9.21+n amd64 [installed]
libfwupd3/noble,now 1.9.21+n amd64 [installed,automatic]
```

Sean-StarLabs, 2024-06-20:
Can you run:
```
echo -e "[0fc25c8c-ffa8-54ad-a216-d13cfe75bee4]\nPlugin = flashrom" | sudo tee /var/lib/fwupd/quirks.d/flashrom.quirk
sudo systemctl restart fwupd
```

Then try the switch?

pelter, 2024-06-20:
```bash
echo -e "[0fc25c8c-ffa8-54ad-a216-d13cfe75bee4]\nPlugin = flashrom" | sudo tee /var/lib/fwupd/quirks.d/flashrom.quirk

[0fc25c8c-ffa8-54ad-a216-d13cfe75bee4]
Plugin = flashrom
```

```bash
sudo systemctl restart fwupd
```

outputs no message

```bash
fwupdmgr refresh --force

Updating lvfs
Downloading…             [          |                            ]
Successfully downloaded new metadata: 3 local devices supported
```

```bash
fwupdmgr update

Devices with the latest available firmware version:
 • StarLite (bios)
 • System Firmware
 • UEFI dbx
Devices with no available firmware updates: 
 • Star Drive SATA SSD

```

Sean-StarLabs, 2024-06-20:
`fwupd switch-branch` is the command to change branches

pelter, 2024-06-20:
```bash
fwupdmgr switch-branch
````

This has now resolved the issue and the StarLite Mk IV now has Coreboot firmware installed.

Thanks for the support @Sean-StarLabs