StarLabsLtd/firmware #171: [Starbook MK VI - Intel]  Max charging level not always honoured

divico, 2024-06-15:
As noted previously https://github.com/StarLabsLtd/firmware/issues/151#issuecomment-2019908224 , the charge level set in CFR/coreboot is not always honoured. Most of the time it is, and I have not been able to find out when exactly it is not.

Now I have managed to do an EC dump when the battery is at 98% but the max charge is set to 60%, maybe you can find out something.

If not, let me know how I can further help.

Currently running Coreboot/EC `24.04` but I have observed this on previous versions too.

```
ectool -d
EC RAM:

00: 18 04 e7 76 01 00 61 e1 e7 bb dd fd 00 a8 99 01 
10: 01 de d6 9d d4 92 a8 00 b7 df aa 00 00 65 22 cf 
20: 3c f3 ea a1 12 a6 0e f9 c3 54 1e 8c 52 fa 3c dc 
30: 1a 83 11 44 89 97 94 93 42 ae a4 5f 22 70 98 81 
40: 00 30 38 2f 32 37 2f 34 31 00 5d 32 30 32 34 2f 
50: 30 34 2f 32 34 00 7b 63 18 fb ab 39 09 3e 1e d3 
60: c8 2f 5a 18 dc fb 5b f2 f9 5d 14 bf 89 48 42 aa 
70: 3c c0 ce 24 f8 ac ab 1f c4 4c 82 38 f2 b5 75 01 
80: 07 01 24 00 fc 15 1e 2d d6 14 00 00 08 74 00 90 
90: 14 f0 32 63 00 81 00 58 c7 b8 3b 96 a9 54 83 83 
a0: 64 b1 74 ce a5 f5 1d 41 24 cf 3f 65 e7 4c 18 37 
b0: 5d b7 13 24 4f df fd c2 ff e3 39 5e ee 93 56 fc 
c0: c3 d3 51 ae bb ae 24 b6 43 ef 6b 28 01 69 c0 50 
d0: e3 30 96 4b 38 bc 94 57 87 a8 82 46 6d 5d 16 e1 
e0: 60 c3 78 e2 6b 0d 3b 8a f9 af 0a b9 6f a4 2c d8 
f0: 5a c6 d6 90 02 2e b6 6b 80 5b 96 f0 94 d3 4b b2 

```

Sean-StarLabs, 2024-06-15:
If you leave it charging, when does it stop discharging (charging LED will go from purple to red or blue)?

divico, 2024-06-15:
> If you leave it charging, when does it stop discharging (charging LED will go from purple to red or blue)?

Not sure I understood the question. If I leave it charging, is it supposed to discharge at some point?

Meanwhile it reached 100% charge, LED is purple and EC dump is

```
ectool -d
EC RAM:

00: 18 04 e7 76 01 00 61 e1 e7 bb dd fd 00 a8 99 01 
10: 01 de d6 9d d4 92 a8 00 b7 df aa 00 00 65 22 cf 
20: 3c f3 ea a1 12 a6 0e f9 c3 54 1e 8c 52 fa 3c dc 
30: 1a 83 11 44 89 97 94 93 42 ae a4 5f 22 70 98 81 
40: 00 30 38 2f 32 37 2f 34 31 00 5d 32 30 32 34 2f 
50: 30 34 2f 32 34 00 7b 63 18 fb ab 39 09 3e 1e d3 
60: c8 2f 5a 18 dc fb 5b f2 f9 5d 14 bf 89 48 42 aa 
70: 3c c0 ce 24 f8 ac ab 1f c4 4c 82 38 f2 b5 75 01 
80: 07 01 24 00 fc 15 1e 2d 6e 15 00 00 08 0f 00 e4 
90: 14 c1 32 61 00 81 00 58 c7 b8 3b 96 a9 54 83 83 
a0: 64 b1 74 ce a5 f5 1d 41 24 cf 3f 65 e7 4c 18 37 
b0: 5d b7 13 24 4f df fd c2 ff e3 39 5e ee 93 56 fc 
c0: c3 d3 51 ae bb ae 24 b6 43 ef 6b 28 01 69 c0 50 
d0: e3 30 96 4b 38 bc 94 57 87 a8 82 46 6d 5d 16 e1 
e0: 60 c3 78 e2 6b 0d 3b 8a f9 af 0a b9 6f a4 2c d8 
f0: 5a c6 d6 90 02 2e b6 6b 80 5b 96 f0 94 d3 4b b2 

```

Sean-StarLabs, 2024-06-15:
Yeah, so just leave it connected to the charger - see where it goes down to.

divico, 2024-06-16:
It never discharged. LED is still purple

```
ectool -d
EC RAM:

00: 18 04 e7 76 01 00 61 e1 e7 bb dd fd 00 a8 99 01 
10: 01 de d6 9d d4 92 a8 00 b7 df aa 00 00 65 22 cf 
20: 3c f3 ea a1 12 a6 0e f9 c3 54 1e 8c 52 fa 3c dc 
30: 1a 83 11 44 89 97 94 93 42 ae a4 5f 22 70 98 81 
40: 00 30 38 2f 32 37 2f 34 31 00 5d 32 30 32 34 2f 
50: 30 34 2f 32 34 00 7b 63 18 fb ab 39 09 3e 1e d3 
60: c8 2f 5a 18 dc fb 5b f2 f9 5d 14 bf 89 48 42 aa 
70: 3c c0 ce 24 f8 ac ab 1f c4 4c 82 38 f2 b5 75 01 
80: 07 01 24 00 fc 15 1e 2d 6e 15 00 00 08 0f 00 e4 
90: 14 c1 32 61 00 81 00 58 c7 b8 3b 96 a9 54 83 83 
a0: 64 b1 74 ce a5 f5 1d 41 24 cf 3f 65 e7 4c 18 37 
b0: 5d b7 13 24 4f df fd c2 ff e3 39 5e ee 93 56 fc 
c0: c3 d3 51 ae bb ae 24 b6 43 ef 6b 28 01 69 c0 50 
d0: e3 30 96 4b 38 bc 94 57 87 a8 82 46 6d 5d 16 e1 
e0: 60 c3 78 e2 6b 0d 3b 8a f9 af 0a b9 6f a4 2c d8 
f0: 5a c6 d6 90 02 2e b6 6b 80 5b 96 f0 94 d3 4b b2 

```

Sean-StarLabs, 2024-06-16:
Did you just leave it idle, or have you been using it? Dump shows RSOC has gone down by 2%, which could be right - drain will be nominal as it's going to be primarily running of the charger - it'll only hit the battery for hybrid power.

divico, 2024-06-16:
It was idle most of the time

Sean-StarLabs, 2024-06-16:
Might not ever go down in that case, if you're leaving it, a `while :; do :; done` in a terminal should make the battery go down

divico, 2024-06-16:
I don't know what you want to test. If you could tell me more about what the test is about, I could act accordingly.

Sean-StarLabs, 2024-06-16:
Testing if it goes down to 55%, then goes back up to 60%

divico, 2024-06-18:
I will update this thread the next time I encounter the problem. Unfortunately this time I could not wait for it to discharge as this is my daily driver

divico, 2024-07-02:
So it happened again today. The charge level was <8%, possibly <5% but I'm not sure, so I plugged the charger, turned it off and let it charge overnight. The charge level is now 94% even though the max charge level is set to 60%.

EC dump is

```
ectool -d
EC RAM:

00: 18 04 e7 76 01 00 61 f1 c7 bb dd dd 00 a8 9d 01 
10: 01 ca 54 9d 54 9a a0 00 b7 df aa 00 00 61 03 cf 
20: 3c b3 eb a1 92 c7 0e b9 07 55 1e 8c 53 59 3c df 
30: da 83 11 44 89 96 84 bb 42 ae a4 1f 22 70 58 a1 
40: 00 30 38 2f 32 37 2f 34 31 00 59 32 30 32 34 2f 
50: 30 34 2f 32 34 00 7b e1 10 fb ab 19 0d 3f 1e f3 
60: c0 2f da 10 dc fb 5f f2 79 5d 54 be 89 58 42 2a 
70: 34 c8 ce 24 f8 ac 8b 0d c4 44 82 18 f6 b5 7d 01 
80: 07 01 24 00 fc 15 1e 2d 5f 13 00 00 08 00 00 3d 
90: 12 83 32 5e 00 85 00 78 c7 b9 3f 96 b9 58 8b 83 
a0: 64 b0 74 8e a0 b1 1d 41 24 ce 3e 65 e7 4c 1a 37 
b0: 5d b7 03 24 5f df fd c2 ff e3 39 1e ee 93 56 fc 
c0: e3 f3 50 ae b9 ae 04 e7 43 67 6b 2c 01 69 d0 50 
d0: e3 35 96 43 3a ac 9c 57 87 a8 82 46 6c 5d 16 e1 
e0: 60 c3 78 e2 6b 05 3b 8b 78 af 0a b9 7f a4 ac d8 
f0: 7a c6 d6 90 03 2e b6 6b 80 5b 96 f0 d4 d3 4b 90
```

I will now leave the charger plugged in and try to discharge, and I will report back on how low it went before charging again.

divico, 2024-07-02:
It does not seem possible to discharge with the charger plugged in. Even with a constant CPU load of ~90% (`stress -c 14`) it will not go down a single percent per hour.

Sean-StarLabs, 2024-07-02:
It'll only go down when the StarBook uses more power than the battery can provide, or power draw changes a lot. Disconnecting the charger for a bit will do the same thing - I was just leaning towards doing it with the charger connected for you to see it more than anything

divico, 2024-07-02:
Yes, it did when down to ~55% before charging up to ~65%

Sean-StarLabs, 2024-07-03:
Cool - so all good? The 5% (sometimes OS will show 6%) margin is normal

divico, 2024-07-03:
No, it happened several times that it did charge above 90% despite the max charge level set to 60%

Sean-StarLabs, 2024-07-03:
Okay, just ping over an EC dump when you see that next

divico, 2024-07-03:
there are two such EC dump above

https://github.com/StarLabsLtd/firmware/issues/171#issue-2355086958

https://github.com/StarLabsLtd/firmware/issues/171#issuecomment-2202008880

Sean-StarLabs, 2024-07-03:
Neither of those are charging

closed 2024-09-02 by Sean-StarLabs

divico, 2024-11-01:
I'm now able to reproduce.
1. Discharge battery below 5%
2. Shutdown machine
3. Connect charger (USB-C or jack) and wait until the charge is complete.

This will result in a battery charged above the max charge threshold set in the bios.

Sean-StarLabs, 2024-11-04:
Point 1 wouldn't be relevant; if it's charging in G3, coreboot hasn't run so it'll charge to 100%