StarLabsLtd/firmware #174: Support for (Open)SUSE

radekholy24, 2024-06-23:
Hello,

I decided to give a try to SLES (SUSE Linux Enterprise Server) on my StarBook Mk VI. So far so good but `fwupdmgr` does not seem to recognize the firmware which implies that I might not be able to update it via `fwupdmgr`, I guess. Is there anything that can be done on your side (perhaps with my contribution) to support (Open)SUSE or should I rather ask `fwupd` or SUSE or is it just that you do not have enough resources to support it and there is no place where I can request support for it?

Also, would it make sense to update https://support.starlabs.systems/kb/firmware/lvfs-requirements with information regarding this distro?

```shell
$ sudo zypper packages -i | grep 'fwupd
> libflashrom1'
i  | SLE-Module-Desktop-Applications15-SP5-Updates   | fwupd                                          | 1.8.6-150500.4.3.1                        | x86_64
v  | SLE-Module-Desktop-Applications15-SP5-Pool      | fwupd                                          | 1.8.6-150500.2.2                          | x86_64
i  | SLE-Module-Desktop-Applications15-SP5-Updates   | fwupd-bash-completion                          | 1.8.6-150500.4.3.1                        | noarch
v  | SLE-Module-Desktop-Applications15-SP5-Pool      | fwupd-bash-completion                          | 1.8.6-150500.2.2                          | noarch
i  | SLE-Module-Desktop-Applications15-SP5-Pool      | fwupd-efi                                      | 1.2-150400.1.14                           | x86_64
i  | SLE-Module-Desktop-Applications15-SP5-Updates   | fwupd-lang                                     | 1.8.6-150500.4.3.1                        | noarch
v  | SLE-Module-Desktop-Applications15-SP5-Pool      | fwupd-lang                                     | 1.8.6-150500.2.2                          | noarch
i  | SLE-Module-Desktop-Applications15-SP5-Pool      | libflashrom1                                   | 1.2-150500.1.1                            | x86_64
i  | SLE-Module-Desktop-Applications15-SP5-Updates   | libfwupd2                                      | 1.8.6-150500.4.3.1                        | x86_64
v  | SLE-Module-Desktop-Applications15-SP5-Pool      | libfwupd2                                      | 1.8.6-150500.2.2                          | x86_64
$
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.9
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.14.21-150500.55.65-default
runtime   com.hughsie.libjcat           0.1.12
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.3.9
$
$ fwupdmgr -v --show-all get-devices
(fwupdmgr:27212): GLib-DEBUG: 10:19:43.878: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:27212): GLib-GIO-DEBUG: 10:19:43.893: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.893: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.893: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.893: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.893: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.893: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:27212): GLib-GIO-DEBUG: 10:19:43.893: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.895: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.896: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.896: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.896: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:27212): dconf-DEBUG: 10:19:43.896: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:27212): Fwupd-DEBUG: 10:19:43.897: Emitting ::status-changed() [idle]
(fwupdmgr:27212): FuMain-DEBUG: 10:19:43.908: FwupdDevice:
  DeviceId:             7532230c0224920f44faf261297a9432df84be38
  Guid:                 c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
  Plugin:               gpio
  Flags:                registered
  Created:              2024-06-22

(fwupdmgr:27212): FuMain-DEBUG: 10:19:43.908: FwupdDevice:
  DeviceId:             4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  Name:                 13th Gen Intel Core™ i7-1360P
  Guid:                 b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
  Guid:                 30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
  Guid:                 c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
  Guid:                 557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
  Plugin:               cpu
  Flags:                internal|registered
  Vendor:               Intel
  Version:              0x00004121
  VersionFormat:        hex
  VersionRaw:           0x00004121
  Icon:                 computer
  Created:              2024-06-22

(fwupdmgr:27212): FuMain-DEBUG: 10:19:43.908: FwupdDevice:
[... 48 lines cut ...]
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2024-06-22
$
$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
$
$ fwupdmgr -v get-updates
Devices with no available firmware updates: 
 • Star Drive PCIe SSD
No updatable devices
$
$ fwupdmgr switch-branch
No supported devices
$
$ uname -a
Linux radek-holy-laptop-star-labs 5.14.21-150500.55.65-default #1 SMP PREEMPT_DYNAMIC Thu May 23 04:57:11 UTC 2024 (a46829d) x86_64 x86_64 x86_64 GNU/Linux
$
$ cat /sys/class/dmi/id/bios_version
24.04
$ cat /sys/class/dmi/id/ec_firmware_release
24.4
```

Sean-StarLabs, 2024-06-23:
What's the output of:
```
sudo fwupdtool get-devices --plugins flashrom --verbose
```

radekholy24, 2024-06-23:
```shell
$ sudo fwupdtool get-devices --plugins flashrom --verbose
20:34:09:0146 FuDebug              Verbose debugging enabled (on console 1)
20:34:09:0147 FuMain               locked /run/lock/fwupdtool
Loading…                 [-                                      ]20:34:09:0151 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
20:34:09:0151 FuCommon             reading /etc/fwupd/daemon.conf with 2224 bytes
20:34:09:0151 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
20:34:09:0151 FuConfig             using autodetected max archive size 4.3 GB
20:34:09:0151 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]20:34:09:0152 FuRemoteList         loading /usr/share/fwupd/metainfo
20:34:09:0152 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
20:34:09:0152 XbSilo               file: 1ed6cad6-7cc2-21c9-d6d9-ab23a37952c6, current:1ed6cad6-7cc2-21c9-d6d9-ab23a37952c6, cached: (null)
20:34:09:0152 XbSilo               loading silo with file contents
20:34:09:0152 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
20:34:09:0152 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
20:34:09:0152 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
20:34:09:0152 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
20:34:09:0152 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
20:34:09:0152 FuRemoteList         ordering lvfs-testing=lvfs+1
20:34:09:0152 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]20:34:09:0152 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
20:34:09:0152 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:34:09:0152 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
20:34:09:0153 Jcat                 loaded 1 certificates
20:34:09:0153 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
20:34:09:0153 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:34:09:0153 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
20:34:09:0153 Jcat                 loaded 1 certificates
20:34:09:0153 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
20:34:09:0153 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
20:34:09:0153 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]20:34:09:0153 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
20:34:09:0153 FuHistory            got schema version of 8
Loading…                 [*                                      ]20:34:09:0159 FuPlugin             load(/usr/lib64/fwupd-1.8.6/libfu_plugin_modem_manager.so)
20:34:09:0160 FuPluginUefiCapsule  failed to get fwupd-efi runtime version: searching /usr/lib/fwupd/efi/fwupdx64.efi: needle of 0x24 bytes was not found in haystack of 0xf578 bytes
20:34:09:0160 FuContext            added udev subsystem watch of platform
20:34:09:0160 FuContext            added udev subsystem watch of mei
20:34:09:0160 FuContext            added udev subsystem watch of block
20:34:09:0160 FuContext            added udev subsystem watch of pci
20:34:09:0160 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
20:34:09:0160 FuPluginDell         Using libsmbios 2.4
20:34:09:0160 FuContext            added udev subsystem watch of i2c
20:34:09:0160 FuContext            added udev subsystem watch of i2c-dev
20:34:09:0160 FuContext            added udev subsystem watch of hidraw
20:34:09:0160 FuContext            added udev subsystem watch of gpio
20:34:09:0160 FuContext            added udev subsystem watch of iommu
20:34:09:0160 FuContext            added udev subsystem watch of msr
20:34:09:0160 FuContext            added udev subsystem watch of mtd
20:34:09:0161 FuContext            added udev subsystem watch of nvme
20:34:09:0161 FuContext            added udev subsystem watch of drm
20:34:09:0161 FuContext            added udev subsystem watch of drm_dp_aux_dev
20:34:09:0161 FuContext            added udev subsystem watch of serio
20:34:09:0161 FuPluginTest         init
20:34:09:0162 FuContext            added udev subsystem watch of thunderbolt
20:34:09:0162 FuContext            added udev subsystem watch of tpm
20:34:09:0162 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
20:34:09:0162 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
20:34:09:0162 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
20:34:09:0162 XbSilo               file: cecf5e19-796e-9c42-1f2e-e24d6964c722, current:cecf5e19-796e-9c42-1f2e-e24d6964c722, cached: (null)
20:34:09:0162 XbSilo               loading silo with file contents
Loading…                 [*                                      ]20:34:09:0162 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
20:34:09:0162 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
[... cut ...]
20:34:09:0177 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
Loading…                 [-                                      ]
20:34:09:0246 FuCommon             creating path /var/cache/fwupd/motd.d
20:34:09:0246 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
20:34:09:0250 FuMain               No detected devices

20:34:09:0251 FuPlugin             finalize(flashrom)
20:34:09:0251 FuPluginTest         destroy
```

Sean-StarLabs, 2024-06-23:
That's just the version of fwupd being too old; usually just a case of waiting for the distro to update it - nudging them won't do any harm (unless you want to go down the rabbit hole of building it from source)

radekholy24, 2024-06-24:
IIUIC, SLES 15 SP6 is going to contain `fwupd` 1.9.10 and should be released in "mid-2024". Would that be enough?

(FTW, I tried to download and install that version from OpenSUSE's repos but it apparently depends on newer glibc which will probably require an update of a lot of packages which sounds too risky, so I would rather wait for the new SLES version.)

Sean-StarLabs, 2024-06-24:
Yup, assuming they build it with flashrom support

radekholy24, 2024-06-24:
Thank you. I'll report back once they release it and I upgrade.

radekholy24, 2024-07-03:
So, I upgraded to SLES 15 SP6 and got `fwupd` 1.9.10. However, the firmware still does not seem to be recognized:

```
$ sudo fwupdtool get-devices --plugins flashrom --verbose
17:40:44.043 FuDebug              verbose to info (on console 1)
17:40:44.043 FuEngine             starting fwupd 1.9.10…
Loading…                 [ -                                     ]17:40:44.047 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
17:40:44.047 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
17:40:44.047 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
17:40:44.047 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
17:40:44.047 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]17:40:44.047 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]17:40:44.052 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
17:40:44.052 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]17:40:44.053 FuContext            SMBIOS Manufacturer=Star Labs
17:40:44.053 FuContext            SMBIOS EnclosureKind=9
17:40:44.053 FuContext            SMBIOS Family=B62-I
17:40:44.053 FuContext            SMBIOS ProductName=StarBook
17:40:44.053 FuContext            SMBIOS ProductSku=B62-I
17:40:44.053 FuContext            SMBIOS BiosVendor=coreboot
17:40:44.053 FuContext            SMBIOS BiosVersion=24.04
17:40:44.053 FuContext            SMBIOS BiosMajorRelease=18
17:40:44.053 FuContext            SMBIOS BiosMinorRelease=02
17:40:44.053 FuContext            SMBIOS FirmwareMajorRelease=18
17:40:44.053 FuContext            SMBIOS FirmwareMinorRelease=04
17:40:44.053 FuContext            SMBIOS BaseboardManufacturer=Star Labs
17:40:44.053 FuContext            SMBIOS BaseboardProduct=StarBook
17:40:44.053 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
17:40:44.054 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]17:40:44.054 FuEngine             loading metadata for remote 'vendor-directory'
17:40:44.054 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
17:40:44.054 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
17:40:44.054 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
17:40:44.056 FuEngine             2641 components now in silo
Loading…                 [***                                    ]17:40:44.066 FuEngine             plugins disabled: modem_manager, uefi_capsule, acpi_dmar, acpi_facp, acpi_ivrs, acpi_phat, amd_pmc, analogix, android_boot, ata, aver_hid, bcm57xx, bios, ccgx, ccgx_dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros_ec, dell, dell_dock, dfu, dfu_csr, ebitdo, elantp, elanfp, emmc, ep963x, fastboot, focalfp, fpc, fresco_pd, genesys, genesys_gl32xx, goodixmoc, goodixtp, gpio, hailuck, igsc, intel_me, intel_usb4, iommu, jabra, jabra_gnp, kinetic_dp, lenovo_thinklmi, linux_display, linux_lockdown, linux_sleep, linux_swap, linux_tainted, logind, logitech_hidpp, logitech_bulkcontroller, logitech_rallysystem, logitech_scribe, logitech_tap, mediatek_scaler, msr, mtd, nitrokey, nordic_hid, nvme, optionrom, parade_lspcon, pci_bcr, pci_mei, pci_psp, pixart_rf, powerd, qsi_dock, realtek_mst, redfish, rts54hid, rts54hub, steelseries, scsi, superio, synaptics_cape, synaptics_cxaudio, synaptics_mst, synaptics_prometheus, synaptics_rmi, system76_launch, test, test_ble, thelio_io, thunderbolt, ti_tps6598x, tpm, uefi_dbx, uefi_esrt, uefi_pk, uefi_recovery, uf2, upower, usi_dock, vbe, vli, wacom_raw, wacom_usb, wistron_dock
17:40:44.066 FuEngine             enabling flashrom due to HwId 12122d1c-c383-5583-9cb7-3ba8d220913d
17:40:44.066 FuEngine             enabling flashrom due to HwId 12122d1c-c383-5583-9cb7-3ba8d220913d
Loading…                 [***                                    ]17:40:44.066 FuContext            battery threshold now 30
Loading…                 [-                                      ]17:40:44.067 FuPluginFlashrom     coreboot table found at 0x76895000.
17:40:44.068 FuPluginFlashrom     WARNING: No chipset found. Flash detection will most likely fail.
17:40:44.068 FuPluginFlashrom     ========================================================================
17:40:44.068 FuPluginFlashrom     You seem to be running flashrom on an unknown laptop. Some
internal buses have been disabled for safety reasons.
17:40:44.068 FuPluginFlashrom     Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
[... cut ...]
If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

Does it mean that also `flashrom` is too old like in https://github.com/StarLabsLtd/firmware/issues/129 ?

Sean-StarLabs, 2024-07-03:
It does indeed

closed 2024-09-02 by Sean-StarLabs