StarLabsLtd/firmware #175: [StarLite Mk V] Touchscreen intermittent

Sean-StarLabs, 2024-06-25:


Sean-StarLabs, 2024-06-27:
Fix released:
```
Fixed an issue which made the touchscreen intermittently stop responding
Improved Bluetooth compatibility
Decreased the delay on the power button when an OS is loaded
Added an option to set the charging speed
Added an option to disable the lid switch
```

This build has been uploaded to the LVFS, so if you want to install it from there, you can find the steps [here](https://support.starlabs.systems/kb/firmware/methods). However, the LVFS is not designed for speed, so if you'd like to manually install this firmware, please follow the below steps:

First, install flashrom:
Ubuntu:
`sudo apt install flashrom`
Fedora
`sudo dnf install flashrom`
Arch
`sudo pacman -Syyuu flashrom`

Then, connect your charger and run:
```
wget https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkV/coreboot/24.06/24.06.rom
sudo flashrom -p internal -w 24.06.rom -i bios --ifd -n -N
sudo shutdown now
```

Once that has finished, please **shutdown** (not a reboot), wait for 12 seconds until you see the LEDs flicker. Once that happens, you can reconnect the charger and carry on.

If you see a message Message recipient disconnected from message bus without replying  or /dev/mem mmap failed: Operation not permitted, you'll need to add the `iomem=relaxed` kernel parameter. The two below commands will do this for you:

```
sudo sed -i 's/quiet/quiet iomem=relaxed/g' /etc/default/grub
sudo update-grub
```

EverlastingOS, 2024-07-01:
Have installed the latest bios (24.06). But seems not to still be able to connect to some bluetooth devices, ex my Surface Mobile Mouse, with my stabook can connect to.

Sean-StarLabs, 2024-07-01:
> Have installed the latest bios (24.06). But seems not to still be able to connect to some bluetooth devices, ex my Surface Mobile Mouse, with my stabook can connect to.

Distro?

EverlastingOS, 2024-07-01:
> > Have installed the latest bios (24.06). But seems not to still be able to connect to some bluetooth devices, ex my Surface Mobile Mouse, with my stabook can connect to.
> 
> Distro?

Debian 12, elementaryOS, ZorinOS and Fedora.

Sean-StarLabs, 2024-07-01:
See #180

EverlastingOS, 2024-07-01:
> See #180

Done, but same issue, can connect to my airpods but not the surface mouse.

closed 2024-07-05 by Sean-StarLabs