StarLabsLtd/firmware #176: [Star Lite V] Wacom tablet not detected

mpetuska, 2024-06-29:
In Gnome settings (on silverblue 40), wacom tablet tab is visible, yet opening it says that wacom tablet is not detected and won't allow to configure anything there. Pen works fine in gimp and other drawing software, it's just that I cannot configure anything at os level.

Sean-StarLabs, 2024-06-29:
There isn't any Wacom hardware in the StarLite - what are you wanting to configure?

EverlastingOS, 2024-07-01:
Do also have the same issue for a hour or two, after that and some reboots Fedora/debian do enable the touchscreen. Dont know why it doing so.

Sean-StarLabs, 2024-07-01:
> Do also have the same issue for a hour or two, after that and some reboots Fedora/debian do enable the touchscreen. Dont know why it doing so.

That's not related - see #175

mpetuska, 2024-07-01:
I was hoping to configure pen buttons there.

Sean-StarLabs, 2024-07-01:
As far as I know, there isn't a graphical way to change them, but it can be done with xinput

thmichel, 2024-07-03:
It think Gnome refers to any touchsscreen as "Wacom tablet". On the Microsoft Surface Pro the touchscreen is listed as Wacom tablet as well though it isn't one. Ther you can configure the Pen buttons. Would love to see this working on the StarLite as well.

ZekeZDev, 2024-07-03:
> It think Gnome refers to any touchsscreen as "Wacom tablet". On the Microsoft Surface Pro the touchscreen is listed as Wacom tablet as well though it isn't one. Ther you can configure the Pen buttons. Would love to see this working on the StarLite as well.

This was my understanding as well. I found some reference to it on the arch linux page about it however the commands seem to be for x11 to add a device manually I am not sure what the process for wayland would be and if it would work with this pen. But having it report to gnome settings would be nice, its really painful to use inkscape with the pen because so much functionality relies on right clicking.

ZekeZDev, 2024-07-03:
https://github.com/linuxwacom/libwacom having a look at this repo it might be possible to make a .tablet description to add this pen as a supported device. Otherwise purchasing one of the devices listed here and using that stylus with the starlite instead of the provided one.

thmichel, 2024-07-03:
Thanks for that hint, I got it working now. Might need some tweaks, but you can configure the buttons!

This is what I did:

Create the directory /etc/libwacom
in that directory, create a file called starlite.tablet with following content:
#

[Device]
Name=StarLabs Starlite
Class=ISDV4
DeviceMatch=i2c:27C6:0111
IntegratedIn=Display;System;

[Features]
Stylus=true
Touch=true

recreate libwacom database using "libwacom-update-db /etc/libwacom/".

After that, the device shows up in Gnome system settings (pen settings appear once you touched the display with it).

ZekeZDev, 2024-07-03:
> Thanks for that hint, I got it working now. Might need some tweaks, but you can configure the buttons!
>
> This is what I did:
>
> Create the directory /etc/libwacom in that directory, create a file [...]

I was working on this at the same time as you and I got so close but I messed up the ids ;-;.

```
DeviceMatch=i2c:0018:27C6
```
this is what I had, anyways great job seems like your change works. I don't think you need to put in /etc/wacom (I think thats the old way?) I put mine in /usr/share/libwacom with the rest of the files and it worked. Also it would probably be better to use the version of the tablet in the name incase starlabs release more in the future with changes to the touch screen. I am also not 100% sure if this change would work for the 2k model of the starlites as they could have a different touch screen id? 

To be on the safeside here is what my starlabs-starlite-v-3k.touch file looks like:
```
[Device]
Name=StarLabs Starlite V 3K
Class=ISDV4
DeviceMatch=i2c:27C6:0111
IntegratedIn=Display;System;

[Features]
Stylus=true
Touch=true
```

**If anyone has the 2k model of starlite and can test this works that would be appreciated.**

Steps to test:

```
sudo nano /usr/share/starlabs-starlite-v.touch
```
Copy this into file
```
[Device]
Name=StarLabs Starlite V 3K
Class=ISDV4
DeviceMatch=i2c:27C6:0111
IntegratedIn=Display;System;

[Features]
Stylus=true
Touch=true
```
Save and open wacom settings in gnome (or any de that supports libwacom, eg. kde)

ZekeZDev, 2024-07-03:
@thmichel I currently can only config one button on the pen wbu?

thmichel, 2024-07-03:
Agreed, putting the version in is a good idea. Regarding file location: I'm using Fedora Silverblue where the root filesystem is read only so you can't put the file into /usr/share/libwacom directly. According to the  libwacom documentation you should use /etc/libwacom from version 1.10 onwards; only older versions should use /usr/share/libwacom.

ZekeZDev, 2024-07-03:
oh I had it backwards then... my bad, updated my comment to reflect that. As for the buttons are you able to configure both @thmichel?

thmichel, 2024-07-03:
> @thmichel I currently can only config one button on the pen wbu?

I can see both buttons; not sure if that was because I played around with a Bamboo pen. Rebooted the device and used the Starlabs pen only, still can configure both buttons.

ZekeZDev, 2024-07-03:
hmm let me just reboot my tablet then

ZekeZDev, 2024-07-03:
its working now, I think it might be because I did old method before, not sure tho.

ZekeZDev, 2024-07-03:
[image]

Works Perfectly :D

ZekeZDev, 2024-07-03:
@thmichel do you want me to start a draft pr on upstream (un-draft when 2k users have tested) or make one yourself?

thmichel, 2024-07-03:
> @thmichel do you want me to start a draft pr on upstream (un-draft when 2k users have tested) or make one yourself?

If you could do it would be great, haven't done it before so wouldn't know how to do it anyways :-)

ZekeZDev, 2024-07-03:
neither have I! what could possibly go wrong

Sean-StarLabs, 2024-07-03:
Nice.

`DeviceMatch=i2c:27C6:0111` would be better to use an HID - that'll allow for different displays too. Happy to leave it to me?

ZekeZDev, 2024-07-03:
> Nice.
> 
> `DeviceMatch=i2c:27C6:0111` would be better to use an HID - that'll allow for different displays too. Happy to leave it to me?

that is what is being used?

anyways I already made a pr and a descriptor, I am assuming you might have some 2k starlites on hand would there be any hardware difference to cause issue with this pr?

https://github.com/linuxwacom/libwacom/pull/735

ZekeZDev, 2024-07-03:
@Sean-StarLabs If you have any changes ping me in the pr, ill include them asap and once the 2k is tested ill un-draft.

Sean-StarLabs, 2024-07-03:
> that is what is being used?

No, that's not an HID, that's an I2C identifier.

ZekeZDev, 2024-07-03:
Ah misunderstood the wording, if you can provide what the hid would look like I can give it a test

ZekeZDev, 2024-07-03:
I was mainly going off the surface example they had which used i2c

Sean-StarLabs, 2024-07-03:
Probably `acpi:STAR0002` but can't say for sure until the 24.07 changes are finalised

thmichel, 2024-07-03:
> Nice.
> 
> `DeviceMatch=i2c:27C6:0111` would be better to use an HID - that'll allow for different displays too. Happy to leave it to me?

Happy to leave it to anyone more qualified than me :-)

I just  did this as first hack to get it working, probably height and width should be added as well.  Looking at the tablet files from libwacom they all use either usb or i2c for DeviceMatch, so assumed that would be the way to go.

ZekeZDev, 2024-07-03:
@Sean-StarLabs I am not sure if that works in the context of libwacom? since its hardcoded to take 3 values for the DeviceMatch acpi:STAR0002 is 2

ZekeZDev, 2024-07-03:
ok I am thinking you might have just had a typo since acpi:STAR:0002 works.

ZekeZDev, 2024-07-03:
do you think any changes are needed for 2k starlite or should it work?

ZekeZDev, 2024-07-03:
pushed the changes

Sean-StarLabs, 2024-07-03:
I think it needs testing thoroughly, as unstitching incorrect quirks for various distros is a painful process.

ZekeZDev, 2024-07-03:
> I think it needs testing thoroughly, as unstitching incorrect quirks for various distros is a painful process.

thats fair, ill leave it in draft for now. the hid change you suggested didn't actually work, I thought it did but there was a caching issue, removing the cache and updating the db it didnt work. I have reverted my pr to i2c for testing. I think the way its done seems standard for libwacom but if you have suggestions add them to the pr review.

ZekeZDev, 2024-07-03:
id also argue this is quite an important feature for a device optionally shipping with a first party pen so the sooner this can be upstream the better.

ZekeZDev, 2024-07-03:
I cant think of much in the way of distro quirks for this besides de's that dont support libwacom. the whole process is pretty abstracted I dont see why this wouldn't work across distros. If I get some time I might give it a test it should be easy enough to load ventoy with all your oem images and see if it has any issues.

Sean-StarLabs, 2024-07-03:
> the hid change you suggested didn't actually work

It wouldn't, because you don't have one of the firmware changes that it would require.

> all your oem images and see if it has any issues

That wouldn't yield anything helpful.

> id also argue this is quite an important feature

There's no need to argue, but stability will always trump features. Even if we find no issues with your exact PR, what's the point in sending one upstream now, only for it to need changing in 3 weeks or so?

ZekeZDev, 2024-07-03:
> > the hid change you suggested didn't actually work
> 
> It wouldn't, because you don't have one of the firmware changes that it would require.

Ok I misunderstood that it was already pushed.

> > all your oem images and see if it has any issues
> 
> That wouldn't yield anything helpful.

I was kinda just going to do this anyways lol

> > id also argue this is quite an important feature
> 
> There's no need to argue, but stability will always trump features. Even if we find no issues with your exact PR, what's the point in sending one upstream now, only for it to need changing in 3 weeks or so?

Not meaning to argue, only expression. Ill keep an eye on the firmware update for this change and I can unrevert the commit. I'd be happy to transfer the pr to you if you would prefer, I am not sure best way to do that, either make a new pr or I add you to the repo? There is also the wacom-hid-descriptors issue but I am not sure if anything more needs adding to that.

ZekeZDev, 2024-07-03:
Ive done some further testing and its really bizarre, second button is reporting but bindings for it dont work. The first button does work and I have been able to bind that. For context first button is the one furthest from the tip of the pen.

ZekeZDev, 2024-07-03:
it seems like there is some janky behaviour elsewhere in the stack I just tested with my wacom intuos it has the same weird issue.

thmichel, 2024-07-03:
> it seems like there is some janky behaviour elsewhere in the stack I just tested with my wacom intuos it has the same weird issue.
> 
> I am going to grab a new pen soon ill be curious if this behavior also happens with that.

This doesn't seem to be StarLite specific. I got the same result when testing with a Microsoft Surface book. "libinput debug-events" does not show the press of the second button either. tested with a Bamboo pen and the StarLabs pen.

ZekeZDev, 2024-07-04:
> This doesn't seem to be StarLite specific. I got the same result when testing with a Microsoft Surface book. "libinput debug-events" does not show the press of the second button either. tested with [...]

interesting, I am wondering if it has something to possibly do with eraser function? Since I have found that the device works with both buttons in krita. With the one I cant bind set to eraser, and the binded one the correct binding.

thmichel, 2024-07-04:
Could be. I guess the problem is that non-wacom tablets always use the generic pen definitions as they don't deliver a pen id (at least that's what the documentation says). Maybe one of the buttons is automatically mapped to the eraser.

presslab-us, 2024-07-04:
For some reason with Manjaro I needed to do things a bit different.  Here is the config file (note file extension and pipe instead of colon in device id)

**/etc/libwacom/starlabs-starlite-v.tablet** 
```
[Device]
Name=StarLabs Starlite V
Class=ISDV4
DeviceMatch=i2c|27C6|0111
IntegratedIn=Display;System;

[Features]
Stylus=true
Touch=true
```

**sudo libwacom-update-db --buildsystem-mode > /etc/udev/hwdb.d/66-libwacom.hwdb**

rogercreagh, 2024-07-08:
So despite following the lead of @thmichel and @ZekeZDev  at about https://github.com/StarLabsLtd/firmware/issues/176#issuecomment-2205791567 above I am still getting No Tablet Detected in settings after a reboot.

When I ran `libwacom-update-db /etc/libwacom` I got a message saying libwacom not installed, use apt install libwacom which I did and then and it seemed to accept the libwacom-update-db but still no tablet detected.

My liteV has come with Ubuntu 24.04 - if you have got earlier ones maybe you have 22.04 and that is different? 22.04 default install excludes a lot of useful stuff that was standard in 22.04, maybe there is something else missing?

How do I find out whether I have 2k or 3k screen - I ordered on the pre-order back in September and mails subsequently said all those orders would have 3k screens. My display resolution is shown as 2880x1920 and firmware version is 24.06

rogercreagh, 2024-07-08:
On Ubuntu 24.04 libwacom is not installed by default so before following the guidance above you need to `sudo apt install libwacom-bin`

Also note the extension name for the file you create in `'/etc/libwacom/` must be `.tablet` not `.touch` as in some posts above. 

It takes a while to recognise the 'wacom tablet' after rebooting and I've had a few system errors since when using Settings app and as noted above the button 2 (bottom of the bar) doesn't map to anything.

ZekeZDev, 2024-07-08:
> On Ubuntu 24.04 libwacom is not installed by default so before following the guidance above you need to `sudo apt install libwacom-bin`
> 
> Also note the extension name for the file you create in `'/etc/libwacom/` must be `.tablet` not `.touch` as in some posts above. This was why mine didn't work in the post above. Needed a full power off, not a restart before it eventually worked, initially the tablet was there but not the pen.
> 
> It takes a while to recognise the 'wacom tablet' after rebooting and I've had a few system errors since when using Settings app and as noted above the button 2 (bottom of the bar) doesn't map to anything.

oh yeah my bad I meant tablet but typed touch I have fixed that. I am surprised that ubuntu doesnt ship libwacom since its a soft dependency of gnome settings, regardless that's good to know and glad you have it working now, I am really confused still what causes the button 2 mapping issue but ill look into it more later.

Sean-StarLabs, 2024-09-05:
Fixed with https://github.com/linuxwacom/libwacom/commit/6229e9e50917c6a5f27eef711d6428dc823a9274

closed 2024-09-05 by Sean-StarLabs

LionsPhil, 2024-12-23:
For anyone else trying to temporarily pull the now-upstream version ( https://github.com/linuxwacom/libwacom/blob/master/data/starlite-mk-v.tablet ) atop the current Ubuntu libwacom release, note that upstream changed the DeviceMatch format in https://github.com/linuxwacom/libwacom/commit/767ae7ecaaf47213bce967dfce115ed6a6f9118f . The colons in the version in this bug are correct for libwacom 2.10.0-2, current in Ubuntu at the time of writing, but later versions will need pipes.

What I've done is put the (edited to colons) upstream version in `/usr/share/libwacom/`, so dpkg *should* overwrite it when it's actually part of the package, and I'll never even have to know the transition happened.