StarLabsLtd/firmware #179: How to update Starlite MKV firmware with fwupd?

M1cha, 2024-07-01:
Even though I see the coreboot update on the LVFS website, the CLI client does not.
* archlinux.
* I manually compiled and installed flashrom `flashrom 1.4.0-devel (git:v1.2-1356-gfb14ceb3)`
* `fwupdmgr get-plugins`:
```
...
flashrom:
  Flags:                • Disabled
                        • Required hardware was not found
                        • Enabled if hardware matches
                        • Loaded from an external module
                        • Will measure elements of system integrity around an update
...
```
* `fwupdmgr get-devices`
```
│
├─Internal SPI Controller:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             DMI:Star Labs
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         8332559d-efc1-5670-9baa-e06a25556a96 ← MTD\VENDOR_Star-Labs&NAME_0000:00:1f.5
│                         f6c775ba-3b67-581a-a5a4-f3afc4da245c ← MTD\VENDOR_Star-Labs&PRODUCT_StarLite&NAME_0000:00:1f.5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│
├─N200:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000017
│     Vendor:             Intel
│     GUIDs:              90cc499c-3166-5538-b337-ac47d715d50b ← CPUID\PRO_0&FAM_06&MOD_BE
│                         dcb3a326-6c55-59ef-a947-d5ae46f7b4ec ← CPUID\PRO_0&FAM_06&MOD_BE&STP_0
│     Device Flags:       • Internal device
│
└─NVME 512GB SSD:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    T1103N0L
      Vendor:             Silicon Motion, Inc. (NVME:0x126F)
      Serial Number:      2024011200374
      GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
                          165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
                          1cbae981-d77f-584d-9a85-843d4fcb56d5 ← NVME 512GB SSD
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

Sean-StarLabs, 2024-07-01:
> I manually compiled and installed flashrom flashrom 1.4.0-devel (git:v1.2-1356-gfb14ceb3)

That's specific - does flashrom on its own work?

Sean-StarLabs, 2024-07-01:
Cool, what's the output of:
`sudo fwupdtool get-devices --plugins flashrom --verbose`

M1cha, 2024-07-01:
```
12:31:21.475 FuDebug              verbose to info (on console 1)
12:31:21.475 FuEngine             starting fwupd 1.9.21…
12:31:21.476 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]12:31:21.477 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:31:21.477 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:31:21.477 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:31:21.477 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs-testing[1], lvfs[0]
Loading…                 [                                       ]12:31:21.478 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]12:31:21.485 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:31:21.485 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]12:31:21.486 FuContext            SMBIOS Manufacturer=Star Labs
12:31:21.486 FuContext            SMBIOS EnclosureKind=20
12:31:21.486 FuContext            SMBIOS Family=I5
12:31:21.486 FuContext            SMBIOS ProductName=StarLite
12:31:21.486 FuContext            SMBIOS ProductSku=I5
12:31:21.486 FuContext            SMBIOS BiosVendor=coreboot
12:31:21.486 FuContext            SMBIOS BiosVersion=24.05
12:31:21.486 FuContext            SMBIOS BiosMajorRelease=18
12:31:21.486 FuContext            SMBIOS BiosMinorRelease=05
12:31:21.486 FuContext            SMBIOS FirmwareMajorRelease=18
12:31:21.486 FuContext            SMBIOS FirmwareMinorRelease=05
12:31:21.486 FuContext            SMBIOS BaseboardManufacturer=Star Labs
12:31:21.486 FuContext            SMBIOS BaseboardProduct=StarLite
12:31:21.486 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
12:31:21.486 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]12:31:21.486 FuEngine             loading metadata for remote 'vendor-directory'
12:31:21.486 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
12:31:21.486 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
12:31:21.486 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
12:31:21.488 FuEngine             3110 components now in silo
Loading…                 [***                                    ]12:31:21.503 FuEngine             plugins disabled: modem_manager, uefi_capsule, acpi_dmar, acpi_facp, acpi_ivrs, acpi_phat, algoltek_usb, amd_pmc, amd_gpu, analogix, android_boot, ata, audio_s5gen2, aver_hid, bcm57xx, bios, ccgx, ccgx_dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros_ec, dell, dell_dock, dfu, dfu_csr, ebitdo, elantp, elanfp, emmc, ep963x, fastboot, focalfp, fpc, fresco_pd, genesys, genesys_gl32xx, goodixmoc, goodixtp, gpio, hailuck, igsc, intel_me, intel_usb4, iommu, jabra, jabra_gnp, kinetic_dp, lenovo_thinklmi, linux_display, linux_lockdown, linux_sleep, linux_swap, linux_tainted, logind, logitech_hidpp, logitech_bulkcontroller, logitech_rallysystem, logitech_scribe, logitech_tap, mediatek_scaler, msr, mtd, nitrokey, nordic_hid, nvme, optionrom, parade_lspcon, pci_bcr, pci_mei, pci_psp, pixart_rf, powerd, qsi_dock, realtek_mst, redfish, rts54hid, rts54hub, steelseries, scsi, superio, synaptics_cape, synaptics_cxaudio, synaptics_mst, synaptics_prometheus, synaptics_rmi, synaptics_vmm9, system76_launch, test, test_ble, thelio_io, thunderbolt, ti_tps6598x, tpm, uefi_dbx, uefi_esrt, uefi_pk, uefi_recovery, uf2, upower, usi_dock, vbe, vli, wacom_raw, wacom_usb, wistron_dock
Loading…                 [***                                    ]12:31:21.503 FuContext            battery threshold now 30
Loading…                 [-                                      ]12:31:21.504 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
Loading…                 [************************************** ]12:31:21.530 FuEngine             FuUsbBackend:
  Name:                 usb
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
DoneColdplug:           true
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false

12:31:21.536 FuEngine             resetting update motd timeout

No detected devices
12:31:21.540 FuMain               No detected devices
```

`disabling plugin because: failed to startup using flashrom: no HwIDs found` looks interesting.

Sean-StarLabs, 2024-07-01:
```echo -e "[3d9415bb-3027-541b-99b7-cf21e5383bdb]\nPlugin = flashrom" | sudo tee /var/lib/fwupd/quirks.d/flashrom.quirk
sudo systemctl restart fwupd`

M1cha, 2024-07-01:
that worked and it updated successfully, thx :)
Do other distros need that as well or do they already ship that by default? Either way, it'd probably a good idea to mention that in your guide.

Sean-StarLabs, 2024-07-01:
Nope - Arch won't soon either once it gets the latest build of fwupd

closed 2024-07-01 by Sean-StarLabs

mpetuska, 2024-07-08:
Odd, fwupd only lists N200 and NVME for me, no SPI controller (nor touchscreen for that matter). I'm on f40 silverblue

mpetuska, 2024-07-09:
Could we reopen this for touchscreen hardware not detected issue?

Sean-StarLabs, 2024-07-09:
Can you elaborate on the issue?

mpetuska, 2024-07-13:
Here's my `fwupdmgr get-devices` output to better illustrate the issue.
```
$ fwupdmgr get-devices
Star Labs StarLite
│
├─N200:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000017
│     Vendor:             Intel
│     GUIDs:              90cc499c-3166-5538-b337-ac47d715d50b ← CPUID\PRO_0&FAM_06&MOD_BE
│                         dcb3a326-6c55-59ef-a947-d5ae46f7b4ec ← CPUID\PRO_0&FAM_06&MOD_BE&STP_0
│     Device Flags:       • Internal device
│   
├─NVME 1TB SSD:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    T1103N0L
│     Vendor:             Silicon Motion, Inc. (NVME:0x126F)
│     Serial Number:      2024011200035
│     GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
│                         165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
│                         abd4111a-12e7-5c7c-9e79-35da2766ab3a ← NVME 1TB SSD
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
```
Note how `Internal SPI Controller` is not listed and there's nothing about touchscreen here (not sure if it's supposed to be).

Sean-StarLabs, 2024-07-13:
> Cool, what's the output of: `sudo fwupdtool get-devices --plugins flashrom --verbose`

Will need this

mpetuska, 2024-07-14:
@Sean-StarLabs here it is:
```
12:17:42.694 FuDebug              verbose to info (on console 1)
12:17:42.695 FuEngine             starting fwupd 1.9.21…
Loading…                 [ -                                     ]12:17:42.736 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:17:42.737 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:17:42.737 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:17:42.738 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]12:17:42.739 FuEngine             client certificate now exists: nothing to do
Loading…                 [*                                      ]12:17:42.759 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:17:42.760 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]12:17:42.823 FuContext            SMBIOS Manufacturer=Star Labs
12:17:42.823 FuContext            SMBIOS EnclosureKind=20
12:17:42.823 FuContext            SMBIOS Family=I5
12:17:42.823 FuContext            SMBIOS ProductName=StarLite
12:17:42.823 FuContext            SMBIOS ProductSku=I5
12:17:42.823 FuContext            SMBIOS BiosVendor=coreboot
12:17:42.823 FuContext            SMBIOS BiosVersion=24.05
12:17:42.823 FuContext            SMBIOS BiosMajorRelease=18
12:17:42.823 FuContext            SMBIOS BiosMinorRelease=05
12:17:42.823 FuContext            SMBIOS FirmwareMajorRelease=18
12:17:42.823 FuContext            SMBIOS FirmwareMinorRelease=05
12:17:42.823 FuContext            SMBIOS BaseboardManufacturer=Star Labs
12:17:42.823 FuContext            SMBIOS BaseboardProduct=StarLite
12:17:42.823 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
12:17:42.824 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]12:17:42.824 FuEngine             loading metadata for remote 'vendor-directory'
12:17:42.824 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
12:17:42.825 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
12:17:42.825 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
12:17:42.830 FuEngine             2648 components now in silo
Loading…                 [***                                    ]12:17:42.878 FuEngine             plugins disabled: modem_manager, uefi_capsule, acpi_dmar, acpi_facp, acpi_ivrs, acpi_phat, algoltek_usb, amd_pmc, amd_gpu, analogix, android_boot, ata, audio_s5gen2, aver_hid, bcm57xx, bios, ccgx, ccgx_dmc, cfu, ch341a, ch347, colorhug, corsair, cpu, cros_ec, dell, dell_dock, dfu, dfu_csr, ebitdo, elantp, elanfp, emmc, ep963x, fastboot, focalfp, fpc, fresco_pd, genesys, genesys_gl32xx, goodixmoc, goodixtp, gpio, hailuck, igsc, intel_me, intel_usb4, iommu, jabra, jabra_gnp, kinetic_dp, lenovo_thinklmi, linux_display, linux_lockdown, linux_sleep, linux_swap, linux_tainted, logind, logitech_hidpp, logitech_bulkcontroller, logitech_rallysystem, logitech_scribe, logitech_tap, mediatek_scaler, msr, mtd, nitrokey, nordic_hid, nvme, optionrom, parade_lspcon, pci_bcr, pci_mei, pci_psp, pixart_rf, qsi_dock, realtek_mst, redfish, rts54hid, rts54hub, steelseries, scsi, superio, synaptics_cape, synaptics_cxaudio, synaptics_mst, synaptics_prometheus, synaptics_rmi, synaptics_vmm9, system76_launch, test, test_ble, thelio_io, thunderbolt, ti_tps6598x, tpm, uefi_dbx, uefi_esrt, uefi_pk, uefi_recovery, uf2, upower, usi_dock, vbe, vli, wacom_raw, wacom_usb, wistron_dock
Loading…                 [***                                    ]12:17:42.879 FuContext            battery threshold now 30
Loading…                 [ -                                     ]12:17:42.879 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
Loading…                 [**************************             ]12:17:42.921 FuBackend            adding suitable BlueZ device: /org/bluez/hci0/dev_DA_90_5D_26_68_19
Loading…                 [************************************** ]12:17:42.922 FuEngine             FuUsbBackend:
  Name:                 usb
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
DoneColdplug:           true
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false

12:17:42.929 FuEngine             resetting update motd timeout

No detected devices
12:17:42.934 FuMain               No detected devices
```

Sean-StarLabs, 2024-07-14:
> ```
> echo -e "[3d9415bb-3027-541b-99b7-cf21e5383bdb]\nPlugin = flashrom" | sudo tee /var/lib/fwupd/quirks.d/flashrom.quirk
> sudo systemctl restart fwupd
> ```

This is the solution

mpetuska, 2024-07-17:
That did nothing for me.

Sean-StarLabs, 2024-07-17:
Flashrom version?

mpetuska, 2024-07-22:
```
$ flashrom --version                                                        ─╯
flashrom v1.3.0 on Linux 6.9.9-200.fc40.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
```

Sean-StarLabs, 2024-07-22:
Thanks, what's the output of:
`sudo flashrom -p internal`

Sean-StarLabs, 2024-07-22:
You need to set `iomem` kernel param to relaxed