StarLabsLtd/firmware #180: [StarLite Mk V] Bluetooth LE discovery broken on latest SFI

Sean-StarLabs, 2024-07-01:
SFIs newer than 20420220 can't discover LE devices.

Can fix with the below:

```
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/snapshot/linux-firmware-20240220.tar.gz
sudo tar -xzvf linux-firmware-20240220.tar.gz --strip-components=2 --directory /usr/lib/firmware/intel linux-firmware-20240220/intel/ibt-0040-2120.sfi linux-firmware-20240220/intel/ibt-0040-2120.ddc
```

EverlastingOS, 2024-07-01:
Have tried, same issue, can connect to my airpods but not the surface mouse.

sodiboo, 2024-07-01:
I'm also experiencing this same issue. On NixOS, with this patch (https://github.com/sodiboo/nix-config/commit/c5b735eec853fac5c84ea34bf255f6b740584993) applied followed by a reboot, the issue persists. Still cannot connect my mouse.

sodiboo, 2024-07-01:
I also noticed in #175, one stated change is

> Improved Bluetooth compatibility

So i updated my BIOS to the latest version following those instructions. It went well, but didn't solve this issue. Still cannot connect my mouse.

rogercreagh, 2024-07-05:
Same here but Bluetooth 3 mouse (actually mice) works ok after the BIOS update.

edpenz, 2024-07-06:
Fixed my Logitech G604 mouse too.

I also needed to do a `sudo update-initramfs -r` and reboot for it to take effect on Ubuntu 24.04.

sodiboo, 2024-07-06:
@EverlastingOS What distro do you use? It's interesting that the fix works for several others, but not for me or for you. NixOS *is* weird and the installation instructions don't work outright, so in that case my failure might be because i'm doing it wrong or something. But it doesn't work for you. Are you also using NixOS or a similar non-standard distro that might cause these issues, or were you able to just straight up run the suggested commands, but it still didn't work?

EverlastingOS, 2024-07-07:
> @EverlastingOS What distro do you use? It's interesting that the fix works for several others, but not for me or for you. NixOS _is_ weird and the installation instructions don't work outright, so [...]

Hey, i have tried with Fedora 40 and Elementary OS.

rbm78bln, 2024-07-07:
> Have tried, same issue, can connect to my airpods but not the surface mouse.

Can confirm. Discovering the surface mouse still doesn't work with neither firmware nor BIOS update. (Distro is Arch)

Nevertheless connecting to an already paired BLE device does work indeed: I had the bd address of a CSR Bluetooth device (RasPi3b+) [temporarily changed](https://github.com/bluez/bluez/blob/master/tools/bdaddr.rst) to the address of my StarLite, paired my surface mouse with it, and transferred the relevant files and keys from /var/lib/bluetooth from the "donor device" back to my StarLite. Now as a result both know each other's keys and addresses and I can use my mouse nicely. This also confirms that it's basically only a discovery issue.

Maybe this will also do as a temporary workaround for you guys (@EverlastingOS, @sodiboo, etc.)

sodiboo, 2024-07-07:
oh wow i did not even think of using a separate device to pair them! haha that sounds like a great hacky workaround, thanks @rbm78bln

EverlastingOS, 2024-07-07:
> Can confirm. Discovering the surface mouse still doesn't work with neither firmware nor BIOS update. (Distro is Arch)
>
> Nevertheless connecting to an already paired BLE device does work indeed: I had [...]

Did copy the files for my mouse, put it onto the starlite device, so can see that the mouse it there in the list but cannont connect, did try connect with CLI, "got Failed to pair: org.bluez.Error.Failed"

rbm78bln, 2024-07-07:
> got Failed to pair: org.bluez.Error.Failed

Are you trying to pair the mouse with the starlite again? After transferring the pairing keys to the tablet, I just used "connect xx:xx:xx...." to connect to the mouse from the tablet on the bluetoothctl-command line. While doing the nasty stuff, additionally I made sure that all involved devices had bluetooth turned off completely and also rebooted all hosts afterwards.

EverlastingOS, 2024-07-08:
> Are you trying to pair the mouse with the starlite again? After transferring the pairing keys to the tablet, I just used "connect xx:xx:xx...." to connect to the mouse from the tablet on the [...]

Do get "Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local". Do you mod something in the files from your raspi device? I just copied the two files "attributes" and "Info", did you change anything in them as key or so?

rbm78bln, 2024-07-09:
> Do you mod something in the files from your raspi device?

Nope, I didn't change the files, but intermittently I had changed the hardware MAC address of the raspi to match one from my starlite. Then, after copying the files and rebooting the Starlite, I didn't try pairing again, but just used the mouse.

EverlastingOS, 2024-07-15:
Still having this issue even with  24.0710 release

Sean-StarLabs, 2024-07-15:
Nothing in 24/07/10 relating to Bluetooth - should get some stuff back from Intel tomorrow to hopefully fix

EverlastingOS, 2024-07-15:
oh okey, great to know :) Lets see if they can give us a fix :)

Sean-StarLabs, 2024-07-26:
Got a fix. Complexity of implementing said fix into coreboot is probably 312 out of 10, so might take a few days :/

rogercreagh, 2024-08-09:
Bluetooth 5.1 keyboard (see discussion https://github.com/StarLabsLtd/firmware/discussions/204) is detected and connecting fine with firmware 24.07. Bluetooth 5.0 mouse still not detected although 3.0 is (as previously)

MRDGH2821, 2024-09-03:
I updated my firmware to 24.07 - no mouse detected
I did the steps as mentioned - no mouse detected
`sudo update-initramfs -u` - no mouse detected

After each step I rebooted & tried connecting to mouse, same result - no mouse detected.

I'm using [Logitech Vertical Ergonomic Mouse](https://www.logitech.com/en-us/products/mice/lift-vertical-ergonomic-mouse.910-006466.html) and the bluetooth mode is not detected by my tablet

Sean-StarLabs, 2024-09-06:
Fixed with 00bd1d2ebfecc59f98d72892159bdee07ae12d36

jothgard, 2024-09-07:
> Fixed with [00bd1d2](https://github.com/StarLabsLtd/firmware/commit/00bd1d2ebfecc59f98d72892159bdee07ae12d36)

Seems still not find my Microsoft mice on Ubuntu 24.04. Did I miss something?

Sean-StarLabs, 2024-09-09:
> Did I miss something?

How did you install it?

stronnag, 2024-09-09:
```
$ cat /sys/class/dmi/id/bios_version
24.08
```
Still fails to discover **any** BLE devices.

Sean-StarLabs, 2024-09-09:
> Still fails to discover any BLE devices.

How did you install it?

stronnag, 2024-09-09:
Everything else looks great (Battery getting above 97% at last, no more annoying keyboard backlight flashes). Just no BLE discovery for 2 x serial devices and 1 Omcon BP monitor.

Arch Linux with LQX 10.9 kernel.

jothgard, 2024-09-16:
> `sudo fwupdmgr --force install /tmp/coreboot-I5.cab`

Trying to install, do only get :
No supported devices found

@Sean-StarLabs How can I install oreboot-I5.cab manually? Do see that the EC version is still on 24.07 and not updated to 24.08 as the Bios version.

Sean-StarLabs, 2024-09-16:
You can't manually update an EC, it'll update automatically when you connect a charger in G3

jothgard, 2024-09-16:
> You can't manually update an EC, it'll update automatically when you connect a charger in G3

Do you med i need to have the charger connected to the starlite 5 while typing: flashrom -p internal -w 24.08.rom  -i bios --ifd -n -N

Sean-StarLabs, 2024-09-16:
No - it's automatic, i.e. you don't do anything

Just turn if off, wait for 12 seconds so it's in G3, then connect a charger.

jothgard, 2024-09-16:
> Just turn if off, wait for 12 seconds so it's in G3, then connect a charger.

Oh okey, yeah, did just do this and sees that im in 24.08 on EC as well now, but cannot connect to the surface mouse.

adamtclark, 2024-09-17:
Same issue for me - my bluetooth keyboard (Logitech Keys-to-Go 2) doesn't show up in the manager as visible on my Starlite Mk V (tried both via GUI and bluetoothctl, running Ubuntu 24.04.1 LTS, firmware version 24.08 installed via LVFS). The keyboard connects no problem on every other device I own (other Ubuntu computers, Android phone), so seems to be an issue with the Mk V. Any suggestions on how to proceed?

adamtclark, 2024-09-20:
And just for completeness - I can confirm that I am able to connect to the keyboard on my Mk V without problems through a bluetooth dongle (TP-link UB500), after running:
sudo modprobe btusb
sudo systemctl restart bluetooth.service

and unplugging/replugging the dongle, as described here: https://askubuntu.com/questions/1423752/ubuntu-20-04-unable-to-use-a-bluetooth-dongle-tp-link-ub500

srund, 2024-09-23:
Same issue; BLE devices are not discovered on my Starlite. Annoying since I work with BLE devices daily. The possible fix ([here](https://github.com/StarLabsLtd/firmware/issues/180#issue-2383781373)) didn't resolve the issue.

Running Ubuntu 24.04  as installed by Starlabs, updated to kernel 6.8.0-45-generic and using BIOS 24.08/24.8 as updated with `fwupd` .

System logs look reasonable to me (except the lack of BLE events). Even when Bluez log on debug level. Sniffed usb traffic to the wireless chip look somewhat fishy but I need compare with a working device before saying too much.

Sean-StarLabs, 2024-10-17:
Fixed with d6de2690dd547a014309086a9cc435e2b58f091e

jothgard, 2024-10-17:
> Fixed with [d6de269](https://github.com/StarLabsLtd/firmware/commit/d6de2690dd547a014309086a9cc435e2b58f091e)

Thanks! How can i download the latest? did try with wget for 24.09, seems not to download the rom.

Sean-StarLabs, 2024-10-18:
LVFS - or download the rom from the repo:
https://github.com/StarLabsLtd/firmware/tree/main/StarLite/MkV/coreboot/24.09

jothgard, 2024-10-18:
> LVFS - or download the rom from the repo: https://github.com/StarLabsLtd/firmware/tree/main/StarLite/MkV/coreboot/24.09

trying to download the link above with wget, but the rom file is only some kb, should be 20 mb+?

MRDGH2821, 2024-10-19:
Flashed 24.09, but bluetooth mouse & my android phone still not detected.
I also see coreboot showing 0 MB RAM.

jothgard, 2024-10-20:
> Flashed 24.09, but bluetooth mouse & my android phone still not detected. I also see coreboot showing 0 MB RAM.
> 
> Update: I downgraded to 24.08. On 24.09, CPU usage hit 100% almost everytime I opened Firefox or VS Code. And the system becomes almost unresponsive for few seconds. In 24.08, CPU usage rarely ever goes 100% & the system is responsive even at 90%+ CPU usage.

yes, same here, thats why my starlite is crazy slow on 24.09 even om boot up..

Sean-StarLabs, 2024-10-21:
What kernels and SFI's please?

24.09 on max benchmarks 20% higher than 24.08, so is this just a "first notice and bail out"?

MRDGH2821, 2024-10-21:
> What kernels and SFI's please?

No idea about SFI. I have reinstalled various OSes in my tabnlet, so I'm assuming latest SFI provided by OS (currently on [Endeavour OS](https://endeavouros.com/))
[Liquorix Kernel](https://liquorix.net/) v6.11.2 & v6.11.3

> 24.09 on max benchmarks 20% higher than 24.08, so is this just a "first notice and bail out"?
I stayed on 24.09 for 2-3 days before downgrading.

Sean-StarLabs, 2024-10-21:
Okay, can you chuck the latest ones on:

```
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/snapshot/linux-firmware-20241017.tar.gz
sudo tar -xzvf linux-firmware-20241017.tar.gz --strip-components=2 --directory /usr/lib/firmware/intel linux-firmware-20240220/intel/ibt-0040-2120.sfi linux-firmware-20240220/intel/ibt-0040-2120.ddc
```

> I stayed on 24.09 for 2-3 days before downgrading.

Default firmware settings and P-State (assuming Endeavour sets the P-State)?

MRDGH2821, 2024-10-21:
I fixed the date than ran the commands.
```
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/snapshot/linux-firmware-20241017.tar.gz
sudo tar -xzvf linux-firmware-20241017.tar.gz --strip-components=2 --directory /usr/lib/firmware/intel linux-firmware-20241017/intel/ibt-0040-2120.sfi linux-firmware-20241017/intel/ibt-0040-2120.ddc
```

After rebooting my [tablet](https://starlabs.systems/pages/starlite), [phone](https://in.nothing.tech/pages/phone-1), & [computer](https://www.gigabyte.com/us/Motherboard/X670-AORUS-ELITE-AX-rev-10-12) and switching off-on my [bluetooth keyboard](https://amzn.in/d/bV09yXF) & [wireless mouse](https://www.logitech.com/en-us/products/mice/lift-vertical-ergonomic-mouse.html), mobile-tablet-pc were interconnected with each other correctly without any problems. 
Keyboard was now discoverable by tablet & successfully connected too. Only the mouse was not able to connect.

> Default firmware settings and P-State (assuming Endeavour sets the P-State)?

I changed charging rate to `1.0 C` in BIOS settings. Apart from that I did not change anything else. 
Don't know about the P-State. 
Endeavour OS is arch based, so I'm assuming whatever default Arch is setting, E-OS is also using the same.

Sean-StarLabs, 2024-10-21:
> Only the mouse was not able to connect.

How about with 24.09? If not, what's the mouse?

> I'm assuming whatever default Arch is setting, E-OS is also using the same.

Probably - that's just balanced which means the firmware settings have much more impact. If you want more performance, just a case of turning them up.

MRDGH2821, 2024-10-21:
> How about with 24.09? If not, what's the mouse?

Flashed 24.09, mouse is not detected.
It is Logitech Lift Vertical Ergonomic Mouse - https://www.logitech.com/en-us/products/mice/lift-vertical-ergonomic-mouse.html 

I'll observe my tablet for few days in 24.09 firmware in regards to performance.

bergmannf, 2024-10-21:
I finished the 24.09 update as well, but it's still not detecting a Razer Basilik X bluetooth mouse.
On top of that I can confirm the performance issues already mentioned earlier that make the tablet feel very sluggish (same issue with it saturating CPUs - so I also opted to downgrade, as it feels really slow) - I've opened a new issue #221  for what feel like performance regression, to maybe not sidetrack this one too much.

Sean-StarLabs, 2024-10-21:
Ha - that's one of the ones I tested with. Bring the latest SFI's in as above.

bergmannf, 2024-10-22:
Hm I tried that on Fedora, but it didn't work - I am not sure I did it entirely correct though, as the firmware files on that distribution seem to be `xz` compressed - so I tried it without compressed files *and* compressed files (and rebuild the initram with `dracut -f` but neither worked).

I am also not sure if I should even need to install any upstream files, given that the package shipped is already the firmware files from 17.10:

```
Name         : linux-firmware
Version      : 20241017
```

And looking at the installed files from that package shows that the ones extracted from the tar files are symlinks:

```
lrwxrwxrwx. 1 root root 20 Oct 19 02:00 /usr/lib/firmware/intel/ibt-0040-2120.ddc.xz -> ibt-0040-0041.ddc.xz
lrwxrwxrwx. 1 root root 20 Oct 19 02:00 /usr/lib/firmware/intel/ibt-0040-2120.sfi.xz -> ibt-0040-1020.sfi.xz
```

jothgard, 2024-10-30:
running 24.10, still cannot find my Microsoft mouse. Using Microsoft Modern Mobile Mouse.

jothgard, 2024-10-31:
> 24.10 hasn't even been tested yet, so it'll just complicate things. Please post the SFI versions and kernel

Running Fedora silverblue on kernel 6.11.3-200.fc40.x86_64, how can i see what SFI version im on?

Sean-StarLabs, 2024-10-31:
`sudo dmesg  | grep hci0`

jothgard, 2024-10-31:
> sudo dmesg | grep hci0

got output:

[   27.803755] Bluetooth: hci0: Device revision is 2
[   27.803760] Bluetooth: hci0: Secure boot is enabled
[   27.803761] Bluetooth: hci0: OTP lock is enabled
[   27.803762] Bluetooth: hci0: API lock is enabled
[   27.803763] Bluetooth: hci0: Debug lock is disabled
[   27.803764] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   27.803765] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[   27.803829] Bluetooth: hci0: DSM reset method type: 0x00
[   27.845244] Bluetooth: hci0: Found device firmware: intel/ibt-0040-2120.sfi
[   27.845261] Bluetooth: hci0: Boot Address: 0x100800
[   27.845263] Bluetooth: hci0: Firmware Version: 151-5.24
[   29.378012] Bluetooth: hci0: Waiting for firmware download to complete
[   29.378716] Bluetooth: hci0: Firmware loaded in 1497525 usecs
[   29.378755] Bluetooth: hci0: Waiting for device to boot
[   29.395737] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[   29.395742] Bluetooth: hci0: Device booted in 16606 usecs
[   29.400525] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0040-2120.ddc
[   29.402750] Bluetooth: hci0: Applying Intel DDC parameters completed
[   29.406429] Bluetooth: hci0: Firmware timestamp 2024.5 buildtype 1 build 151
[   29.406437] Bluetooth: hci0: Firmware SHA1: 0x00000000
[   29.409967] Bluetooth: hci0: Fseq status: Success (0x00)
[   29.409973] Bluetooth: hci0: Fseq executed: 00.00.02.36
[   29.409975] Bluetooth: hci0: Fseq BT Top: 00.00.02.36

adamtclark, 2024-11-03:
Still same problem for me - BLE keyboard and mouse work fine if I connect them over a USB bluetooth dongle, but don't work at all using the internal Bluetooth of my MkV. Output from dmsg is pasted below - I'm still running Ubuntu 24.04.01.

[    3.848780] Bluetooth: hci0: Device revision is 2
[    3.848786] Bluetooth: hci0: Secure boot is enabled
[    3.848787] Bluetooth: hci0: OTP lock is enabled
[    3.848788] Bluetooth: hci0: API lock is enabled
[    3.848789] Bluetooth: hci0: Debug lock is disabled
[    3.848790] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    3.848792] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[    3.848886] Bluetooth: hci0: DSM reset method type: 0x00
[    3.851894] Bluetooth: hci0: Found device firmware: intel/ibt-0040-2120.sfi
[    3.851913] Bluetooth: hci0: Boot Address: 0x100800
[    3.851914] Bluetooth: hci0: Firmware Version: 151-5.24
[    5.387316] Bluetooth: hci0: Waiting for firmware download to complete
[    5.387549] Bluetooth: hci0: Firmware loaded in 1499656 usecs
[    5.387613] Bluetooth: hci0: Waiting for device to boot
[    5.404547] Bluetooth: hci0: Device booted in 16564 usecs
[    5.404561] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[    5.405741] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0040-2120.ddc
[    5.407565] Bluetooth: hci0: Applying Intel DDC parameters completed
[    5.410578] Bluetooth: hci0: Firmware timestamp 2024.5 buildtype 1 build 151
[    5.410584] Bluetooth: hci0: Firmware SHA1: 0x00000000

rbm78bln, 2024-11-03:
Yes, same for me. Also booting takes considerably longer until the device is finally up with 24.09.

Before I had a workaround running to get me over the ble pairing hurdle, now even already the paired mouse doesn't connect properly anymore. Actually the mouse connects for a short period time, service resolution seems to fail and then it is disconnected again.

I have created a few logs including versions and trances and everything. Please find them attached.

BTW: Wouldn't it make sense to reopen this ticket again? Clearly this doesn't seem to be solved.

[blebug_bluetoothctl.txt](https://github.com/user-attachments/files/17611054/blebug_bluetoothctl.txt)
[blebug_dmesg.txt](https://github.com/user-attachments/files/17611055/blebug_dmesg.txt)
[blebug_hcitrace.txt](https://github.com/user-attachments/files/17611056/blebug_hcitrace.txt)
[blebug_syslog.txt](https://github.com/user-attachments/files/17611057/blebug_syslog.txt)
[blebug_hcitrace_pcapng_base64.txt](https://github.com/user-attachments/files/17611062/blebug_hcitrace_pcapng_base64.txt)

stronnag, 2024-11-03:
Apropos boot time, you are probably waiting for the non-appearing tpm device. May be resolved by:

```
sudo systemctl mask dev-tpmrm0.device
```

rbm78bln, 2024-11-03:
> you are probably waiting for the non-appearing tpm device

Naah, was fixing this mid-year already... 😎

bergmannf, 2024-11-24:
Just as an update - it seems the new Bios 24.10 did not help with this issue either:
```
sudo dmidecode -s bios-version
24.10
```

And the HCI info:

```
[   49.073343] Bluetooth: hci0: Device revision is 2
[   49.073352] Bluetooth: hci0: Secure boot is enabled
[   49.073353] Bluetooth: hci0: OTP lock is enabled
[   49.073354] Bluetooth: hci0: API lock is enabled
[   49.073354] Bluetooth: hci0: Debug lock is disabled
[   49.073355] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   49.073357] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[   49.073419] Bluetooth: hci0: DSM reset method type: 0x00
[   49.116960] Bluetooth: hci0: Found device firmware: intel/ibt-0040-2120.sfi
[   49.116982] Bluetooth: hci0: Boot Address: 0x100800
[   49.116984] Bluetooth: hci0: Firmware Version: 151-5.24
[   50.643101] Bluetooth: hci0: Waiting for firmware download to complete
[   50.643424] Bluetooth: hci0: Firmware loaded in 1490686 usecs
[   50.643457] Bluetooth: hci0: Waiting for device to boot
[   50.660353] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[   50.660356] Bluetooth: hci0: Device booted in 16519 usecs
[   50.663812] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0040-2120.ddc
[   50.665385] Bluetooth: hci0: Applying Intel DDC parameters completed
[   50.668370] Bluetooth: hci0: Firmware timestamp 2024.5 buildtype 1 build 151
[   50.668378] Bluetooth: hci0: Firmware SHA1: 0x00000000
[   50.672355] Bluetooth: hci0: Fseq status: Success (0x00)
[   50.672363] Bluetooth: hci0: Fseq executed: 00.00.02.36
[   50.672365] Bluetooth: hci0: Fseq BT Top: 00.00.02.36
```

Still won't detect LE devices.

Sean-StarLabs, 2024-11-25:
>  Just as an update - it seems the new Bios 24.10 did not help with this issue either:

Nothing in there that would

> [   50.668370] Bluetooth: hci0: Firmware timestamp 2024.5 buildtype 1 build 151

The fix is in the week 33 builds,  so you'll need to manually update those or wait for you distro to get the newer firmware.

altatechor, 2024-11-28:
> The fix is in the week 33 builds

Where can I find those, and how would I update manually? With `flashrom`?

thebatdan, 2024-12-29:
I'm having no luck connecting bluetooth mice. I've tried two different models, Dell MS5120W and Logitech G604, neither will connect. Updating the SFI makes no difference.

Running Fedora 41 with firmware version 24.10

[60 more replies]

closed 2025-07-18 by Sean-StarLabs