StarLabsLtd/firmware #181: Wayland completely unusable on new StarLite V (arch linux)

isomorphian, 2024-07-04:
I have just received my StarLite V and so far have had nothing but issues. I have 5 other intel-based devices running Hyprland with no issues. On this device, running Hyprland, any graphical application will crash after about 30 seconds. 

In addition, with either Gnome or KDE on wayland, the desktop environment will flicker uncontrollably before crashing after about 30 seconds. 

Also, I am experiencing random kernel panics running the stock Arch kernel. 

I have had somewhat better experience running X11. Apps will not crash, although every so often, upon starting the X server, the device will lock up completely, and not even allow me to switch virtual terminals. The only way out of this situation is to cut the power by holding the power button. 

I am on the latest firmware that was just released to solve the intermittent touchscreen issue.

Sean-StarLabs, 2024-07-04:
kernel 6.9 by any chance?

isomorphian, 2024-07-04:
> kernel 6.9 by any chance?

Yes, 6.9.7-arch-1-1 installed with the "linux" package on arch

Sean-StarLabs, 2024-07-04:
6.9 has bugs; should be fine if you use 6.8 instead

thmichel, 2024-07-04:
Works well with Fedora Kernel 6.9.7 by the way.

isomorphian, 2024-07-05:
Thank you for the information. I'm out of town for a few days but when I get back I will try downgrading my kernel.

I'm glad to hear things are working better on Fedora. I'd prefer to stay on Arch because that's what I'm most familiar with, but I'm glad to know that's an option.

stronnag, 2024-07-06:
It continues with the latest Arch updates (`6.9.8-zen1-1-zen`),  the journal reporting:
```
Jul 06 08:26:55 piglet kernel: i915 0000:00:02.0: [drm] *ERROR* Fault errors on pipe A: 0x00000080
```
Though not as frequently as previously.

PhilT, 2024-07-09:
I'm using Wayland on NixOS with Kernel 6.9.7. No flickering problems that I've noticed so far.

isomorphian, 2024-07-12:
Seems that the problem is in libwayland-client.so

I noticed that applications only crash after being interacted with via touchscreen. As long as I do not touch the screen at all, there are no problems. But if I interact with applications via touchscreen, they will continue to function for about 30 seconds before crashing.

I recorded a crash in Rnote, and a crash in Firefox.

Dmesg reports this message upon Rnote crashing:
rnote[843]: segfault at 603aa950dc8d ip 0000790883a122af sp 00007ffe3dda1e40 error 4 in libwayland-client.so.0.23.0[790883a0d000+6000] likely on CPU 0 (core 0, socket 0)

Dmesg reports another, similar message upon Firefox crashing:
traps: crashreporter[1349] general protection fault ip:764328dcfcc4 sp:7ffde78d7ff8 error:0 in libwayland-client.so.0.23.0[764328dce000+6000]

isomorphian, 2024-07-12:
apologies for closing and reopening. I have never submitted an issue before, I did not realize how it worked.

isomorphian, 2024-07-12:
Ack! Sorry !

isomorphian, 2024-07-12:
I tried all three officially supported Arch Linux kernels:

linux 6.9.8
linux-zen 6.9.8
linux-lts 6.6.38

None of them make any difference.

Is it possible that there is some firmware that this device requires, which is not included in the linux-firmware package?

isomorphian, 2024-07-12:
Interestingly, this does not affect all applications on my machine, but it does affect most of them.

Apps that crash upon touch input:
Firefox
Rnote
LibreOffice
Pavucontrol
Thunar File Manager
PCManFM (GTK3 version)

Apps that do not crash upon touch input:
KeepassXC
Blueman
GNU Octave
PCManFM (QT6 version)
qBittorrent
Discord

isomorphian, 2024-07-12:
Just finished recovering my system from a kernel panic that interrupted a system upgrade. This machine really is not happy to be running Arch Linux.

isomorphian, 2024-07-12:
I think the kernel panics during system upgrade could be an upstream issue with Arch, according to this forum post:

https://forum.endeavouros.com/t/kernel-panic-during-update-broke-my-system-thanks-god-and-linux-community-for-the-arch-chroot/51839/20?page=3

(they are talking about Endeavour, not Arch, but close enough to make me think it could be upstream issue)

isomorphian, 2024-07-12:
I made the mistake of enabling gdm.service to see if starting Gnome that way would work. But I should have started it instead of enabling. Now I am trapped inside a completely unusable Gnome that freezes my system after a few seconds of runtime. Impossible to get to a terminal. Have to boot into recovery USB for the 50th time since I got this machine.

isomorphian, 2024-07-12:
Both Gnome and KDE Plasma on wayland are completely unusable, only way to escape is to power off. This is not good. Wayland is the default option.

They start fine on X11 but have issues with the pointer not tracking the pen correctly.

isomorphian, 2024-07-12:
i3 and awesomewm seem to work fine on X11 although the pen is lacking pressure-sensitivity.

Hyprland is broken, but I say that's probably their fault anyways.

Interestingly, Sway seems to work perfectly! :)

isomorphian, 2024-07-12:
I apologize for spamming this issue. Tomorrow I will summarize my findings in another issue and I will close this one.

diggit, 2024-07-12:
@isomorphian
I am also on archlinux with same issues.
In KDE/Gnome, "solution" to disable output to "ghost" output. Then the flickering stopped. SDDM probably also has some way of output configuration to apply the same same workaround.

Due to some reason, these DEs and SDDM (and probably more) see this `Unknown-1` output and try to use it.

Contents of `/sys/class/drm/`, notice the `coreboot8/simple-framebuffer.0`
```
card0 -> ../../devices/platform/BOOT0000:00/coreboot8/simple-framebuffer.0/drm/card0/
card0-Unknown-1 -> ../../devices/platform/BOOT0000:00/coreboot8/simple-framebuffer.0/drm/card0/card0-Unknown-1/
card1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/
card1-DP-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-1/
card1-DP-2 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/
card1-eDP-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/
card1-HDMI-A-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-1/
card1-HDMI-A-2 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-2/
card1-HDMI-A-3 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-3/
renderD128 -> ../../devices/pci0000:00/0000:00:02.0/drm/renderD128/
```

isomorphian, 2024-07-12:
> @isomorphian
> 
> I am also on archlinux with same issues.
> 
> In KDE/Gnome on wayland, workaround is to disable "ghost" output. Then the flickering stopped. SDDM probably also has some way of output configuration to apply the same same workaround.
> 
> 
> 
> Due to some reason, these DEs and SDDM (and probably more) see this `Unknown-1` output and try to use it.
> 
> 
> 
> Contents of `/sys/class/drm/`, notice the `coreboot8/simple-framebuffer.0`
> 
> ```
> 
> card0 -> ../../devices/platform/BOOT0000:00/coreboot8/simple-framebuffer.0/drm/card0/
> 
> card0-Unknown-1 -> ../../devices/platform/BOOT0000:00/coreboot8/simple-framebuffer.0/drm/card0/card0-Unknown-1/
> 
> card1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/
> 
> card1-DP-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-1/
> 
> card1-DP-2 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-DP-2/
> 
> card1-eDP-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-eDP-1/
> 
> card1-HDMI-A-1 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-1/
> 
> card1-HDMI-A-2 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-2/
> 
> card1-HDMI-A-3 -> ../../devices/pci0000:00/0000:00:02.0/drm/card1/card1-HDMI-A-3/
> 
> renderD128 -> ../../devices/pci0000:00/0000:00:02.0/drm/renderD128/
> 
> ```

Interesting. I also noticed that Unknown-1 device, and I had assumed it had to do with the touchscreen. I'll experiment with this when I get home.

TheJackiMonster, 2024-07-17:
I might have found a cause for this... potentially. I had some issue running into a framebuffer problem during boot (which appeared in log from `dmesg`). So I added the i915 module to `/etc/mkinitcpio.conf` for [early KMS start](https://wiki.archlinux.org/title/Kernel_mode_setting#Early_KMS_start). Now my issue during boot is gone but I encounter heavy flickering in my Wayland session. Didn't happen before that.

So my guess is that it's coming from this module...

stronnag, 2024-07-17:
The Arch Testing / Zen kernels (6.9.x) do not work well on the Starlite V.
The AUR LQX kernel (6.9.9) works very, very well (Wayland, Gnome).  No flickering etc. Touch and GJS OSK are stable and predictable. 
Bit of pain to build the kernel from source, but it was worth it.

isomorphian, 2024-07-22:
@stronnag Thank you! This solved all my problems. 

I wonder what's different in the Liquorix kernel vs the official Arch kernels. Whatever it is, it is working great!

TheJackiMonster, 2024-07-22:
> I wonder what's different in the Liquorix kernel vs the official Arch kernels. Whatever it is, is working great!

Tested the `linux-rt` kernel now and it seems to work fine. So that might be a potential option which is pre-built in the official repositories.

stronnag, 2024-07-22:
> > I wonder what's different in the Liquorix kernel vs the official Arch kernels. Whatever it is, is working great!
> 
> Tested the `linux-rt` kernel now and it seems to work fine. So that might be a potential option which is pre-built in the official repositories.

Any Arch kernel prior to 6.9 will work. I'd prefer LTS or 6.8.9 over `-rt` any day,

isomorphian, 2024-07-22:
> Any Arch kernel prior to 6.9 will work. I'd prefer LTS or 6.8.9 over `-rt` any day,

Last time I tested LTS it had the same issues, but I'll give it another shot.

andrewwebber, 2024-07-24:
Just adding my two pence:

Use Arch BTW with Sway/Warbar. Installed using archinstall and selecting minimal + sway. No complaints as Wayland works no less usable than on my desktop hardwares (e.g. full screen Slack crashes on both). 

However I found it impossible to get VLC Player working on the Starlite. 
As a workaround mpv was recommended and worked perfectly.

DavidG-Ghb, 2024-09-16:
Starlite V on Manjaro. Kernel 6.6 LTS works. 6.8 was working, but Manjaro have dropped support, so had to revert to 6.6. Current kernels 6.9, 6.10 and the experimental 6.11 all have the black flickering issues with Wayland. Slightly concerning that this issue remains in newer kernels, which suggests there could be issues with long term support for the Starlite V.

A workaround would be welcome.

stronnag, 2024-09-16:
This still occurs with Arch  6.9. 6.10 kernels on 24.08.

It does not occur with the Arch "lqx" 6.10 kernels (which, AFAIK, are build with Debian build flags). So it is probably an Arch (and derivatives) build issue rather than an upstream kernel issue.

DavidG-Ghb, 2024-09-16:
Yes, tested on 24.08

isomorphian, 2024-09-18:
yes, I've been using lqx with no problems, but the official arch kernels all have issues with graphical apps.

MRDGH2821, 2024-10-06:
In my distro hopping journey, I faced problems in Endeavour OS (which is arch based)
So I fixed it in this way:

https://forum.endeavouros.com/t/unknown-1-display-causing-screen-flickering-on-main-display/60435

https://forum.endeavouros.com/t/after-updating-kernal-framebuffer-coreboot8-probe-with-driver-framebuffer-failed-with-error-17/60389

And later switched to Fedora 40 (which is working fine with kernel 6.10.10).

I feel like this is a common issue with coreboot related systems🤔

Waiting for a permanent fix because I wanna use Endeavour OS

diggit, 2024-10-06:
@MRDGH2821 yeah that is what I wrote in my comment above https://github.com/StarLabsLtd/firmware/issues/181#issuecomment-2226431176

It's interesting, that sway (probably all wlroots based wayland compositors) don't even list nor enable this output, so no issues there.

TheJackiMonster, 2024-10-08:
> I freshly installed Endeavour OS in online mode & it installed kernel `v6.11.2-arch1-1`. So far no flickering & doesn't crash when external monitors are plugged in.
> 
> In other words, I didn't do any special [steps](https://github.com/StarLabsLtd/firmware/issues/181#issuecomment-2395501283). It just worked out of the box.
> 
> I feel that others can try kernel v6.11 from default source/upstream.

Doesn't work for me on Archlinux. Using 6.11.2-arch1-1 still gives me flickering as soon as GDM (which is using Wayland) appears and logging into GNOME as Wayland session even results in it freezing so I have to press the power button for shutdown.

MRDGH2821, 2024-10-08:
> Doesn't work for me on Archlinux. Using 6.11.2-arch1-1 still gives me flickering as soon as GDM (which is using Wayland) appears and logging into GNOME as Wayland session even results in it freezing so I have to press the power button for shutdown.

I see this inside `/proc/cmdline`

```
initrd=\0b6dd5cb07694ca283ac1472cda598c8\6.11.2-arch1-1\initrd nvme_load=YES nowatchdog rw root=UUID=a56e2122-bf54-4394-a71a-25f540c4cd52 rw root=UUID=a56e2122-bf54-4394-a71a-25f540c4cd52 systemd.machine_id=0b6dd5cb07694ca283ac1472cda598c8
```

Referred this https://wiki.archlinux.org/title/Kernel_parameters

I'm using EndeavourOS KDE Plasma 6.1.5 & Wayland if that matters

stronnag, 2024-10-08:
> Doesn't work for me on Archlinux. Using 6.11.2-arch1-1 still gives me flickering as soon as GDM (which is using Wayland) appears and logging into GNOME as Wayland session even results in it freezing [...]

It does not freeze the computer, it freezes the screen. It is possible to log in via `ssh` and reboot into a known good (lqx?) kernel.

mixmastamyk, 2024-10-12:
Been using Fedora 40 with Phosh (Wayland) for months without issue (since the Mint it came with was too old).  First kernel was 6.8 I think, now on 6.10.x.  Have updated the firmware twice, now on 24.08.

The root problem you're having does not appear to be Wayland.  Might want to edit the title of this issue.

MRDGH2821, 2024-10-12:
> I freshly installed Endeavour OS in online mode & it installed kernel `v6.11.2-arch1-1`.
> So far no flickering & doesn't crash when external monitors are plugged in.
>
> In other words, I didn't do any [...]

My bad, next day I started my tablet & Framebuffer error came back.

It just works after fresh install. 2nd boot onwards it won't boot into login screen.

I switched to lqx kernel.

Sean-StarLabs, 2024-10-15:
What version of wayland?

Sean-StarLabs, 2024-10-17:
From my testing, d6de2690dd547a014309086a9cc435e2b58f091e does it, but let me know if not

stronnag, 2024-10-17:
Lets see ...
* Downloaded latest from https://fwupd.org/lvfs/devices/com.starlabs.I5.coreboot
* booted with iomem==relaxed

```
# fwupdmgr --force install 24.09-coreboot-I5.cab
Decompressing…           [            /                          ]
No supported devices found
```

Anything else required?

stronnag, 2024-10-17:
> Lets see ...
>
> * Downloaded latest from https://fwupd.org/lvfs/devices/com.starlabs.I5.coreboot
> * booted with `iomem=relaxed`
>
> ```
> # fwupdmgr --force install 24.09-coreboot-I5.cab [...]

Same result with Arch LTS (6.6.56) and Ach LQX (6.11.3) kernels.

Sean-StarLabs, 2024-10-17:
> Anything else required?

Don't know what you're using - see https://support.starlabs.systems/kb/firmware/getting-started

stronnag, 2024-10-17:
As posted I'm using `fwupdmgr` and **your**  latest publicly published firmware. So lets use `flashrom` instead ...
```
flashrom 1.4.0 (git:v1.4.0) on Linux 6.6.56-1-lts (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Reading ich descriptor... done.
Using region: "bios".
linux_mtd_wp_read_cfg: ioctl: Operation not supported
Failed to unlock flash status reg with wp support.
Reading old flash chip contents... done.
Erase/write done from 600000 to ffffff
```
OK, messages not so encouraging, but let's see ...

It boots. 
```
$ cat  /sys/class/dmi/id/ec_firmware_release 
24.9
```
It appears to be the latest firmware.

It **still flickers**. Not quite as much as before, but still "not fit for purpose".  Not fixed.

Sean-StarLabs, 2024-10-17:
Okay, can you post:
+ a video of the flicker? Or ludicrously verbose description
+ Config of a working and non working kernel
+ Wayland version (or whatever you are using)

stronnag, 2024-10-17:
* (ludicrously) verbose description.  As soon as GDM starts the screen flickers, ie. it alternates between displaying the expected GDM screen and the screen being blank/black. Frequency of the flicker is variable (seemly random) but tending more to the black mode as over time. If you get in quickly enough, you can select user and enter password. User's Gnome Shell starts, screen shows same behaviour, alternating between Gnome screen and blank. The mouse will start as a a normal pointer, but after a while (on movement) becomes a square blob (perhaps 3 times the size of the pointer). Sometimes, the screen will not recover from the blank state (which may be reason for some reports of "lock up"). However, the machine is still responsive e.g to ssh login. But it is unusable as a graphical desktop.
* Good config: Standard Arch linux-lqx kernel. 
[config-good.gz](https://github.com/user-attachments/files/17416658/config-good.gz)
* Bad config: Standard Arch (Zen) kernel. Any Arch kernel post 6.8.9 will exhibit bad behaviour.  
[config-bad.gz](https://github.com/user-attachments/files/17416678/config-bad.gz)
* Wayland version: At least 1.23.0 and 1.23.1
* Same behaviour with 'Weston' as with Gnome Shell.
* On the "bad" kernels, mulitple instances of
```
Oct 17 17:34:55 piglet kernel: i915 0000:00:02.0: [drm] *ERROR* Fault errors on pipe A: 0x00000080
```
seen in the journal.
* From prior experiments, none of the usual kernel flags for bad Intel drm behaviour (e.g. one of more  of ` i915.enable_psr=0 intel_idle.max_cstate=1 i915.enable_dc=0` el al), have any effect on the issue.

I will attempt to record a video, probably tomorrow.

stronnag, 2024-10-17:
Interesting ... 

Working Arch Linux-LQX kernel:
```
Oct 17 17:51:03 piglet kernel: efifb: probing for efifb
Oct 17 17:51:03 piglet kernel: efifb: framebuffer at 0x90000000, using 5400k, total 5400k
Oct 17 17:51:03 piglet kernel: efifb: mode is 1440x960x32, linelength=5760, pages=1
Oct 17 17:51:03 piglet kernel: efifb: scrolling: redraw
Oct 17 17:51:03 piglet kernel: efifb: Truecolor: size=0:8:8:8, shift=0:16:8:0
```
Flickering unusable Arch Linux / Arch Linux Zen kernels:
```
Oct 17 17:34:32 piglet kernel: [drm] Initialized simpledrm 1.0.0 for simple-framebuffer.0 on minor 0
Oct 17 17:34:32 piglet kernel: fbcon: Deferring console take-over
Oct 17 17:34:32 piglet kernel: simple-framebuffer simple-framebuffer.0: [drm] fb0: simpledrmdrmfb frame buffer device
```

bkfirmen, 2024-10-17:
I can also confirm that on arch there is flickering (Im using Gnome). Used EndeavourOS before and there was an issue after an update that I installed yesterday. If you logged out of Gnome you cant log in again into your user account. After a reboot also gdm throwed an error message. A fresh install of endeavour os also didnt fixed it. Logged in the first time worked but after that it was the same as before.

DavidG-Ghb, 2024-10-18:
I can also confirm that I'm still encountering flickering on Manjaro, kernel 6.11.2-4 with firmware 24.09.

I'm running Wayland KDE Plasma 6.1.5 and Wayland KDM so I can use Malit virtual keyboard before login. I have the flickering during login, seems to flicker less if the mouse isn't moving. The flickering disappears after login because I've disabled the 'Unknown-1' virtual display device.

Also surprisingly, after updating the firmware, the laptop is now reporting 0% battery with 100% battery health. If I unplug it, it continues running, so the battery is there.

Sean-StarLabs, 2024-10-18:
`DRM_SIMPLEDRM`  and `FB_EFI` set in the bad config,

Sean-StarLabs, 2024-10-18:
Can you try adding `framebuffer-coreboot` to initramfs?

stronnag, 2024-10-18:
> Can you try adding `framebuffer-coreboot` to initramfs?

Hangs in early boot:
```
framebuffer coreboot8: probe with driver framebuffer failed with error -17
:: running hook udev
:: Triggering uevents ...
```

Sean-StarLabs, 2024-10-18:
Hmm, the order might be relevant, or efifb might be fighting it - I'll try to reach out to an arch dev

Sean-StarLabs, 2024-10-18:
What's the output of:
`sudo dmesg | grep -'efifb|simpledrm'`

And
`cat /proc/fb`?

stronnag, 2024-10-18:
Working (Arch LQX 6.11.4):
```
$ dmesg | grep -e 'efifb\|simpledrm'
[    0.764561] efifb: probing for efifb
[    0.764583] efifb: framebuffer at 0x90000000, using 5400k, total 5400k
[    0.764585] efifb: mode is 1440x960x32, linelength=5760, pages=1
[    0.764586] efifb: scrolling: redraw
[    0.764587] efifb: Truecolor: size=0:8:8:8, shift=0:16:8:0
$ cat /proc/fb 
0 i915drmfb
```
Not working (Arch / Arch-Zen 6.11.4):
```
$ dmesg | grep -e 'efifb\|simpledrm'
[    0.378524] [drm] Initialized simpledrm 1.0.0 for simple-framebuffer.0 on minor 0
[    0.379835] simple-framebuffer simple-framebuffer.0: [drm] fb0: simpledrmdrmfb frame buffer device
[    3.137671] [drm] Initialized simpledrm 1.0.0 for simple-framebuffer.0 on minor 0
[    3.141034] simple-framebuffer simple-framebuffer.0: [drm] fb1: simpledrmdrmfb frame buffer device
$ cat /proc/fb
0 i915drmfb
1 simpledrmdrmfb
```
(the user does not require `sudo` for `dmesg`)

Sean-StarLabs, 2024-10-18:
Thanks, so last suggestion, what if you blacklist `efifb`?

stronnag, 2024-10-18:
> Thanks, so last suggestion, what if you blacklist `efifb`?

Nothing, as it's a builtin on the working LQX kernel and is never loaded on the non-working kernel.
Unless I misunderstand what you meant by "blacklist".

sim31, 2024-10-31:
I use arch as well and I see an issue that might be related to simpledrm. In displays settings I see some unknown display which does not actually exist. Searching around this issue lead me to discover [this post](https://bbs.archlinux.org/viewtopic.php?pid=2189699
), which implies that it is probably a simpledrm issue.

I don't have any other symptoms so I've been ignoring this. There are some problems with screensharing in some apps, but investigating it I was lead to believe it's unrelated.

stronnag, 2024-10-31:
Confirming that adding `initcall_blacklist=simpledrm_platform_driver_init` to the kernel boot line solves the flickering problem for me.

DavidG-Ghb, 2024-11-01:
Testing, this also works for me. The 'unknown' display disappears. No further issues with the KDM Wayland login.

[6 more replies]

closed 2025-01-24 by Sean-StarLabs