StarLabsLtd/firmware #182: Disable Touchpad while typing not working

thmichel, 2024-07-04:
I noticed that the setting "disable touchpad while typing" does not work on Fedora. Not sure about other distros. This is pretty annoying while typing longer texts.

Solution is to create a local override for libinput. To do so, create a file called "/etc/libinput/local-overrides.quirks" with following content:

[Serial Keyboards]
MatchVendor=0x27C6
MatchProduct=0x0111
MatchUdevType=keyboard
AttrKeyboardIntegration=internal

Sean-StarLabs, 2024-07-04:
> I noticed that the setting "disable touchpad while typing" does not work on Fedora.

Which way? And what model?

thmichel, 2024-07-04:
Ah, sorry. Starlite MK V. Amended the title. Without that quirk, the palm on the  palm on the touchpad moves the vorsor around while typing and sometimes initiates a mouse click.

Sean-StarLabs, 2024-07-26:
Fixed with 39dfcc167040e8ee36cc760e584a350142bdc4c7

closed 2024-07-26 by Sean-StarLabs

aral, 2024-09-22:
Just wanted to mention that I’m running firmware 24.08 and the issue still exists on Fedora Silverblue.

aral, 2024-09-25:
Hmm, I wonder if this is actually an issue with tap-to-click not getting disabled when the disable touchpad while typing setting is on in GNOME and/or the StarLite firmware.

I’ve just re-enabled the trackpad but turned off tap to click and I don’t seem to be seeing the problem (and the mouse cursor does disappear while typing, which is did before too).