StarLabsLtd/firmware #184: 24.07 Release

Sean-StarLabs, 2024-07-05:
## StarLite
- [ ] Closes #183: Fix S3 issue
- [ ] Closes #182: Tell the OS keyboard is internal
- [ ] Closes #182: Fix intermittant touchscreen

clhodapp, 2024-07-07:
I used `flashrom` to flash version 24.06 on my StarLite Mk V, and was able to see the version update in `/sys/class/dmi/id/ec_firmware_release`. However, after following the same procedure with this 24.07 daily build, I'm still seeing 24.06 in `/sys/class/dmi/id/ec_firmware_release` and actually 24.05 in `/sys/class/dmi/id/bios_release`.

Is this expected, and 24.07 just hasn't had its baked-in version string updated? Or did something go wrong with my flashing?

PhilT, 2024-07-08:
^^ Description links to issue 182 twice.

thmichel, 2024-07-08:
Installed the update. Seems #182 isn't fixed, if I remove the quirks file the keyboard is not recognized as internal. Also I have a frequent screen flicker after the upgrade.

Sean-StarLabs, 2024-07-08:
> Is this expected

Yes

> Installed the update. Seems https://github.com/StarLabsLtd/firmware/issues/182 isn't fixed

Haven't made any changes yet for #182 

> ^^ Description links to issue 182 twice.

Done

dev-da77, 2024-07-11:
My StarLabe Lite 5 runs EndeavourOS (Archlnux).
flashrom is installed.

I executed

`sudo flashrom -p internal -w <filename>.rom -i bios --ifd -n -N`

with 24.06.rom and 240710.rom

and received:

```
flashrom v1.3.0 on Linux 6.9.8-arch1-1 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Opened /dev/mtd0 successfully
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Reading ich descriptor... done.
Using region: "bios".
Error: Image size (292660 B) doesn't match the expected size (16777216 B)!

```

So I can not update from 24.5 to one of the newer versions.

My SSD in encrypted, if this is from interest.

Sean-StarLabs, 2024-07-11:
Whatever the file is that you're trying to flash, it's too big to fit on the chip

dev-da77, 2024-07-11:
First try: I downloaded with the wrong path (file size: 168,1 KB):

`https://github.com/StarLabsLtd/firmware/blob/master/StarLite/MkV/coreboot/24.06/24.06.rom`

Correct path is (file size: 16,8 MB):

`https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkV/coreboot/24.06/24.06.rom`

Than I could update to 24.6

rogercreagh, 2024-07-13:
Trying to do 24.07 on Ubuntu 24.04

flashrom already installed from previous update to 24.06

`wget https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkV/coreboot/Daily%20Builds/240710.rom` appeared to work ok.  File  240710.rom has arrived with size 16777216 bytes

then running  `sudo flashrom -p internal -w coreboot.rom -i bios --ifd -n -N`
generated this output:

```
$ sudo flashrom -p internal -w coreboot.rom -i bios --ifd -n -N
[sudo] password for roger: 
flashrom 1.4.0-devel on Linux 6.8.0-36-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x7689f000.
Found chipset "Intel Alder Lake-N".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00600000-0x00ffffff) is read-write.
FREG2: Management Engine region (0x00001000-0x00412fff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
At least some flash regions are read protected. You have to use a flash
layout and include only accessible regions. For write operations, you'll
additionally need the --noverify-all switch. See manpage for more details.
OK.
Unable to identify chip, mfg_id: 0xa1, model_id: 0x2818
Found Programmer flash chip "Opaque flash chip" (16384 kB, Programmer-specific) on internal.
Reading ich descriptor... done.
Using region: "bios".
Error: opening file "coreboot.rom" failed: No such file or directory
```
why is it looking for coreboot.rom when no such file was downloaded. Should the command be:
`sudo flashrom -p internal -w 240710.rom -i bios --ifd -n -N`
or the wget be:
`wget https://github.com/StarLabsLtd/firmware/raw/master/StarLite/MkV/coreboot/Daily%20Builds/coreboot.rom`

??? or something else???  - rename 240710.rom to coreboot.rom?  Also what's all that stuff about read-only and locked regions - I dont remember seeing that when going to 24.06 (but maybe I did?)

Sean-StarLabs, 2024-07-13:
Instructions updated

altatechor, 2024-07-13:
did not get errors during the flashing process. but now I cannot boot anymore, always drop into the UEFI shell. Any tips?

Sean-StarLabs, 2024-07-14:
Most likely the Boot Entry wasn't created automatically - what distro are you using?

Most distros can be started with this (replace ubuntu with whatever you're using):
```
fs0:
cd EFI
cd ubuntu
grubx64.efi
```

altatechor, 2024-07-14:
Your help already reached me indirectly [via](https://reddit.com/comments/1e2mmg2/comment/ld4cnuu) someone else you helped, thx!

vdessotti, 2024-07-16:
Hello

When I try the flashrom command it gives me:
`Warning: No chipset found. Flash detection will most likely fail`

I did update to 24.06 before and it went fine.

Thanks,

Sean-StarLabs, 2024-07-17:
Need more info, full output, version, distro

tobbsn, 2024-07-17:
I did this upgrade successfully. At least I think so, since most problems I had with `24.6` seem to be gone now. But both `/sys/class/dmi/id/bios_version` and `/sys/class/dmi/id/ec_firmware_release` still return the old `24.06` and `24.6`. Is it possible that I am actually still running the old firmware?

Sean-StarLabs, 2024-07-17:
Unlikely, but you can check the build date to be sure

vdessotti, 2024-07-21:
Hello,

I am on Zorin OS 17.1 Core
Kernel is 6.5.0-44-generic
24.06 update went fine. 

I am trying to update for 24.07 but it gives this error:

```
flashrom` v1.2 on Linux 6.5.0-44-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Cannot open file stream for /dev/mtd0
coreboot table found at 0x7689f000.
WARNING: No chipset found. Flash detection will most likely fail.
========================================================================
You may be running flashrom on an unknown laptop. We could not
detect this for sure because your vendor has not set up the SMBIOS
tables correctly. Some internal buses have been disabled for
safety reasons. You can enforce using all buses by adding
  -p internal:laptop=this_is_not_a_laptop
to the command line, but please read the following warning if you
are not sure.

Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

Thanks,

dev-da77, 2024-07-25:
@Sean-StarLabs : Can you provide a change list for 240725.rom?

dev-da77, 2024-07-25:
My device is only charging, when shutdown.
If it is in standby, it is not charging.
If it is running, it is discharging, while it is plugged-in.

Sean-StarLabs, 2024-07-25:
> Can you provide a change log for 240725.rom?

As above

> not charging.

No charge LED?

dev-da77, 2024-07-25:
When I plug in the USB cable, the LED is constant red (not blinking) - independent if device is running, standby or shotdown. Gnome battery icon is also showing, that it should charge. But percentage value is counting down.

Sean-StarLabs, 2024-07-25:
What percentage range? Centre of the battery icon in GNOME - is it green or white?

dev-da77, 2024-07-25:
Battery icon is white.

[image]

dev-da77, 2024-07-25:
gnome-power-manager:

[image]

dev-da77, 2024-07-25:
Nevertheless it is discharging. Now it is at 80%.

Sean-StarLabs, 2024-07-25:
> Battery icon is white.

That means charged connected, not charging.

Stays at 80%??

dev-da77, 2024-07-25:
No, it don't stay.

Charge-USB is connected
Charge-LED is red
Gnome battery icon shows, it is charging
Gnome-power-manager shows it is charging

BUT charge value is going down, not up like expected.

At coreboot, charge speed is 0,5

Sean-StarLabs, 2024-07-25:
Presumably, if you flash the previous build, it will charge again?

dev-da77, 2024-07-25:
I'm back on version 24.06 and charging works again.

With 240719 I had the same charging issue like with the newest version 240725, before and after downgrade. Than I downgraded to 240705, but my display flickered. Now I'm back to 24.06. Seams to be working better for me.

closed 2024-07-29 by Sean-StarLabs

echo-devnull, 2024-07-30:
It could be me (probably is) but how does one make use of this "`Add an option to disable waking from suspend by opening the lid`"  feature? ]

I cannot seem to find it in the coreboot menu. (I do see thee "`Adjust the charging frequency to improve stability`" option though and I see that `24.07` is the latest version)

echo-devnull, 2024-07-30:
> Not you, set to off at the moment waiting to see if anyone says anything :)

Ah allright then.   I won't say a thing.

JamesOsborn-SE, 2024-07-30:
> I used `flashrom` to flash version 24.06 on my StarLite Mk V, and was able to see the version update in `/sys/class/dmi/id/ec_firmware_release`. However, after following the same procedure with this [...]

If you run the command below, you can see the version like so:
```
SMBIOS BiosVendor=coreboot
SMBIOS BiosVersion=24.07
```

```shell
sudo fwupdtool get-devices --plugins flashrom --verbose
```

tomellis, 2024-07-31:
Not sure if it's just me but after installing this the screen rotation is now incorrect again. I am using the [workaround described on the starlabs support page](https://support.starlabs.systems/kb/guides/starlite-fixing-rotation-on-older-kernel) but after firmware upgrade I'm back to the wrong orientation. After installing the gnome-shell screen rotation add-on I can get the screen around the right way ([as per reddit](https://www.reddit.com/r/starlabs_computers/comments/1dyebde/starlite_mkv_autorotate_solution/)).

I know it's mentioned this is a symptom of older kernels and fixed upstream but which version fixes it?
I've tried newer [vanilla kernel builds](https://fedoraproject.org/wiki/Kernel_Vanilla_Repositories) in Fedora Silverblue but those didn't seem to include the fix. Default Fedora Silverblue kernel is currently 6.9.11-200.fc40.

Sean-StarLabs, 2024-07-31:
The commands in the support article should be all needed if the kernel doesn't support it. Whether the kernel supports it differs depending on the build (patch was upstreamed into 6.8).

Only other things that's known to cause issues is having an entry in /usr/lib/udev/hwdb.d/60-sensor.hwdb for the StarLite - but AFAIK, no distro has that in it.