StarLabsLtd/firmware #190: StarLite Mk V Pen does not disable touchscreen

rogercreagh, 2024-07-05:
When using the Pen the touchscreen should be disabled or you can't rest your hand on the screen while drawing.

The touchscreen should disable as soon as the pen is detected and the disable should last for a configurable period after the pen goes out of range to cover the case where you lift the pen but leave your hand resting on the ~~paper~~ screen before reapplying it in a different area.

ZekeZDev, 2024-07-06:
I am not sure if its software or firmware level but atleast in krita, xournal and a couple others it seems to do palm rejection with the pen. In the case of krita you can also completely disable touch drawing and require usage of pen which is what I have enabled.

rogercreagh, 2024-07-08:
Ah, so I guess it is the individual applications that have to handle palm rejection - I was assuming it could be done globally whenever the pen is in play.

I'll leave this open for a bit in case there are other suggestions for getting a non-touchscreen-aware application to use the pen and not the touchscreen. I prefer to use the pen rather than finger on the touchscreen almost all the time. I wonder if it is possible to disable the touch screen while leaving pen working - and then if it is possible to have a script which detects the pen and turns off the touch when it sees the pen and renables it when the pen is out of range.

Thanks @ZekeZDev for the pointer to two useful looking apps that I hadn't stumbled upon.

rogercreagh, 2024-07-08:
OneNote (microsoft) does do palm rejection and switch to drawing mode when it sees a pen

rogercreagh, 2024-07-08:
> osk not opening when text fields are viewed

There's an oddity there. Having installed the OSK which works with Gnome 46 (https://extensions.gnome.org/extension/5949/gjs-osk/) I now find that the OSK never opens automatically when click or tap or tab to a text field. The gjs-osk has an icon in the top try which opens it (or closes if already open) and the original poor layout osk still opens if you swipe up from the bottom.

Sean-StarLabs, 2024-07-26:
Fixed with 39dfcc167040e8ee36cc760e584a350142bdc4c7

closed 2024-07-26 by Sean-StarLabs