StarLabsLtd/firmware #191: [Starlite V] Closing the lid doesn't disable the touchscreen

mpetuska, 2024-07-08:
As the title says, the touchscreen remains active after connecting it to external display and closing the lid (keyboard). This also happens when simply turning off internal screen to only have external one active.

This makes it impossible to use the device with external monitor while having it closed (or anything sitting on top of the touchscreen) as touch input keeps getting registered on the touchscreen from contact with keyboard.

rogercreagh, 2024-07-08:
It also seems as if it doesnt go into suspend mode when you close the lid, although it does seem to blank the screen.

If you close it without the power connected and come back an hour or two later you'll likely find your battery flat.

PhilT, 2024-07-08:
Yeah, I've experienced these issues as well. Definitely come to it on when I know I put it to sleep. "Lid" is normally closed when I do this too.

PhilT, 2024-07-11:
Not sure if this is related, possibly a symptom but Firefox also crashes when closing the lid. 
If not related, I can open a separate issue. Let me know.

rogercreagh, 2024-07-12:
Ah, so does Edge. Other apps seem to be ok so far. It (MS Edge browser) also crashes if it goes to sleep without losing the lid.

rogercreagh, 2024-07-12:
I also notice that if you close the lid with the keyboard light on it remains on all the while. So currently if closing the lid have to remember to shutdown the browser and turn off the keyboard backlight at least - which is a hassle when you want to quickly put it aside. The safe option is to always shut it down :-(

Sean-StarLabs, 2024-07-12:
@mpetuska What distro and what position is the Lite in when you close the lid?

mpetuska, 2024-07-12:
Fedora 40 Silverblue. It's horizontal with the hinge at the bottom.

mpetuska, 2024-07-12:
But touch remains active if i lay it flat and turn off the screen in settings.

Sean-StarLabs, 2024-07-12:
"turn off the screen" - Display settings, i.e. stop outputting video?

mpetuska, 2024-07-13:
Yes. In gnome settings turn off internal screen and leave only external screen on -> touch input is still being registered on internal screen even though it's not outputting any video.

cybersynk, 2024-07-21:
I have issues with the lid switch too. I mostly use the the starlite with it's kickstand (about 130 degrees between the screen and the keyboard), and when I go afk I like to simply fold up the keyboard (one of my cats are very interested in tasting my starlite) so it can act like a screen protector. I have experimented a bit with the lid switch and how the event is handled on my system and my diagnosis is that the sensor inside the starlite and the magnet inside the keyboard simply get misaligned, so the event is not being triggered when it should, at least not with the magnet in the keyboard. I did most of my testing with evtest (device event 0), and I've tried multiple (neodymium) magnet to trigger it and basically a bigger magnet does trigger the event very reliably, which the magnet inside the keyboard does not. My best guess is that the magnet in the keyboard is just too small and maybe a bit misplaced since it does not "account" for the saggy placement of the keyboard in certain angles.

Is there maybe a way to increase the sensitivity of the sensor inside the starlite? If not, I might just have to swap the magnet in the keyboard myself, so a peek inside that section (right under the right arrow key) would be appreciated. And as for more long term solutions, a better keyboard would be great!

PhilT, 2024-07-21:
Great write up. This sounds like my experience as well. I noticed lying flat, the tablet seems to suspend correctly (most of the time) but on the stand it's very hit and miss.

rogercreagh, 2024-07-21:
If you simply close the tablet onto the keyboard all it does is turn the screen off. Everything else continues to run and drain the battery at same rate. As noted above the keyboard backlight does not turn off when you shut the lid:

(Incidentally touching the RH side of the screen about half way down with a magnet also turns it off, as you might expect given @cybersynk findings above. So one solution to that problem of missing the magnet might be to cut a length of self-adhesive flexible magnet to stick down the side of the keyboard.)

[image]

The 30 mins straight line is while the lid was closed and the screen off (also the keyboard backlight was turned off before closing).  AFAIK no power hungry apps running. The power light remained on. When opening the lid nothing happens to wake it up, pressing keys does nothing. Pressing the power button once makes it blink slowly off-on. Pressing it again cause the screen to wake up after about 10 secs. This is nothing like the expected behaviour which would be just closing the lid puts it into suspend (low power mode) and opening the lid and pressing the {ANY} key wakes it up.

Note also that in one hour, including the 30mins with lid closed, the battery has dropped from 95% to 75% implying a maximum low use battery life of only 5hrs from 100% to zero - somewhat less than the advertised "up to 11 hours battery life". 

Is this expected or have I got a duff one? It is certainly a lot worse than my StarLiteII when its batteries were new (now on its second battery after 4 years and down to 2 hours life)

What are other people getting for battery life - this really seems pretty poor. The quoted 11 hours life must be with it in suspend mode!

rogercreagh, 2024-07-21:
> Not sure if this is related, possibly a symptom but Firefox also crashes when closing the lid. If not related, I can open a separate issue. Let me know.

Edge also crashes when it goes into suspend mode. On restarting it it it reports the crash and you can restore tabs. On StarLite with Ununtu 22.04 and not using coreboot you can put it in suspend and when you wake it up again Edge is still there intact with all its tabs.

KeepassXC is the similar - if you suspend it looses its database connection - not so on StarLiteII. 

Something has very much changed for the worse with this new StarLiteV - whether it is Ubuntu 24.04, or Wayland instead of X11, or Coreboot instead of AMD I can't say, but it is not good.

rogercreagh, 2024-07-22:
So putting it into Suspend mode on the Ubuntu tray power menu (top right) first and then closing the lid does do the expected thing:
[image]
Opening the lid then brings everything back to life exactly as it was (there is a few seconds delay while it yawns and stretches and puts its feet on the floor and stands up) with all apps intact.

During 8 hrs in suspend the battery went down about 7%,  so 21% a day or if nearly full when suspended something over 4 days in suspended animation mode before it would die. Not as good as my phone, but just about acceptable.

What is needed is for closing the lid to put it into proper suspend mode, not just turning the screen off.

Sean-StarLabs, 2024-07-26:
Fixed with 39dfcc167040e8ee36cc760e584a350142bdc4c7

closed 2024-07-26 by Sean-StarLabs

mpetuska, 2024-08-01:
How does one update to it or will it come automatically via an OTA update?

Sean-StarLabs, 2024-08-01:
https://support.starlabs.systems/kb/firmware/lvfs-requirements
https://support.starlabs.systems/kb/firmware/lvfs-terminal