StarLabsLtd/firmware #193: [StarLite Mk V] Battery doesn't charge to 100%

rbm78bln, 2024-07-16:
Hi all!

The issue is easily summarized: The battery doesn't charge to 100%, eventhough it's selected in the BIOS settings.
This behaviour is independent from the selected charging speed, max charge level is honoured, just doesn't ever reach 1 hundret percent, even if selected. Please find a few screenshots attached.

The device is brand-new, so this shouldn't be a hardware issue.

Please help.

[image]
[image]
[image]

Sean-StarLabs, 2024-07-16:
There's nothing wrong there; it's just the overcharge protection. 

If you're interested:
```
static uint16_t calculate_charging_current(int16_t temperature)
{
	uint8_t	charge_level;
	uint8_t	stop_charging	= false;

	switch (emem_max_charge) {
	case 0xaa:	// 60%
		charge_level = 60;
		break;
	case 0xbb:	// 80%
		charge_level = 80;
		break;
	default:	// 100%
		charge_level = 100;
		break;
	}

	/*
	 * We stop charging if any of the below conditions are met:
	 *	Any alarm bits are set
	 *	The battery reports that it's fully charged
	 *	The battery has charged to target level
	 */

	/* Alarm bit(s) set */
	if (CONFIG(BATTERY) && check_battery_alarm())
		stop_charging = true;

	/* Fully charged */
	if (CONFIG(BATTERY) && battery_is_fully_charged())
		eram_battery_overcharge_protection = 1;

	/* Charged to target level */
	if ((charge_level < 100) && (eram_battery_relative_state_of_charge >= charge_level))
		eram_battery_overcharge_protection = 1;

	/* Overcharge protection active */
	if (eram_battery_overcharge_protection) {
		/* Check if the battery has drained 5% */
		if (eram_battery_relative_state_of_charge < (charge_level - 5))
			eram_battery_overcharge_protection = 0;
		else
			stop_charging = true;
	}

	if (temperature >= 450)		/* 0.0C */
		stop_charging = true;

	/* If there isn't a battery online, trickle charge */
	if (!eram_battery_status)
		return 256;

	if (stop_charging)
		return 0;

	if (temperature < 150 || emem_charging_speed == 2)	/* Below 15.0°C or charging speed 0.2C */
		return eram_battery_design_capacity / 5;	/* 0.2C */

	if (eram_battery_voltage >= (CONFIG_LIMITED_CHARGING_VOLTAGE / 100 * 95) ||
	    emem_charging_speed == 1)				/* Voltage >= 95% of limited charging voltage or charging speed 0.5C */
		return eram_battery_design_capacity / 2;	/* 0.5C */

	return eram_battery_design_capacity;			/* 1.0C */
}
```

closed 2024-07-18 by Sean-StarLabs

rbm78bln, 2024-07-18:
Thank you, great to know!