StarLabsLtd/firmware #194: [StarLite Mk V - Magnetic Keyboard] Certain function key combos like Alt+F4 do not work

edpenz, 2024-07-17:
Per `libinput debug-events --show-keycodes` certain function keys report nothing when pressed at the same time as certain modifier keys:

F4:
* Left Alt + Fn + F4 -> LEFTALT (only)
* Right Alt + Fn + F4 -> RIGHTALT + F4
* F4 -> LEFTMETA + P
* Fn + F4 -> F4
* Left Alt + F4 -> LEFTALT + LEFTMETA+ P

F6:
* Left Alt + Fn + F6  -> LEFTALT (only)
* Right Alt + Fn + F6 -> RIGHTALT + F6

All other tested function key combos seem to work fine.

Tangential remarks:
* Is there a full list of other Fn combos? Ones I've found are:
  * Fn + Left/Right -> HOME/END
  * Fn + Up/Down -> PAGEUP/PAGEDOWN
* IMO any modifier key (shift, alt, ctrl, meta) combo with a function key should automatically behave as though the Fn key is also pressed, as shortcut combos with function keys are significantly more common than combos with media keys.
* A function lock combo as in https://github.com/StarLabsLtd/firmware/issues/192 would also be nice

All tested on stock Ubuntu 24.04 as shipped with the StarLite.

Sean-StarLabs, 2024-07-17:
I'm trying to follow your list, are some just examples?

> IMO any modifier key (shift, alt, ctrl, meta) combo with a function key should automatically behave as though the Fn key is also pressed, as shortcut combos with function keys are significantly more common than combos with media keys.

We can't mix and match them; that'll just be too confusing.

edpenz, 2024-07-17:
> I'm trying to follow your list, are some just examples?

Yes, I've edited it down for clarify. It's just left alt + fn + f4/f6 that are the two the combo that are not reported to the OS as expected.

> We can't mix and match them; that'll just be too confusing.

That's understandable, but raises the importance of the function lock feature. Many OS and app shortcuts that are easy with a modifier + function key become very unergonomic when needing to press all three keys (modifier + fn + function) at once.

Sean-StarLabs, 2024-10-22:
Current way to update: https://github.com/StarLabsLtd/starlite_keyboard_update

edpenz, 2024-10-28:
Tested the new firmware today. Alt+F4 works with fn lock enabled, but still doesn't work directly with Fn+Alt+F4.

Sean-StarLabs, 2024-10-28:
It won't, they share pins so only way will be with fn lock